ghdl/ghdl #1810: Mod for enumerations - VHDL-2008

JimLewis, 2021-06-30:
**Description**

I get a compile failure on:
```VHDL
    ------------------------------------------------------------
    procedure InitSeed (T : time ; UseNewSeedMethods : boolean := TRUE ) is
    ------------------------------------------------------------
      variable ChurnSeed : real ;
    begin
      -- Allow use of UseNewSeedMethods but ignore it
      -- as this is a new method
      -- Convert T to an integer that is no more than 2**30 bits
      RandomSeed := GenRandSeed((T mod (2**30 * std.env.resolution_limit))/std.env.resolution_limit) ;
      Uniform(ChurnSeed, RandomSeed) ;
    end procedure InitSeed ;
```

Message:
```
ghdl -a --std=08 -Wno-hide --work=osvvm --workdir=/c/SynthWorks/Dev/_osvvm/_sim/ghdl/VHDL_LIBS/GHDL-2.0.0-dev/osvvm/v08 -P/c/SynthWorks/Dev/_osvvm/_sim/ghdl/VHDL_LIBS/GHDL-2.0.0-dev /c/SynthWorks/Dev/_osvvm/OsvvmLibraries/osvvm/RandomPkg.vhd
                                 C:/SynthWorks/Dev/_osvvm/OsvvmLibraries/osvvm/RandomPkg.vhd:474:36: no function declarations for operator "mod"
      RandomSeed := GenRandSeed((T mod (2**30 * std.env.resolution_limit))/std.env.resolution_limit) ;
```

Is it mod or is it std.env.resolution_limit?

**Expected behaviour**

I expect it to compile

**How to reproduce?**

Check out the dev branch of OSVVM and compile the osvvm utility library.

**Context**
Please, provide the following information:

- OS:    Windows 10
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

Paebbels, 2021-06-30:
@JimLewis when you split your expression into elementary expressions and multiple variables, you'll what operator fails.

`mod` for physical types was added with VHDL-2008. Maybe this is a missing VHDL-2008 feature in GHDL.

JimLewis, 2021-06-30:
I could recode it as:
``` vhdl
GenRandSeed(T /std.env.resolution_limit) ;
```

Given that GHDL is stuck using fs, then I have 2**31 fs before I have a problem, 
which is 64K seconds = 18 hours of simulation time.   

OTOH, I could calculate it (using REM and knowing both numbers are positive):
```vhdl
      RandomSeed := GenRandSeed((T - (T/2**30)*2**30)/std.env.resolution_limit) ;
```

OK it is ugly but it gets the job done.

commit f33ab7ad (2021-06-30): vhdl: handle mod/rem for physical.  Fix #1810
closed 2021-06-30 by tgingold