ghdl/ghdl #1812: Licensing (GHDL, libghdl, libghw, GRT, pyGHDL, etc.)

umarcor, 2021-07-01:
Let's talk about licensing of the GHDL pieces. The current state is described in [ghdl.github.io/ghdl/licenses](https://ghdl.github.io/ghdl/licenses.html).

- I think we are all good with GHDL's core being GNU GPLv2 or later.
- We would all want VITAL libraries to be updated and released under Apache 2.0, as other IEEE libs, but that is not something we can decide here.
- Docs are CC-BY-SA. I think we all agree with keeping that.
- libghdl and libghw are GPL too. I believe that's not an issue for anyone to use them.

Hence, the pieces which we might have different opinions about are:

- GRT and therefore the executable models or the shared libraries generated with LLVM and GCC backends, or when using `--bind` and `--list-link` with an "external" compiler. GRT is GPL and requires the VHDL sources to be made available along with the executable/library models.
  - This is apparently an stopper for some companies to use GHDL in comparison to Verilator. We already discussed that the case is not exactly the same, and that a ghdlator might be possible (which would not include GRT). On the other hand, we also commented about maybe making GRT LGPL? I'm not sure about the practical difference between GPL and LGPL in this case.
- pyGHDL.dom and therefore the plugins/extensions developed on top of it (e.g. VUnit's parser, Symbolator, Sphinx extensions, asciidoctor extensions...). If the DOM is GPL, all of those projects need to be GPL too.
  - In order for pyGHDL not to be GPL, pyGHDL.libghdl would need to be permissive.
- pyGHDL.lsp might be ok as GPL, since it is used as a CLI tool, not expected to be imported into code sources.
- scripts contains some resources which we might want to be licensed permissively. For instance, the msys2 recipes or the vendor scripts.

Now, the pieces outside of the main repo:

- ghdl-yosys-plugin is licensed under GPL v3. That is not a problem for using it as module in Yosys, however it makes Yosys GPL if the plugin is built into it.
  - We might want to relicense the examples under a permissive license?
- extended-tests is unlicensed. It contains one workflow, four scripts (2 bash, 2 powershell), and a python script.
- setup-ghdl-ci is licensed under MIT just because that's the repo template I used. I believe I'm the author of all the content currently available in that repo.
- ghdl-cosim is currently unlicensed, however, we might consider it an extension of GHDL's main repo, so the documentation is CC-BY-SA and the sources are GPL.
  - There is a C header file with a different license: `vhpi_user` (from IEEE). Similarly, I think we might want to license `vffi_user.h` under a permissive license.
  - (Re)licensing the dockerfiles, workflows and testing scripts would be easy, because I'm the author of those. However, the C/VHDL sources of some of the examples might be more difficult to track. I'm quite sure that all the sources I used were written by either Tristan, Yann or Martin (apart from the contributions which are tracked in the repo), but I wouldn't be able to tell which specific snippets were picked from where. On the one hand, I got several of the examples through search engine archeology (they didn't have an author but I assume it was Tristan). On the other hand, it is arguable whether the most basic examples are actually copyrightable or just the only possible implementation (which is not copyrightable per se).
  - Overall, I would be good with making examples (the whole ghdl-cosim repo) available under a permissive license.
- ghdl-language-server is unlicensed, but the license specified in the package.json of vscode-client is GPL v2 or later. I don't have a good understanding of the possibilities/implications in this case.
- ghdl/docker is currently unlicensed. I think that all the authors might be good with using a permissive license. I'd suggest Apache-2.0, which is the one used in hdl/containers.

Paebbels, 2021-07-01:
We can put the Python code in `pyGHDL` directory under MIT (commonly used by Python projects) or Apache License 2.0 (big brother of MIT license). While we could license LSP differently, I suggest to keep it simple by saying everything in that subdirectory has license X. I suggest to use Apache License 2.0 for it.

I'm ok with having vendor compile scripts as GPL - I can't imagine a problem here. Even when script files are containing pre-defined functions that can be sourced by other scripts, it's still interpreted, so no compilation. I don't assume GPL would expand to other scripts sourcing these files.

I suggest to license `extended-test` as MIT as it is just some CI scripts in YAML, Bash, Powershell

tgingold, 2021-07-03:
I am OK with licensing pyGHDL with the Apache License 2.0

umarcor, 2021-07-03:
@tgingold what about using Apache 2.0 for ghdl-cosim and ghdl/docker too?

tgingold, 2021-07-03:
I'm fine with that.  Aren't you the main author of these repos ?

umarcor, 2021-07-03:
I am the main author of ghdl/docker and of most of the "new" content in ghdl-cosim. However, in ghdl-cosim there are examples which I believe are yours (see the 'rand', 'sin' and 'custom C' examples in https://ghdl.github.io/ghdl-cosim/vhpidirect/examples/index.html), as well as most of the sections about "Type declarations", "Wrapping a simulation" and "Linking object files". Furthermore, I would be ok with you being the gatekeeper (the copyright holder in the header files), in coherence with other C sources in the GHDL repo. Otherwise, should I put myself in the copyright line? Shall I write "GHDL contributors"?

tgingold, 2021-07-03:
I am ok with the license change.
I am not sure about `GHDL contributors` legal meaning.  But for the small examples, we can let them as they are.

ktbarrett, 2022-02-07:
@umarcor 
>libghdl and libghw are GPL too. I believe that's not an issue for anyone to use them.

It is an issue. Linking against a GPL library is considered a combined work and would require the user to release their program under the GPL. I recommend all compiled shared libraries that are intended to be reused outside of GHDL (the application) be released under the LGPL v2.1.

From [gnu.org's FAQ](https://www.gnu.org/licenses/gpl-faq.en.html#IfLibraryIsGPL):
> **If a library is released under the GPL (not the LGPL), does that mean that any software which uses it has to be under the GPL or a GPL-compatible license?**
>Yes, because the program actually links to the library. As such, the terms of the GPL apply to the entire combination. The software modules that link with the library may be under various GPL compatible licenses, but the work as a whole must be licensed under the GPL.

oscargus, 2024-02-19:
Any updates on this? Especially, the possible LGPL licensing of libghw (I am consider using that in a project that is EUPL-1.2, so very close to GPL and GPL is claimed to be "compatible" with EUPL, but, without being a lawyer, I'd prefer if libghw was licensed as LGPL, also this is a Rust project, and they tend to pull in dependencies with lots of different licenses, making it even more challenging to straighten out what actually holds...).

Formally, all authors should approve this, so it requires a bit of work, but if you @tgingold are OK, I can open another thread and try to get permissions from everyone I can find in the logs.

hyfloac, 2025-08-19:
I would also like to assert my interest in having *libghdl* licensed under an LGPL variant.

I have a personal interest in making hardware, and would like to do so professionally. As an individual I have an interest in protecting my IP from being stolen, but I don't have a problem with people using my IP in a fair use context, which is why I provide some of my source code as Source Available (other things I care about less I typically release under MIT). Which is to say, if my IP happens to contain a snippet that solves a problem for someone, even a corporation, they could easily get away with using that, but a company can't take my IP and release it as a product.

If I wanted to release an executable simulation of the hardware to potential employers, I would have to release all of the IP under GPL to that company under the current licensing (and such a right could probably be asserted by the receiver even if I don't explicitly tell them that the IP is covered by the GPL license). While such a simulation would rely quite heavily on the GHDL runtime, it wouldn't usurp the purpose of GHDL.

While GHDL being licensed under GPL doesn't restrict my ability to perform development, it does restrict my ability to allow potential employers to freely test my work.

I know that a change to LGPL would require getting all contributors to agree to the new license, or removing their contributions, but it's still a change I'd like to see.

tgingold, 2025-08-25:
@hyfloac There is already a link exception for grt, so you can build an executable of your vhdl model and distribute it with your own license.

hyfloac, 2025-10-29:
> [@hyfloac](https://github.com/hyfloac) There is already a link exception for grt, so you can build an executable of your vhdl model and distribute it with your own license.

I see that I misunderstood what libghdl was, however I don't see any reasonable way to build an executable using GRT from VHDL without using GHDL.

While many (though not all) the GRT files do have a linking exception, there are at least two assertions throughout the documentation that any application built with GHDL must be licensed as GPL due to the runtime license: 
https://github.com/ghdl/ghdl/blob/master/doc/licenses.rst?plain=1#L9-L11
https://github.com/ghdl/ghdl-cosim/blob/master/doc/vhpidirect/index.rst?plain=1#L20-L23

Here are some of the translation units that do not have the linking exception despite being a part of GRT:
https://github.com/ghdl/ghdl/blob/master/src/grt/grtlink.ads
https://github.com/ghdl/ghdl/blob/master/src/grt/grt-vpi.adb
https://github.com/ghdl/ghdl/blob/master/src/grt/grt-vpi.ads
https://github.com/ghdl/ghdl/blob/master/src/grt/grt-vhpi.adb
https://github.com/ghdl/ghdl/blob/master/src/grt/grt-vhpi.ads
https://github.com/ghdl/ghdl/blob/master/src/grt/grt-readline_gnu.ads
https://github.com/ghdl/ghdl/blob/master/src/grt/grt-no_sundials_c.c

If the documentation is incorrect, it would be nice if it were updated, since that is what drew my attention to this issue in the first place. Additionally, while your statement probably acts as a contractual license, it would be preferable if all of the GRT translation units declared the linking exception to remove any doubt.