ghdl/ghdl #1814: aggregate assignment to a single bit vector port at an instance

jeras, 2021-07-05:
**Context**

This is all using the Git master version of GHDL on Ubuntu 20.04.
```Bash
$ ghdl --version
GHDL 2.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

**Description**

I created an entity with a single bit vector port (generics are used in the actual RTL):
```VHDL
port (
...
   wen : in  std_logic_vector(0 downto 0);
...
);
```
At the instance I used an aggregate assignment to convert the a `std_logic` signal into `std_logic_vector(0 downto 0)`:
```VHDL
signal write : std_logic;
...
port map(
...
wen => (others => write),
...
);
```
While the code compiles just fine, elaboration (`-e`) crashes with:
```Bash
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:553
Call stack traceback locations:
0x55c5da00f2e9 0x55c5da00f6c9 0x55c5da00f945 0x55c5da21baf8 0x55c5da22c9e6 0x55c5da1d9712 0x55c5da1f61b0 0x55c5da1f6493 0x55c5da1f62a0 0x55c5da1c082a 0x55c5da1c0955 0x55c5da1c0fae 0x55c5da1c1744 0x55c5da1c32de 0x55c5da1c3433 0x55c5da1c684f 0x55c5da1cb652 0x55c5da239b26 0x55c5da1a1145 0x55c5da19e298 0x55c5da1ac3da 0x55c5da23e495 0x55c5da17ef86 0x55c5da0b15d5 0x55c5da241db2 0x55c5d9e54af7 0x7f5f290ee0b1 0x55c5d9e531fc 0xfffffffffffffffe
```

**Expected behaviour**

The issue can be workaround by using an intermediate single bit vector signal as a port connection.

**How to reproduce?**

```vhd :file: ent.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity ent is
generic (
  W : integer := 1
);
port (
  wen : in std_logic_vector(W-1 downto 0)
);
end entity ent;

architecture rtl of ent is
begin

  process begin
    report "Hello world from ent." severity note;
    wait;
  end process;

end rtl;
```

```vhd :file: top.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity top is
generic (
  W : integer := 1
);
end entity top;

architecture rtl of top is

  signal write : std_logic;
  -- workaround
  signal wen   : std_logic_vector(W-1 downto 0);

begin

  process begin
    report "Hello world from top" severity note;
    wait;
  end process;

  write <= '0';

  -- workaround
  wen <= (others => write);

  u_ent: entity work.ent
  generic map(
    W => W
  )
  port map(
  -- workaround
    wen => (others => write)
--  wen => wen
  );

end rtl;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 --work=work ent.vhd
ghdl -a --std=08 --work=work top.vhd
ghdl -e --std=08 --work=work top
ghdl -r --std=08 --work=work top
```

jeras, 2021-07-14:
I tested this on another simulator just to be sure this was legal code, and it is.

umarcor, 2021-07-16:
I can confirm on MSYS2:

```
# ghdl -a --std=08 --work=work top.vhd 

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r348.g7d5c6c48) [Dunoon edition]
Compiled with GNAT Version: 10.3.0
Target: x86_64-w64-mingw32
T:\msys2\setup-msys2\test-ghdl\
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 --work=work -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o top.o top.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:553
Call stack traceback locations:
0x7ff77c434112 0x7ff77c434551 0x7ff77c4347ef 0x7ff77c51c2b5 0x7ff77c52df4a 0x7ff77c59a028 0x7ff77c56c462 0x7ff77c56c865 0x7ff77c56c5d9 0x7ff77c5c05a5 0x7ff77c5c070a 0x7ff77c5c0ea2 0x7ff77c5c1922 0x7ff77c5c3d97 0x7ff77c5c3ecf 0x7ff77c5c7cc1 0x7ff77c5cd8c1 0x7ff77c58706e 0x7ff77 0x7ff77c57952c 0x7ff77c5d998d 0x7ff77c5db40e 0x7ff77c5dcce5 0x7ff77c3a2702 0x7ff77c3a13bf 0x7ff77c3a14f4 0x7ffb1acd7032 0x7ffb1b3c264f
******************************************************************
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

tgingold, 2021-08-24:
Now fixed.

closed 2021-08-24 by tgingold