ghdl/ghdl #1815: Alternative way for invoking ghdl to use code coverage

jocorso, 2021-07-08:
I am currently evaluating the use of VHDL code coverage with ghdl-gcc and gcov.

All the information that I found related with ghdl and code coverage uses -a and -e arguments in addition to other arguments for gcc, to create .gcna and .gcdo files. These files are necessary to generate html results.
I normally invoke ghdl with -i and -m instead of -a and -e.

I prefer -i & -m because you can import all the files to ghdl simulation with no hierarchy dependencies, and you can create entities with unconstrained ports.

I created the following repository on [gitlab](https://gitlab.com/jcorrecher/ghdl_code_coverage) as an example. 
During the development of that mini-project I realized the following things.

- When I use -a & -e all the steps works ok (as you can see in the following copied terminal text).

```console
[user@archlin sim]$ make
cd workspace/ && ghdl -a -g --std=08 -fprofile-arcs -fsynopsys -ftest-coverage ../../rtl/full_adder.vhd
cd workspace/ && ghdl -a -g --std=08 -fprofile-arcs -fsynopsys -ftest-coverage ../../sim/full_adder_tb.vhd
cd workspace/ && ghdl -e --std=08 -Wl,-L/usr/lib/gcc/x86_64-pc-linux-gnu/10.2.0 -Wl,-lgcov -Wl,--coverage -fsynopsys full_adder_tb
./workspace/full_adder_tb
../../sim/full_adder_tb.vhd:107:5:@8ns:(assertion note): end of test
simulation stopped @8ns
cd workspace/ && lcov -c -d . -o full_adder_gcov.info
Capturing coverage data from .
Found gcov version: 11.1.0
Using intermediate gcov format
Scanning . for .gcda files ...
Found 2 data files in .
Processing full_adder_tb.gcda
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder_tb.gcno:version 'B02*', prefer 'B11*'
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder_tb.gcda:version 'B02*', prefer version 'B11*'
Processing full_adder.gcda
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder.gcno:version 'B02*', prefer 'B11*'
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder.gcda:version 'B02*', prefer version 'B11*'
Finished .info-file creation
cd workspace/ && genhtml -o ../../resources/html full_adder_gcov.info
Reading data file full_adder_gcov.info
Found 2 entries.
Found common filename prefix "/home/user/work/ghdl_code_coverage"
Writing .css and .png files.
Generating output.
Processing file rtl/full_adder.vhd
Processing file sim/full_adder_tb.vhd
Writing directory view page.
Overall coverage rate:
  lines......: 100.0% (27 of 27 lines)
  functions..: 82.4% (14 of 17 functions)
[user@archlin sim]$
```

I am using Arch-Linux, that uses gcc-11 and ghdl is using gcc-10, that's why you can read the following messages above:
```code
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder_tb.gcno:version 'B02*', prefer 'B11*'
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder_tb.gcda:version 'B02*', prefer version 'B11*'
...
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder.gcno:version 'B02*', prefer 'B11*'
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder.gcda:version 'B02*', prefer version 'B11*'
```

- Although when I use -i & -m something goes wrong and .info file differs from the other invoking way and results generation fails.

```console
[user@archlin sim]$ make run_make
cd workspace/ && ghdl -i --std=08 -g -fsynopsys ../../rtl/full_adder.vhd
cd workspace/ && ghdl -i --std=08 -g -fsynopsys ../../sim/full_adder_tb.vhd
cd workspace/ && ghdl -m --std=08 -Wc,-fprofile-arcs -Wc,-ftest-coverage -Wl,-L/usr/lib/gcc/x86_64-pc-linux-gnu/10.2.0 -Wl,-lgcov -Wl,--coverage -fsynopsys full_adder_tb
analyze ../../rtl/full_adder.vhd
analyze ../../sim/full_adder_tb.vhd
elaborate full_adder_tb
./workspace/full_adder_tb
../../sim/full_adder_tb.vhd:107:5:@8ns:(assertion note): end of test
simulation stopped @8ns
cd workspace/ && lcov -c -d . -o full_adder_gcov.info
Capturing coverage data from .
Found gcov version: 11.1.0
Using intermediate gcov format
Scanning . for .gcda files ...
Found 3 data files in .
Processing full_adder_tb.gcda
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder_tb.gcno:version 'B02*', prefer 'B11*'
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder_tb.gcda:version 'B02*', prefer version 'B11*'
Processing e~full_adder_tb.gcda
/home/user/work/ghdl_code_coverage/sim/workspace/e~full_adder_tb.gcno:version 'B02*', prefer 'B11*'
/home/user/work/ghdl_code_coverage/sim/workspace/e~full_adder_tb.gcda:version 'B02*', prefer version 'B11*'
geninfo: WARNING: could not open /home/user/work/ghdl_code_coverage/sim/workspace/e~full_adder_tb
geninfo: WARNING: some exclusion markers may be ignored
Processing full_adder.gcda
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder.gcno:version 'B02*', prefer 'B11*'
/home/user/work/ghdl_code_coverage/sim/workspace/full_adder.gcda:version 'B02*', prefer version 'B11*'
Finished .info-file creation
cd workspace/ && genhtml -o ../../resources/html full_adder_gcov.info
Reading data file full_adder_gcov.info
Found 3 entries.
Found common filename prefix "/home/user/work/ghdl_code_coverage"
Writing .css and .png files.
Generating output.
Processing file rtl/full_adder.vhd
Processing file sim/full_adder_tb.vhd
Processing file sim/workspace/e~full_adder_tb
genhtml: ERROR: cannot read /home/user/work/ghdl_code_coverage/sim/workspace/e~full_adder_tb
make: *** [makefile:31: run_make] Error 2
[user@archlin sim]$ 
```

Is it possible to do something to fix this and uses -i & -m instead of -a & -e?

tgingold, 2021-07-08:
The issue is only with genhtml.  Does it work better if you remove e~full_adder_tb.gcno and e~full_adder_tb.gcda ?

jocorso, 2021-07-09:
Yes, e~full_adder_tb.gcno and e~full_adder_tb.gcda files are created when I invoke ghdl with -i & -m. So lcov uses them to create full_adder_gcov.info and produces the error.
You are right, if I delete the files it works.
Thanks!

closed 2021-07-09 by jocorso