ghdl/ghdl #1816: unnecessary "range expression must be locally static"

jeras, 2021-07-09:
I have used a generic in a range expression (at an entity instance):
```VHDL
port map (
   drnData_i(DATA_WIDTH_G+1 downto DATA_WIDTH_G) => axiRdResp,
   drnData_i(DATA_WIDTH_G-1 downto 0           ) => dout,
   ...
);
```

And I got errors:

```VHDL
../rtl/EccRam.vhd:206:29:error: range expression must be locally static
      drnData_i(DATA_WIDTH_G+1 downto DATA_WIDTH_G) => axiRdResp,
                            ^
...
```
And genetics are locally static?

umarcor, 2021-07-09:
Hi @jeras! Can you please provide a minimal reproducible example (MWE) which we can actually analyze/elaborate ourselves?

tgingold, 2021-07-09:
Sorry, but ghdl just follows a rule from the LRM.
I think the rule is correct: it allows you to check at analysis time the association.

jeras, 2021-07-14:
**How to reproduce?**

```vhd :file: ent.vhd]
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity ent is
generic (
  W : integer := 8
);
port (
  dat : in std_logic_vector(W-1 downto 0)
);
end entity ent;

architecture rtl of ent is
begin

  process begin
    wait for 1 ns;
    report "dat = " & to_hstring(dat) severity note;
    wait;
  end process;

end rtl;
```

```vhd :file: top.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity top is
generic (
  W : integer := 8
);
end entity top;
  
architecture rtl of top is

  signal dat_h : std_logic_vector(W-1 downto 0);
  signal dat_l : std_logic_vector(W-1 downto 0);

begin
  
  process begin
    report "Hello world from top" severity note;
    wait;
  end process;

  dat_h <= 4X"5";
  dat_l <= 4X"C";

  u_ent: entity work.ent
  port map(
    -- workaround
--    dat => (dat_h & dat_l)
    dat(W  -1 downto W/2) => dat_h,
    dat(W/2-1 downto 0  ) => dat_l
  );

end rtl;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 --work=work ent.vhd
ghdl -a --std=08 --work=work top.vhd
ghdl -e --std=08 --work=work top
ghdl -r --std=08 --work=work top
```

jeras, 2021-07-14:
I tested this on another simulator and got the same ERROR.
So I would agree, this is done correctly in VHDL.

This was my first encounter with a *locally static* kind of restriction.

closed 2021-07-14 by jeras