ghdl/ghdl #1817: Testbench organization generates failure on code coverage

jocorso, 2021-07-10:
**Description**
I am currently evaluating the use of VHDL code coverage with ghdl-gcc and gcov. To carry out this evaluation I have created a [gitlab](https://gitlab.com/jcorrecher/ghdl_code_coverage)  repository. 

**Expected behaviour**
I updated simulation folder with two simulations.
- First simulation declares design under verification first, and process stimulus last.
- Second simulation declares process stimulus first and design under verification last.

Both simulations do the same, but **first works** ok and **second fails**.

**First simulation**

This is the architecture body code from first simulation testbench code.

```vhdl :file: full_adder_tb.vhd
begin
-------------------------------------------------------------------------------
-- * TB0 : DUV + FM
-------------------------------------------------------------------------------
-- ** instance
  DUV : entity work.full_adder
    port map (a => a, b => b, cin => cin, s => s,  cout => cout );
-------------------------------------------------------------------------------
-- * TB1 : Test
-------------------------------------------------------------------------------
  process
  begin
    --  Check each pattern.
    for i in fa_table'range loop
      --  Set the inputs.
      a   <= fa_table(i).a;
      b   <= fa_table(i).b;
      cin <= fa_table(i).cin;
      --  Wait for the results.
      wait for 1 ns;
      --  Check the outputs.
      assert s = fa_table(i).s
        report "bad sum value" severity error;
      assert cout = fa_table(i).cout
        report "bad carry out value" severity error;
    end loop;
    assert false report "end of test" severity note;
    std.env.stop;
    wait;
  end process;
end architecture sim;
```
This is the result provided by terminal.

```console
[user@archlin instance_first.ghdl]$ make
cd workspace/ && ghdl -a -g --std=08 -fprofile-arcs -fsynopsys -ftest-coverage ../../../rtl/full_adder.vhd
cd workspace/ && ghdl -a -g --std=08 -fprofile-arcs -fsynopsys -ftest-coverage ../full_adder_tb.vhd
cd workspace/ && ghdl -e --std=08 -Wl,-L/usr/lib/gcc/x86_64-pc-linux-gnu/10.2.0 -Wl,-lgcov -Wl,--coverage -fsynopsys full_adder_tb
./workspace/full_adder_tb
../full_adder_tb.vhd:107:5:@8ns:(assertion note): end of test
simulation stopped @8ns
cd workspace/ && lcov -c -d . -o full_adder_gcov.info
Capturing coverage data from .
Found gcov version: 11.1.0
Using intermediate gcov format
Scanning . for .gcda files ...
Found 2 data files in .
Processing full_adder_tb.gcda
/home/user/work/ghdl_code_coverage/sim/instance_first.ghdl/workspace/full_adder_tb.gcno:version 'B02*', prefer 'B11*'
/home/user/work/ghdl_code_coverage/sim/instance_first.ghdl/workspace/full_adder_tb.gcda:version 'B02*', prefer version 'B11*'
Processing full_adder.gcda
/home/user/work/ghdl_code_coverage/sim/instance_first.ghdl/workspace/full_adder.gcno:version 'B02*', prefer 'B11*'
/home/user/work/ghdl_code_coverage/sim/instance_first.ghdl/workspace/full_adder.gcda:version 'B02*', prefer version 'B11*'
Finished .info-file creation
cd workspace/ && genhtml -o ../../../resources/html full_adder_gcov.info
Reading data file full_adder_gcov.info
Found 2 entries.
Found common filename prefix "/home/user/work/ghdl_code_coverage"
Writing .css and .png files.
Generating output.
Processing file rtl/full_adder.vhd
Processing file sim/instance_first.ghdl/full_adder_tb.vhd
Writing directory view page.
Overall coverage rate:
  lines......: 100.0% (27 of 27 lines)
  functions..: 82.4% (14 of 17 functions)
[user@archlin instance_first.ghdl]$ 
```

**Second Simulation**

This is the architecture body code from first simulation testbench code.

```vhdl :file: full_adder_tb.vhd
begin
-------------------------------------------------------------------------------
-- * TB0 : Test
-------------------------------------------------------------------------------
  process
  begin
    --  Check each pattern.
    for i in fa_table'range loop
      --  Set the inputs.
      a   <= fa_table(i).a;
      b   <= fa_table(i).b;
      cin <= fa_table(i).cin;
      --  Wait for the results.
      wait for 1 ns;
      --  Check the outputs.
      assert s = fa_table(i).s
        report "bad sum value" severity error;
      assert cout = fa_table(i).cout
        report "bad carry out value" severity error;
    end loop;
    assert false report "end of test" severity note;
    std.env.stop;
    wait;
  end process;
-------------------------------------------------------------------------------
-- * TB1 : DUV + FM
-------------------------------------------------------------------------------
-- ** instance
  DUV : entity work.full_adder
    port map (a => a, b => b, cin => cin, s => s, cout => cout);
end architecture sim;
```
This is the result provided by terminal.

```console
[user@archlin instance_last.ghdl]$ make
cd workspace/ && ghdl -a -g --std=08 -fprofile-arcs -fsynopsys -ftest-coverage ../../../rtl/full_adder.vhd
cd workspace/ && ghdl -a -g --std=08 -fprofile-arcs -fsynopsys -ftest-coverage ../full_adder_tb.vhd
during IPA pass: profile
/home/user/work/ghdl_code_coverage/sim/instance_last.ghdl/workspace/../full_adder_tb.vhd: En la función ‘work__full_adder_tb__ARCH__sim__duv__COMP_ELAB’:
/home/user/work/ghdl_code_coverage/sim/instance_last.ghdl/workspace/../full_adder_tb.vhd:76: error interno del compilador: en coverage_begin_function, en coverage.c:656
   76 |   process
      | 
Por favor, envíe un informe completo de errores,
con el código preprocesado si es apropiado.
Véase <https://bugs.archlinux.org/> para instrucciones.
ghdl:error: exec error
make: *** [makefile:17: run_elab] Error 3
[user@archlin instance_last.ghdl]$ 
```
**Context**

- OS: Arch Linux
- GHDL: 2.0.0-dev (tarball)
- Gcov: 10.2.0
- LCOV: 1.15-2-gb9d6727

tgingold, 2021-07-17:
I will have a look at it.

tgingold, 2021-07-17:
But I was not able to reproduce the crash.  Do you know which gcc backend is used in ghdl ?

jocorso, 2021-07-17:
My [gitlab](https://gitlab.com/jcorrecher/ghdl_code_coverage) repository deploys a [ghdl fedora docker](https://hub.docker.com/layers/ghdl/ghdl/fedora34-gcc-9.2.0/images/sha256-faf2f164f53b5d883f9f6418e466c8910f83e442a2ba3d1717b0a671853cf26a?context=explore) with ghdl.

The version installed in this image is:

```console
sh-5.1# ghdl --version
GHDL 2.0.0-dev (v1.0.0-479-g1da694fe) [Dunoon edition]
 Compiled with GNAT Version: 11.1.1 20210531 (Red Hat 11.1.1-
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

umarcor, 2021-07-17:
In case it can be useful, that container is built in ghdl/docker (https://github.com/ghdl/docker/runs/3082162351?check_suite_focus=true) using the [scripts/ci-run.sh](https://github.com/ghdl/ghdl/blob/master/scripts/ci-run.sh) from this repo. So, it's equivalent to the Ubuntu jobs we have in the CI of this repo, but using Fedora instead.

@jocorso, would you mind trying image `ghdl/ghdl:buster-gcc-8.3.0`? Just for checking whether it is specific to Fedora or to GCC 9.

jocorso, 2021-07-18:
@umarcor I created a new [branch](https://gitlab.com/jcorrecher/ghdl_code_coverage/-/tree/RunBuster) with `ghdl/ghdl:buster-gcc-8.3.0` and it **works**.
I have installed in my laptop Arch-Linux and I have the same problem as fedora version.

umarcor, 2021-07-18:
Gracias!

Did you use Filipe's package of Arch (https://archlinux.org/packages/community/x86_64/ghdl-gcc/) or some of the packages from AUR/marzoul (https://aur.archlinux.org/packages/ghdl-gcc-git/)?

jocorso, 2021-07-18:
I am using [nightly version](https://gitlab.com/aur-packages/ghdl-nighty/-/jobs/1428997311/artifacts/browse).

umarcor, 2021-07-18:
Interesting. I was not aware of that variant.

Summary so far:

- GCC 8.3.0 on Debian Buster works.
- GCC 9.2.0 on Fedora 34  fails.
- GCC 10.2.0 on Arch Linux fails.

tgingold, 2021-07-19:
I think this is now fixed.

closed 2021-07-19 by tgingold

csantosb, 2021-07-19:
> Interesting. I was not aware of that variant.

Author here. Just for the record, this is nothing but Filipe's package build script, using nightly ghdl. I comment out ghdl-mcode and python-pyghdl, as I'm not using it. Then I build a new package every week.

jocorso, 2021-08-13:
> I think this is now fixed.

@tgingold I updated ghdl version at the docker used for [gitlab code coverage](https://gitlab.com/jcorrecher/ghdl_code_coverage) design and now it works.

umarcor, 2021-08-15:
@jocorso, I saw in https://gitlab.com/jcorrecher/ghdl_code_coverage/-/commit/2837a54b561b8e74a189223eb4e90cb30cb98724 that you are now using your own docker image. Is that because of some problem with the ones from ghdl or due to adding extra features/tools?

commit 4be34d78 (2021-07-19): trans-chap9: set line number for gcc.  Fix #1817