ghdl/ghdl #1819: Synth: Can integer literal expansion use the minimum width instead of 32 bits?

tttpgi, 2021-07-14:
Hallo,

I have started using GHDL synthesis as a means of checking that the code synthesizes correctly. I have found one particular feature of GHDL that I like that is not often seen in other tools: the reporting of truncated integers as part of the output, which makes it fairly easy to review this against code to see if the truncation is intended or not.
  The problem I encountered here is that any integer literals used inside the VHDL code, e.g. in a statement like (r+1), are converted first to a 32-bit signed vector. This leads eventually to a truncation to the correct width with a report that is not useful. 

So I have a quetion: is it possible/smart to convert the integer literals to an internal type that only uses as many bits as needed (log2 of the value)?  If yes, I would like to try my hand at implementing this.

Thanks,
   Petr Grillinger

umarcor, 2021-07-14:
@tttpgi, what is the type of `r` in your `(r+1)`?

tttpgi, 2021-07-14:
> @tttpgi, what is the type of `r` in your `(r+1)`?

A signal or a variable (when in a process) that is a subtype of natural. Typically something like this:
  subtype TYPE_PORT_NUMBER is natural range 0 to 2;
  signal r : TYPE_PORT_NUMBER;
  process ....
     variable v: TYPE_PORT_NUMBER;
  begin
     ....
      r <= v;  
     ....
  end process;

Sometimes also a variable

tgingold, 2021-07-17:
Do you have a reproducer ?

This is not a so simple issue, as the addition is really a 32 bit addition.