ghdl/ghdl #1820: GHDL bug ocurred (TYPES.INTERNAL_ERROR raised)

ManuelBilbao, 2021-07-21:
**Description**
When attempting to analyze some files ghdl throws TYPES.INTERNAL.ERROR.

**Expected behaviour**
If I run `ghdl -a etapa.vhd cordic.vhd desenrollado.vhd` it runs fine, but when I run `ghdl -a *.vhd` it crashes.

**How to reproduce?**

```vhd :file: etapa.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity etapa is
	generic (
		SIZE: natural := 16;
		STEP: natural := 1
	);
	port (
		vec: in std_logic;
		x_i: in std_logic_vector(SIZE-1 downto 0);
		y_i: in std_logic_vector(SIZE-1 downto 0);
		z_i: in std_logic_vector(SIZE-1 downto 0);
		atan_i: in std_logic_vector(SIZE-1 downto 0);
		x_o: out std_logic_vector(SIZE-1 downto 0);
		y_o: out std_logic_vector(SIZE-1 downto 0);
		z_o: out std_logic_vector(SIZE-1 downto 0)
	) ;
end entity ; -- etapa

architecture etapa_arq of etapa is

	signal d_i: std_logic;
	signal d_i_slv: std_logic_vector(SIZE-1 downto 0);
	signal d_i_n_slv: std_logic_vector(SIZE-1 downto 0);

	signal x_shifted: std_logic_vector(SIZE-1 downto 0);
	signal y_shifted: std_logic_vector(SIZE-1 downto 0);
	signal c_x_shifted: std_logic_vector(SIZE-1 downto 0);
	signal c_y_shifted: std_logic_vector(SIZE-1 downto 0);
	signal c_z_i: std_logic_vector(SIZE-1 downto 0);

begin

	d_i <= y_i(SIZE-1) when vec = '1' else z_i(SIZE-1);
	d_i_slv <= (SIZE-2 downto 0 => '0') & d_i;
	d_i_n_slv <= (SIZE-2 downto 0 => '0') & not(d_i);

	x_shifted <= std_logic_vector(shift_right(signed(x_i), STEP));
	y_shifted <= std_logic_vector(shift_right(signed(y_i), STEP));

	c_x_shifted <= x_shifted xor (0 to SIZE-1 => d_i);
	c_y_shifted <= y_shifted xor (0 to SIZE-1 => not(d_i));
	c_z_i <= atan_i xor (0 to SIZE-1 => not(d_i));

	x_o <= std_logic_vector(signed(x_i) + signed(c_y_shifted) + signed(d_i_n_slv));
	y_o <= std_logic_vector(signed(y_i) + signed(c_x_shifted) + signed(d_i_slv));
	z_o <= std_logic_vector(signed(z_i) + signed(c_z_i) + signed(d_i_n_slv));

end architecture ; -- etapa_arq
```
```vhd :file: cordic.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use IEEE.math_real.all;

entity cordic is
	generic (
		SIZE: natural := 16
	);
	port (
		vec: in std_logic;
		x_i: in std_logic_vector(SIZE-1 downto 0);
		y_i: in std_logic_vector(SIZE-1 downto 0);
		z_i: in std_logic_vector(SIZE-1 downto 0);
		x_o: out std_logic_vector(SIZE-1 downto 0);
		y_o: out std_logic_vector(SIZE-1 downto 0);
		z_o: out std_logic_vector(SIZE-1 downto 0)
	) ;
end entity ; -- cordic

architecture desenrollado of cordic is

	type matriz is array(natural range <>) of std_logic_vector(SIZE-1 downto 0);
	signal x, y, z: matriz(0 to SIZE);
	signal a: std_logic := '1';

	--type matriz_atan is array (0 to SIZE-1) of std_logic_vector(SIZE-1 downto 0);
	--constant atan: matriz_atan := (
	--	for i in 0 to SIZE-1 loop
	--		std_logic_vector(),
	--	end loop ; 
	--)
	signal PRUEBA: std_logic_vector(SIZE-1 downto 0) := std_logic_vector(to_unsigned(integer(0.25*2**SIZE+0.5), SIZE));
	constant HOLA: real := 0.5;

begin

	x(0) <= x_i;
	y(0) <= y_i;
	z(0) <= z_i;

	x_o <= x(SIZE);
	y_o <= y(SIZE);
	z_o <= z(SIZE);

	etapas : for i in 0 to SIZE-1 generate
		etapa: entity work.etapa
			generic map(
				SIZE => SIZE,
				STEP => i
			)
			port map(
				vec => vec,
				x_i => x(i),
				y_i => y(i),
				z_i => z(i),
				atan_i => (others => '0'),
				x_o => x(i+1),
				y_o => y(i+1),
				z_o => z(i+1)
			);
	end generate ; -- etapas

end architecture ; -- desenrollado
```
```vhd :file: desenrollado.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use IEEE.math_real.all;

entity cordic is
	generic (
		SIZE: natural := 16
	);
	port (
		vec: in std_logic;
		x_i: in std_logic_vector(SIZE-1 downto 0);
		y_i: in std_logic_vector(SIZE-1 downto 0);
		z_i: in std_logic_vector(SIZE-1 downto 0);
		x_o: out std_logic_vector(SIZE-1 downto 0);
		y_o: out std_logic_vector(SIZE-1 downto 0);
		z_o: out std_logic_vector(SIZE-1 downto 0)
	) ;
end entity ; -- cordic

architecture desenrollado of cordic is

	type matriz is array(natural range <>) of std_logic_vector(SIZE-1 downto 0);
	signal x, y, z: matriz(0 to SIZE);
	signal a: std_logic := '1';

	--type matriz_atan is array (0 to SIZE-1) of std_logic_vector(SIZE-1 downto 0);
	--constant atan: matriz_atan := (
	--	for i in 0 to SIZE-1 loop
	--		std_logic_vector(),
	--	end loop ; 
	--)
	signal PRUEBA: std_logic_vector(SIZE-1 downto 0) := std_logic_vector(to_unsigned(integer(0.25*2**SIZE+0.5), SIZE));
	constant HOLA: real := 0.5;

begin

	x(0) <= x_i;
	y(0) <= y_i;
	z(0) <= z_i;

	x_o <= x(SIZE);
	y_o <= y(SIZE);
	z_o <= z(SIZE);

	etapas : for i in 0 to SIZE-1 generate
		etapa: entity work.etapa
			generic map(
				SIZE => SIZE,
				STEP => i
			)
			port map(
				vec => vec,
				x_i => x(i),
				y_i => y(i),
				z_i => z(i),
				atan_i => (others => '0'),
				x_o => x(i+1),
				y_o => y(i+1),
				z_o => z(i+1)
			);
	end generate ; -- etapas

end architecture ; -- desenrollado
```

**Context**
- OS: Ubuntu 18.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `2fb2384dbcfea0b2b6e6efe00934ccdf869d6ee6`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
save_library: cannot handle IIR_KIND_INTEGER_SUBTYPE_DEFINITION (etapa.vhd:29:47)
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (tarball) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/home/manu/Documentos/Facultad/SisDig/TPs/3/src/
Command line:
ghdl -a cordic.vhd desenrollado_tb.vhd etapa.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x5557d190baf9 0x5557d199fd16 0x5557d19a1431 0x5557d1a745d6 0x5557d19a9f91 0x5557d1b373c1 0x5557d175b711 0x7f1af482fbf5 0x5557d1759fd8 0xfffffffffffffffe
******************************************************************
```

**Additional context**
There's no other .vhd files at the directory:
```
$ ls
cordic.vhd  desenrollado_tb.vhd  etapa.vhd  tb.vcd  work-obj93.cf
```

umarcor, 2021-07-21:
Kaixo @ManuelBilbao! The 2nd and 3rd code blocks are exactly the same. Is that an oversight?

ManuelBilbao, 2021-07-21:
> Kaixo @ManuelBilbao! The 2nd and 3rd code blocks are exactly the same. Is that an oversight?

Kaixo (de un descendiente en Argentina)! My mistake, edited.

umarcor, 2021-07-21:
I tested on MINGW64. I cannot reproduce the internal error:

```sh
# ghdl -a *.vhd
cordic.vhd:47:36: unit "etapa" not found in library "work"
                etapa: entity work.etapa
                                   ^
C:\msys64\mingw64\bin\ghdl.exe: compilation error

# ghdl -a cordic.vhd desenrollado.vhd etapa.vhd
cordic.vhd:47:36: unit "etapa" not found in library "work"
                etapa: entity work.etapa
                                   ^
C:\msys64\mingw64\bin\ghdl.exe: compilation error

# ghdl -a etapa.vhd cordic.vhd desenrollado.vhd
```

I guess that is correct. When the wildcard is used, bash expands the list in alphabetical order. Hence, `etapa.vhd` is analysed the last, although it should be the first.

In order to use wildcards with GHDL, you want to use `-i` and `-m`, instead of `-a` and `-e`.

```sh
# ghdl -i *.vhd

# ghdl -m desenrollado_tb
analyze etapa.vhd
analyze cordic.vhd
analyze desenrollado.vhd
elaborate desenrollado_tb
```

The difference between `-i` and `-a` is that "import" will not analyse the sources at the moment. They are parsed only. Analysis and elaboration is done when you run "make". See https://ghdl.github.io/ghdl/using/InvokingGHDL.html#design-rebuilding-commands for further details.

umarcor, 2021-07-21:
While testing it on a container, I could trigger the bug:

```
# docker run --rm -itv /$(pwd)://wrk -w //wrk ghdl/ghdl:ubuntu18-mcode bash

root@b012805ca35d:/wrk# ghdl -a *.vhd
cordic.vhd:47:36:error: unit "etapa" not found in library "work"
                etapa: entity work.etapa
                                   ^

root@b012805ca35d:/wrk# ghdl -a etapa.vhd cordic.vhd desenrollado.vhd 

root@b012805ca35d:/wrk# ghdl -a *.vhd
etapa.vhd:1:1:warning: redefinition of a library unit in same design file: [-Wlibrary]
library IEEE;
^
etapa.vhd:1:1:warning: entity "etapa" defined at etapa.vhd:5:8 is now entity "etapa" [-Wlibrary]
library IEEE;
^
save_library: cannot handle IIR_KIND_INTEGER_SUBTYPE_DEFINITION (etapa.vhd:29:48)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-495-gcfa7ec61) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/wrk/
Command line:
ghdl -a cordic.vhd desenrollado.vhd etapa.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x55dfffbe39bf 0x55dfffc6d408 0x55dfffc6eb23 0x55dfffd45f7e 0x55dfffc77c6f 0x55dfffe09d8f 0x55dfffa1528f 0x7fbf3e05bbf5 0x55dfffa13af8 0xfffffffffffffffe
******************************************************************
```

tgingold, 2021-07-22:
I am testing a fix.
Thank you very much for the reproducer.  I think I saw this crash once or two, but without being able to reproduce it at that time.

commit 5cf12a35 (2021-07-22): vhdl-utils(free_recursive): only free the name (not the entity)

Fix #1820
closed 2021-07-22 by tgingold