ghdl/ghdl #1821: Wrong file list in compile-osvvm.sh

csantosb, 2021-07-21:
**Description**

File array in scripts/vendors/compile-osvvm.sh contains a wrong ordered fileset. Upon launching of this script, it fails.

**Expected behaviour**

Should be this one instead, for example

    NamePkg.vhd
    OsvvmGlobalPkg.vhd
    VendorCovApiPkg.vhd
    TranscriptPkg.vhd
    TextUtilPkg.vhd
    AlertLogPkg.vhd
    MessagePkg.vhd
    SortListPkg_int.vhd
    RandomBasePkg.vhd
    RandomPkg.vhd
    CoveragePkg.vhd
    MemoryPkg.vhd
    ScoreboardGenericPkg.vhd
    ScoreboardPkg_slv.vhd
    ScoreboardPkg_int.vhd
    ResolutionPkg.vhd
    TbUtilPkg.vhd
    OsvvmContext.vhd
    NameStorePkg.vhd
    RandomProcedurePkg.vhd
    ResizePkg.vhd

**How to reproduce?**

Run script with two versions.

**Context**

Please, provide the following information:

- OS: ArchLinux
- Origin:
  https://gitlab.com/aur-packages/ghdl-nighty/-/pipelines

umarcor, 2021-07-21:
@csantosb you need to update to OSVVM 2021.06. That changed between 2021.02 and 2021.06. A few days ago, I changed it so that "latest" OSVVM works: https://github.com/ghdl/ghdl/commit/1409d4ee66fc336d535486cb08bf60070bc9e197.

It might be possible to detect the OSVVM version and use a different filelist. However, the way forward is to actually parse the `*.pro` files. That is currently done in `compile-osvvm.ps1`, but not in the bash variant. The reason is the limitations of bash for dealing with nested and cross-referenced arrays. Due to how `*.pro` files call/import each other, it is not a simple two-layer logic as in UVVM's equivalents.

I discussed this with @Paebbels a few days ago and I suggested that maybe it's time to deprecate the powershell and bash variants and provide a python solution instead. Compared to five years ago, we can assume the dependency of Python to be acceptable. OTOH, converting the current powershell scripts to python should not be difficult, but it needs to be done and none of us can take that task at the moment.

Paebbels, 2021-07-21:
@tgingold some years ago when I started the vendor compile scripts, there was only Bash for macOS/Linux and *.bat for Windows. I converted these old compile instructions to PowerShell. In parallel I developed the vendor compile scripts in Bash and PowerShell as Python what not around in GHDL.

Now we have Python scripting in GHDL as well as a Python API. It would simplify maintaining these scripts if only done in one language. Also some challenges like described in this issue, are to hard to solve in Bash (e.g. recursively reading compile order files).

tgingold, 2021-07-21:
Yes, I agree.

umarcor, 2021-07-22:
FTR, a quick and dirty `compile-osvvm.py`: https://github.com/umarcor/ghdl-extended-tests/blob/py-scripts/verification/OSVVM/compile-osvvm.py. That is the recursive `*.pro` file parsing only. The arguments passed to analysis are wrong, because it puts everything in the same single directory: `'-P={}'.format(str(DESTINATION)), '--workdir={}'.format(str(DESTINATION))`. Nevertheless, CI is successful, meaning the files are all analyzed succesfully.