ghdl/ghdl #1826: Assertion failure when parsing multiple identifiers

umarcor, 2021-07-30:
This is a bug we found while testing the DOM with NEORV32. Precisely, with file https://github.com/stnolting/neorv32/blob/master/rtl/templates/system/neorv32_SystemTop_axi4lite.vhd. By changing https://github.com/stnolting/neorv32/blob/9e2d0f245de1439ec17d3852262e7d4626b57044/rtl/templates/system/neorv32_SystemTop_axi4lite.vhd#L249-L259 to:

```vhdl
  signal ack_read, ack_write : std_ulogic; -- normal transfer termination
  signal err_read, err_write : std_ulogic; -- error transfer termination

  -- AXI bridge control --
  type ctrl_t is record
    radr_received : std_ulogic;
    wadr_received : std_ulogic;
    wdat_received : std_ulogic;
  end record;
  signal ctrl : ctrl_t;
```

That is, moving the signal declarations with multiple identifiers above the record type declaration, we get:

````sh
# ghdl-dom pretty -f /t/neorv32/rtl/templates/system/neorv32_SystemTop_axi4lite.vhd 
================================================================================
                         pyGHDL.dom - Test Application
================================================================================
Parsing file 'T:/neorv32/rtl/templates/system/neorv32_SystemTop_axi4lite.vhd'

raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Subtype_Indication
````

Paebbels, 2021-07-30:
I had a similar problem in the Python code when dealing with signal declaration that have multiple identifiers. In my case, the scan for more identifiers sometimes never ended. Here the subtype field is not filled or it doesn't reference the full filled IRR node depending on how lines are ordered.

tgingold, 2021-07-30:
Well, I am not sure what I can do.  Does it reproduce on master or only in your PR ?

I suppose python code is trying to call Get_Subtype_Indication on a non-object declaration (like the record type declaration).

Paebbels, 2021-07-30:
Maybe my comment was more confusing then helping ...

I meant, I had a similar algorithmic issue in the Python code and fixed it. It was related to the call of `Get_Subtype_Indication` in case multiple identifiers were used.

In the example of @umarcor, libGHDL itself throws an assertion error. What we did is just reorder signal declarations.

Before (OK):
```vhdl
  -- AXI bridge control --
  type ctrl_t is record
    radr_received : std_ulogic;
    wadr_received : std_ulogic;
    wdat_received : std_ulogic;
  end record;
  signal ctrl : ctrl_t;

  signal ack_read, ack_write : std_ulogic; -- normal transfer termination
  signal err_read, err_write : std_ulogic; -- error transfer termination
```

After (assertion error):
```vhdl
  signal ack_read, ack_write : std_ulogic; -- normal transfer termination
  signal err_read, err_write : std_ulogic; -- error transfer termination

  -- AXI bridge control --
  type ctrl_t is record
    radr_received : std_ulogic;
    wadr_received : std_ulogic;
    wdat_received : std_ulogic;
  end record;
  signal ctrl : ctrl_t;

--  signal ack_read, ack_write : std_ulogic; -- normal transfer termination
--  signal err_read, err_write : std_ulogic; -- error transfer termination
```

So in this example, the assertion error is raised, when a signal declaration **with multiple identifiers** is followed by a type definition.

So under some circumstances, libGHDL does not always find or fill the subtype indication field based on the order.

umarcor, 2021-07-30:
I think that Tristan is correct (this is produced by how Python is using libghdl, not by libghdl itself...). A MWE:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity ent is
end entity;

architecture arch of ent is

  signal ack_read, ack_write : std_ulogic;

  type ctrl_t is record
    radr_received : std_ulogic;
    wadr_received : std_ulogic;
    wdat_received : std_ulogic;
  end record;

begin
end architecture;
```

```sh
# ghdl -a ent.vhd

# ghdl-dom pretty -f ent.vhd 
================================================================================
                         pyGHDL.dom - Test Application
================================================================================
Parsing file 'ent.vhd'

raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Subtype_Indication
```

umarcor, 2021-08-02:
I can confirm that this is fixed with the latest updates to #1822, so I'm closing.

closed 2021-08-02 by umarcor