ghdl/ghdl #1828: bug report - dinamic array of std_logic_vector

rochadiego, 2021-07-31:
**Description**
I'm trying to create an dinamic array of std_logic_vector by parsing the size of array as generic integer parameter

**Expected behaviour**
The code do not run in ghdl, i've tested same example in Ada Playgroud and I had no problem

message: 
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (tarball) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/diego/Downloads/network/
Command line:
ghdl -e --std=08 --time-resolution=ns network_tb
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : ortho_code-types.adb:113
Call stack traceback locations:
0x7f8fc2380d12 0x55df1ef3dcbb 0x55df1ef5137d 0x55df1ef5153d 0x55df1ef68b57 0x55df1f24202c 0x55df1f22cdfc 0x55df1f22a0f7 0x55df1f22fc16 0x55df1f1ea598 0x55df1f1ebd09 0x55df1f1ec748 0x55df1f1e156e 0x55df1f1e1a6e 0x55df1f1e1843 0x55df1f1e1aed 0x55df1f24f940 0x55df1f1b7a51 0x55df1f1b4ba4 0x55df1f1c2ca8 0x55df1f2542af 0x55df1f195cac 0x55df1f0cbf8b 0x55df1f257bcc 0x55df1ee82a67 0x7f8fc1f300b1 0x55df1ee811cc 0xfffffffffffffffe
******************************************************************

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: layer.vhd
entity layer is
  generic (
    n_inputs  : integer;
    n_neurons : integer
    -- activation_type : string
  );
  port (
    layer_IN   : in coluna(0 to n_inputs - 1)(n_bits - 1 downto 0);
    layer_W_IN : in coluna(0 to n_inputs * n_neurons - 1)(n_bits - 1 downto 0);
    layer_OUT  : out coluna(0 to n_neurons - 1)(n_bits - 1 downto 0)
  );
end layer;

vhd :file: network.vhd

entity network is
  port (
    INPUTS  : in coluna(0 to n_entradas - 1)(n_bits - 1 downto 0);
    OUTPUTs : out coluna(0 to n_neuros - 1)(n_bits - 1 downto 0));
end network;

architecture rtl of network is

  component layer is
    generic (
      n_inputs        : integer;
      n_neurons       : integer
      -- activation_type : string
    );
    port (
      layer_IN   : in coluna(0 to n_entradas - 1)(n_bits - 1 downto 0);
      layer_W_IN : in coluna(0 to n_entradas * n_neuros - 1)(n_bits - 1 downto 0);
      layer_OUT  : out coluna(0 to n_neuros - 1)(n_bits - 1 downto 0)
    );
  end component;

-- signals and constants declarations...

begin

  unica : layer
  generic map(
    n_inputs        => n_inputs,
    n_neurons => n_neurons
  )
  port map(
    layer_IN   => INPUTS,
    layer_W_IN => mem,
    layer_OUT  => OUTPUTs);

end architecture;

```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

umarcor, 2021-07-31:
Hi @rochadiego! The MWE seems incomplete. The `vhd :file: network.vhd` needs to be a comment (add `--`). Other than that, I get the followin:

```sh
# ghdl -a ent.vhd 
ent.vhd:8:21: no declaration for "coluna"
    layer_IN   : in coluna(0 to n_inputs - 1)(n_bits - 1 downto 0);
                    ^
ent.vhd:9:21: no declaration for "coluna"
    layer_W_IN : in coluna(0 to n_inputs * n_neurons - 1)(n_bits - 1 downto 0);
                    ^
ent.vhd:10:22: no declaration for "coluna"
    layer_OUT  : out coluna(0 to n_neurons - 1)(n_bits - 1 downto 0)
                     ^
ent.vhd:16:18: no declaration for "coluna"
    INPUTS  : in coluna(0 to n_entradas - 1)(n_bits - 1 downto 0);
                 ^
ent.vhd:17:19: no declaration for "coluna"
    OUTPUTs : out coluna(0 to n_neuros - 1)(n_bits - 1 downto 0));
                  ^
ent.vhd:20:21: entity 'network' was not analysed
architecture rtl of network is
                    ^
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

rochadiego, 2021-07-31:
I was looking at ghdl issues and I found this issue https://github.com/ghdl/ghdl/issues/695, I think it is similar to my problem:

someone anwered: "Generic map aspect in interface package declaration are not yet supported. [...] In addition, interface package declarations are also not yet supported in entities..."

is this still valid today?

umarcor, 2021-08-01:
Yes, I think so.
Nonetheless, it would be useful to have a MWE in this issue, so that it can be tested when time comes.

tgingold, 2021-09-14:
I close this issue due to the lack of a reproducer.  If this is still not fixed, please reopen with a reproducer.

closed 2021-09-14 by tgingold