ghdl/ghdl #1829: OSVVM failing on Linux with GCC backend, but working with LLVM or mcode (either Linux or Windows)

umarcor, 2021-08-01:
Some days ago, we added CI to OSVVM/OsvvmLibraries. The latest pushes are tested with GHDL using [ghdl/setup-ghdl-ci](https://github.com/ghdl/setup-ghdl-ci) (see https://github.com/umarcor/OsvvmLibraries/blob/dev/.github/workflows/Test.yml). We found that all jobs are successfull except the GCC backend (which is only available on Linux). For now, it is disabled, but I created a branch for showing the issue:

- https://github.com/umarcor/OsvvmLibraries/commit/acf241be0099d97d1b6f164c5ca3a9258e95d948
- https://github.com/umarcor/OsvvmLibraries/actions/runs/1088038225
  - https://github.com/umarcor/OsvvmLibraries/runs/3215253994?check_suite_focus=true#step:5:48
  
```
library OSVVM_Common
creating library directory /home/runner/work/OsvvmLibraries/OsvvmLibraries/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_common/v08
analyze ./src/StreamTransactionPkg.vhd
/home/runner/work/OsvvmLibraries/OsvvmLibraries/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_common/v08/StreamTransactionPkg.s: Assembler messages:
/home/runner/work/OsvvmLibraries/OsvvmLibraries/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_common/v08/StreamTransactionPkg.s:1702: Error: symbol `osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTISTR' is already defined
ghdl: compilation error
child process exited abnormally
```

tgingold, 2022-03-10:
I will have a look.

tgingold, 2022-03-10:
No real progress, but I think I understand the reason of the issue: it is due to missing RTI generation for package instances.

The-MEO, 2022-06-15:
I use the gdh/vunit:ggc Docker-image and get the same error since the nightly build.
Yesterday everything worked fine with `ghdl/vunit@sha256:e0bb33ee863ed97d49b91f6248e1e55041f690323296c35713f336d45c6417ce`  
Today the new image `ghdl/vunit:gcc with digest ghdl/vunit@sha256:8e1de60940ececa7aa682dde8dfe372fc4be0a956e25ca80bc40e2c826589c3c` fails with 
```
/builds/nvram/axi-nvram/vunit_out/ghdl/libraries/OSVVM_COMMON/StreamTransactionPkg.s: Assembler messages:
/builds/nvram/axi-nvram/vunit_out/ghdl/libraries/OSVVM_COMMON/StreamTransactionPkg.s:1695: Error: symbol `osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTISTR' is already defined
/usr/local/bin/ghdl: compilation error
```
Perhaps a comparison between the builds will help.

The-MEO, 2022-06-17:
I was wrong. It's not the containers.
My CI had some cached builds from llvm and used them in the gcc container.
Interestingly, with build through llvm and subsequent execution with gcc it allows the coverage to be recorded.

ncervino, 2023-03-22:
I'm seeing this same problem outside of docker and CI builds, so no cache involved.  I was wondering if anyone has other ideas on how to resolve this issue.

GHDL Version: 
```
GHDL 4.0.0-dev (3.0.0.r17.gea460cb31) [Dunoon edition]
 Compiled with GNAT Version: 10.2.1 20210110
 GCC 10.2.0 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Error:
```
=== Command output: ===
/repos/roas-fpga/tools/vunit_out/ghdl/libraries/osvvm_common/StreamTransactionPkg.s: Assembler messages:
/repos/roas-fpga/tools/vunit_out/ghdl/libraries/osvvm_common/StreamTransactionPkg.s:1853: Error: symbol `osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTISTR' is already defined
/usr/local/bin/ghdl: compilation error
```

sweeneyal, 2023-07-15:
Can confirm this is still happening. GHDL version:
```
GHDL 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 GCC 9.3.0 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

andreasorio, 2023-09-08:
same problem with

GHDL 4.0.0-dev (3.0.0.r478.ge747890f9.dirty) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 GCC 12.3.0 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

timothystotts, 2023-12-31:
Can confirm this happening on Ubuntu 22.04.3 LTS with:
```
...
ghdl -a --std=08 --work=osvvm -O2 ../OSVVM/OsvvmLibraries/osvvm/ScoreboardGenericPkg.vhd
ghdl -a --std=08 --work=osvvm -O2 ../OSVVM/OsvvmLibraries/osvvm/ScoreboardPkg_slv.vhd
ghdl -a --std=08 --work=osvvm -O2 ../OSVVM/OsvvmLibraries/osvvm/ScoreboardPkg_int.vhd
ghdl -a --std=08 --work=osvvm -O2 ../OSVVM/OsvvmLibraries/osvvm/ResizePkg.vhd
ghdl -a --std=08 --work=osvvm -O2 ../OSVVM/OsvvmLibraries/osvvm/TbUtilPkg.vhd
ghdl -a --std=08 --work=osvvm -O2 ../OSVVM/OsvvmLibraries/osvvm/ReportPkg.vhd
ghdl -a --std=08 --work=osvvm -O2 ../OSVVM/OsvvmLibraries/osvvm/OsvvmContext.vhd
ghdl -a --std=08 --work=OSVVM_Common -O2 ../OSVVM/OsvvmLibraries/Common/src/StreamTransactionPkg.vhd
StreamTransactionPkg.s: Assembler messages:
StreamTransactionPkg.s:7867: Error: symbol `osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTISTR' is already defined
ghdl:error: compilation error
```

```
$ ghdl --version
GHDL 3.0.0 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 GCC 11.4.0 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2024-01-01:
Yes, but the ghdl you are using (3.0.0) is quite old.  Can you try with the head ?  I am pretty sure this has been fixed.

rramunni, 2024-02-01:
Still getting this error: 
```
GHDL 4.0.0-dev (3.0.0.r823.g9e2b6fc95) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 GCC 12.1.0 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
analyze StreamTransactionPkg.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=OSVVM_Common --workdir=/home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev/osvvm_common/v08 -P/home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev ../OsvvmLibraries/Common/src/StreamTransactionPkg.vhd
Error: /home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev/osvvm_common/v08/StreamTransactionPkg.s: Assembler messages:
/home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev/osvvm_common/v08/StreamTransactionPkg.s:2097: Error: symbol `osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTISTR' is already defined
Error: ghdl: compilation error
AnalyzeError: See messages above in "analyze StreamTransactionPkg.vhd"
For tcl errorInfo, puts $::osvvm::AnalyzeErrorInfo
analyze StreamTransactionArrayPkg.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=OSVVM_Common --workdir=/home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev/osvvm_common/v08 -P/home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev ../OsvvmLibraries/Common/src/StreamTransactionArrayPkg.vhd

analyze AddressBusTransactionPkg.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=OSVVM_Common --workdir=/home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev/osvvm_common/v08 -P/home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev ../OsvvmLibraries/Common/src/AddressBusTransactionPkg.vhd
Error: /home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev/osvvm_common/v08/AddressBusTransactionPkg.s: Assembler messages:
/home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev/osvvm_common/v08/AddressBusTransactionPkg.s:2349: Error: symbol `osvvm_common__addressbustransactionpkg__addressbusrectype__writeburstfifo__RTISTR' is already defined
Error: ghdl: compilation error
AnalyzeError: See messages above in "analyze AddressBusTransactionPkg.vhd"
For tcl errorInfo, puts $::osvvm::AnalyzeErrorInfo
analyze AddressBusTransactionArrayPkg.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=OSVVM_Common --workdir=/home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev/osvvm_common/v08 -P/home/rramunni/osvvm_sim/VHDL_LIBS/GHDL-4.0.0-dev ../OsvvmLibraries/Common/src/AddressBusTransactionArrayPkg.vhd
```

Here are the multiple definitions in the StreamTransactionPkg.s file starting on line 2059:
```
osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTISTR:
	.byte	115
	.byte	99
	.byte	111
	.byte	114
	.byte	101
	.byte	98
	.byte	111
	.byte	97
	.byte	114
	.byte	100
	.byte	105
	.byte	100
	.byte	116
	.byte	121
	.byte	112
	.byte	101
	.byte	0
	.globl	osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTI
	.section	.data.rel.ro
	.align 32
	.type	osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTI, @object
	.size	osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTI, 40
osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTI:
	.byte	34
	.byte	0
	.byte	0
	.byte	0
	.zero	4
	.quad	osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTISTR
	.long	1
	.zero	4
	.quad	osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTIARRAY
	.quad	osvvm__scoreboardpkg_slv__scoreboardidtype__STL
	.section	.rodata
	.align 8
	.type	osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTISTR, @object
	.size	osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTISTR, 10
osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTISTR:
	.byte	98
	.byte	117
	.byte	114
	.byte	115
	.byte	116
	.byte	102
	.byte	105
	.byte	102
	.byte	111
	.byte	0
	.globl	osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTIEL
	.section	.data.rel.ro.local
	.align 32
	.type	osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTIEL, @object
	.size	osvvm_common__streamtransactionpkg__streamrectype__burstfifo__RTIEL, 40
```

Is there a solution yet? Do I just need to run ghdl with the llvm/mcode backend?

nhw76, 2024-12-19:
Experiencing the same problem with GHDL 4.1.0.