ghdl/ghdl #1830: MCode terminates with error due to (A_int -1) when A_int is not initialized.

JimLewis, 2021-08-04:
**Description**
During initialization, the signal (in UartRx.vhd) is unitialized, and, hence has the value integer'left.
  signal NumDataBits : integer ;
. . .
  LastDataBit <= '1' when DataBitCount = (NumDataBits - 1) else '0' ;

This is not an issue as during time 0, simulation cycle 1, NumDataBits is initialized to an appropriate value.

**Expected behaviour**
???  Not getting failures out of other simulators.

OTOH, I updated the code as it can indeed be initialized in the declaration.

**How to reproduce?**
It is in the 2021.06 release of OSVVM.

On the Dev branch, I initialized the signal to a value that is not integer'left.  
It will be in the Master branch shortly and released with 2021.07

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
ghdl --elab-run --std=08 --syn-binding --work=osvvm_TbUart --workdir=D:/a/OsvvmLibraries/OsvvmLibraries/VHDL_LIBS/GHDL-2.0.0-dev/osvvm_tbuart/v08 -PD:/a/OsvvmLibraries/OsvvmLibraries/VHDL_LIBS/GHDL-2.0.0-dev TbUart_SendGet1
D:\a\_temp\msys\msys64\mingw32\bin\ghdl.exe:error: overflow detected
in process .tbuart(testharness).uartrx_1@uartrx(model).P1
  from: process osvvm_uart.uartrx(model).P1 at UartRx.vhd:359
D:\a\_temp\msys\msys64\mingw32\bin\ghdl.exe:error: simulation failed
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

Paebbels, 2021-08-04:
@JimLewis ... but `NumDataBits - 1` is underflowing the range of `integer`. The error is correct.

JimLewis, 2021-08-04:
Does it say that in the LRM?   I think only if you put range constraints on objects.

tgingold, 2021-08-14:
Yes, the design is not correct.  It is an error according to LRM08 5.2.3.1

closed 2021-08-14 by tgingold