ghdl/ghdl #1831: TYPES.INTERNAL_ERROR raised

kammoh, 2021-08-05:
**Description**
See the MWE

**Expected behaviour**
Should compile (or at least not raise an exception).

**How to reproduce?**

```vhd :file: sipo.vhdl
library IEEE;
use IEEE.std_logic_1164.all;

entity SIPO is
end entity SIPO;

architecture RTL of SIPO is
  type slv_array_t is array (natural range <>) of std_logic_vector;
  signal block_reg        : slv_array_t(0 to 3)(7 downto 0);
  
  -- the following works though
  -- type block_t is array (0 to 3) of std_logic_vector(7 downto 0);
  -- signal block_reg        : block_t;
begin
  -- same if in a process
  block_reg <= block_reg(1 to 3) & block_reg(0);
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 sipo.vhdl
```

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [x] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `gcf22e6dd`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r549.gcf22e6dd) [Dunoon edition]
Compiled with GNAT Version: 11.2.1 20210728 (Red Hat 11.2.1-
Target: x86_64-redhat-linux
/home/kamyar/src/xyz
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -o sipo.o /home/kamyar/src/xyz/sipo.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1672
Call stack traceback locations:
0x4aa511 0x5975e4 0x5981f0 0x59a7c0 0x598e20 0x599928 0x5c95e8 0x5c8850 0x5c8997 0x5c88cb 0x5c85d9 0x5cc5a9 0x5d64e8 0x5d5efb 0x62554b 0x6256bc 0x625e4e 0x6268e9 0x628cfe 0x628e51 0x612e9f 0x612f43 0x628c95 0x628e51 0x612edd 0x612f43 0x628c95 0x628e51 0x612e9f 0x612f43 0x628c95 0x628e51 0x62cb37 0x631f9a 0x5ee826 0x5e1a74 0x63d08b 0x63ea99 0x6400b6 0x40b42f 0x7fac2d860b73 0x40a5bc 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

umarcor, 2021-08-07:
I can confirm with `GHDL release: 2.0.0-dev (1.0.0.r563.gece6e78f)` on MSYS2/MINGW64 with LLVM backend.

avelure, 2021-08-11:
I hit the same case with

```vhd :file: ent.vhd
library IEEE;
use IEEE.std_logic_1164.all;

entity SIPO is
end entity SIPO;

architecture RTL of SIPO is
  type slv_array_t is array (natural range <>) of std_logic_vector;
  signal block_reg        : slv_array_t(0 to 3)(7 downto 0);
  signal block_reg_0      : std_logic_vector(7 downto 0);

begin
  block_reg <= block_reg(1 to 3) & block_reg_0;
end architecture;
```

commit 904efedb (2021-09-07): trans-chap7: Handle unbounded elements in Translate_Concatenation

Fix #1831
Fix #1657
closed 2021-09-07 by tgingold