ghdl/ghdl #1832: PSL: inf value seems not to be infinite in all cases

tmeissner, 2021-08-05:
**Description**
I've stumbled over the PSL `inf` value. It is parsed and processed without warning, but the value seems to be not "infinite". Sequences like {not a[*0:inf]; a} don't behave correctly in simulation or formal verification with SymbiYosys.

Something like

```vhdl
inf_0 : assert always {a} |=> {not b[*0 to inf]; b};
inf_1 : assert always {a} |=> {not b[*1 to inf]; b};
```

should behave similar to

```vhdl
inf_0 : assert always {a} |=> {not b[*]; b};
inf_1 : assert always {a} |=> {not b[+]; b};
```

**Expected behaviour**
`inf` should be interpreted as "infinite" in ranges.

**How to reproduce?**
Get this file: https://github.com/tmeissner/psl_with_ghdl/blob/master/issues/issue_16xx.vhd

```sh :image: ghdl/vunit:llvm
ghdl -a --std=08 issue_16xx.vhd
ghdl -e --std=08 test_issue
```

If you run it with `stop-time` option:

```sh :image: hdlc/formal:all
ghdl -r --std=08 test_issue 
issue_16xx.vhd:79:3:@8ns:(psl assertion error): Assertion violation
simulation stopped @29ns with status 0
```
The assertion should hold.

- OS: Debian Buster
- Origin:
  - [x] Docker (Image:hdlc/formal:all)

tmeissner, 2021-08-28:
Thanks for your work. I've tested with simulation, which works as intended 👍 

However, synthesis doesn't work:

```
ghdl --synth --std=08 issue_1832.vhd -e issue > work/issue_1832_synth.vhd

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/build/issues/
Command line:
ghdl --synth --std=08 issue_1832.vhd -e issue
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:794
Call stack traceback locations:
0x7f39b691e4f1 0x563c34da1dc3 0x563c34daf209 0x563c34f4e119 0x563c34f4e39c 0x563c34f3bb25 0x563c34f3bcce 0x563c34f3c04a 0x563c34f3c9e7 0x563c34f3ca69 0x563c34f2abdd 0x563c34f2acf8 0x563c34f22d61 0x563c34fad933 0x563c34eb168a 0x563c34fb7409 0x563c34d4a029 0x7f39b652e099 0x563c34d48e18 0xfffffffffffffffe
******************************************************************
```

Would it possible to support `inf` also for synthesis?

tgingold, 2021-08-28:
Ok, but OTOH the `issue` design is not supposed to be synthesizable.

tgingold, 2021-08-28:
Hold on.

tgingold, 2021-08-28:
I think it fails because the assertion is always true.  I will fix that.

tmeissner, 2021-08-29:
This seems to work now, thanks @tgingold 😃