ghdl/ghdl #1834: going out of integer range with a loop => crash (instead of error)

simon-codes-something, 2021-08-09:
**Description**
A loop can currently run out of integer'range. This is not detected by analysis.
When evaluating or running, a "GHDL Bug occurred" message is shown.

**Expected behaviour**
It should give a proper error message like when manually assigning an out of range value.

**How to reproduce?**
Minimal Working Example (MWE):

```vhd :file: example.vhd
library IEEE;
use IEEE.std_logic_1164.all;

entity tb is
end entity;
architecture a of tb is
        signal clk : std_logic := '0';
        signal running : boolean := true;
begin
        clk <= not clk after 5 ns when running else '0';

        process(clk)
                -- This assignment one throws a proper error message:
                --constant var : integer := integer'high+1;
        begin
                if(rising_edge(clk)) then
                        -- Looping out of integer range results in Bug-Message

                        -- Over int'high
                        for i in integer'high to integer'high+1 loop
                        end loop;

                        -- Under int'low
                        for i in integer'low downto integer'low-1 loop
                        end loop;
                        running <= false;
                end if;
        end process;
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a example.vhd
ghdl -e tb
```

**Context**

- OS: Ubuntu 20.04.2 LTS
- Origin:
  - [x] Package manager: `ghdl/focal,now 0.37+dfsg-1ubuntu1 amd64`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/*****/DATEN/git-repos/******/Bugreports/GHDL/
Command line:
/usr/bin/ghdl-mcode -e tb
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:2091 range check failed
Call stack traceback locations:
0x5594e94531d5 0x5594e9453c70 0x5594e9454a90 0x5594e94551f0 0x5594e93dfcef 0x5594e93e0cd9 0x5594e93e0f31 0x5594e93f1c7b 0x5594e93f1ddf 0x5594e93df423 0x5594e93df4cf 0x5594e93f1c3f 0x5594e93f1ddf 0x5594e93f51ed 0x5594e93f9ad4 0x5594e9465088 0x5594e93d153f 0x5594e93ce692 0x5594e93dbffe 0x5594e94697cd 0x5594e93b34f0 0x5594e9346557 0x5594e946cf31 0x5594e91535ff 0x7f34ecb320b1 0x5594e915212c 0xfffffffffffffffe
******************************************************************
```

**Additional context**

umarcor, 2021-08-09:
@simon-codes-something, can you please try with a more recent version of GHDL. I cannot reproduce with an up to date version on MSYS2/MINGW64 using LLVM backend or on MSYS2/MINGW32 using mcode backend. Analysis, elaboration and run are executed without warnings or crashes:

```sh
# ghdl -a example.vhd 

# ghdl -e tb

# ./tb.exe
```

simon-codes-something, 2021-08-10:
I can confirm that nightly build _ghdl-gha-ubuntu-20.04-mcode_ from May 21, 2020 displays no errors or warnings with my example code and also does not crash.

This seems even more strange as integer'high+1 and integer'low-1 should be invalid for type integer, just as they are when doing something like `report integer'image(integer'high+1)`, which results in a `error: bound check failure`.

I guess ghdl somehow does a implicit integer wraparound here. This would lead to both of the ranges being interpreted as null ranges instead of throwing the bound check failure.

umarcor, 2021-08-10:
> I guess ghdl somehow does a implicit integer wraparound here. This would lead to both of the ranges being interpreted as null ranges instead of throwing the bound check failure.

Agree. The for loops seem not to be executed at all (I tried adding report statements in them).