ghdl/ghdl #1835: Test/simulate components that are hardcoded on the FPGA

71GA, 2021-08-09:
**The problem:**

I am figuring out, how to test/simulate components that are hardcoded on the FPGA... For example...

Lattice iCE40UP5k FPGA *(package SG48)* has an internal SPI block that is hardcoded. It can be used throug a macro `SB_SPI`, defined inside Lattice's official [Ice technology library]( https://www.latticesemi.com/view_document?document_id=52206) *(pages 171, 117)*. 

So I used VHDL to write an entity `spi_icebreaker` that instantiates component `C1` of type `SB_SPI`. Instantiation step connects component `C1` to the internal signals of my entity `spi_icebreaker`. This is all inside a file `spi_icebreaker.vhdl`:

    -- A:
    library ieee;
    use ieee.std_logic_1164.all;
    
    -- B:
    entity spi_icebreaker is
    	port(
    		sbclki: in std_ulogic;
    		sbrwi: in std_ulogic;
    		sbstbi: in std_ulogic;
    		sbadri: in std_ulogic_vector (7 downto 0);
    		sbdati: in std_ulogic_vector (7 downto 0);
    		sbdato: out std_ulogic_vector (7 downto 0);
    		sbacko: out std_ulogic;
    		spiirq: out std_ulogic;
    		spiwkup: out std_ulogic;
    		mi: in std_ulogic;
    		so: out std_ulogic;
    		soe: out std_ulogic;
    		si: in std_ulogic;
    		mo: out std_ulogic;
    		moe: out std_ulogic;
    		scki: in std_ulogic;
    		scko: out std_ulogic;
    		sckoe: out std_ulogic;
    		scsni: in std_ulogic;
    		mcsno: out std_ulogic_vector (3 downto 0);
    		mcsnoe: out std_ulogic_vector (3 downto 0)
    	);
    end spi_icebreaker;
    
    
    ----------------------------------------------------------------------------------------------------
    
    
    -- C:
    architecture logic_001 of spi_icebreaker is
    
    	-- E:
    	component SB_SPI is
    		generic(
    			BUS_ADDR74: integer
    		);
    		port(
    			SBCLKI: in std_ulogic; -- System bus: System clock input
    			SBRWI: in std_ulogic; -- System bus: System read/write input
    			SBSTBI: in std_ulogic; -- System bus: Strobe signal
    			SBADRI0: in std_ulogic; -- System bus: control registers address
    			SBADRI1: in std_ulogic;
    			SBADRI2: in std_ulogic;
    			SBADRI3: in std_ulogic;
    			SBADRI4: in std_ulogic;
    			SBADRI5: in std_ulogic;
    			SBADRI6: in std_ulogic;
    			SBADRI7: in std_ulogic;
    			SBDATI0: in std_ulogic; -- System bus: System data input
    			SBDATI1: in std_ulogic;
    			SBDATI2: in std_ulogic;
    			SBDATI3: in std_ulogic;
    			SBDATI4: in std_ulogic;
    			SBDATI5: in std_ulogic;
    			SBDATI6: in std_ulogic;
[... 214 lines cut ...]
    	ghdl -a $(file_main).vhdl
    	ghdl -a $(file_main)_tb.vhdl
    	ghdl -r $(entity_top_tb) --vcd=$(file_main).vcd

I get an error: 

    ┌───┐
    │ $ │ ziga > ziga--workstation > xxx--spi_icebreaker
    └─┬─┘ /dev/pts/1
      └─> make simulate 
    ghdl -s spi_icebreaker.vhdl
    ghdl -s spi_icebreaker_tb.vhdl
    ghdl -a spi_icebreaker.vhdl
    ghdl -a spi_icebreaker_tb.vhdl
    ghdl -r spi_icebreaker_tb --vcd=spi_icebreaker.vcd
    spi_icebreaker.vhdl:161:9:warning: instance "c1" of component "sb_spi" is not bound [-Wbinding]
            C1: SB_SPI
            ^
    spi_icebreaker.vhdl:76:14:warning: (in default configuration of spi_icebreaker(logic_001))
    spi_icebreaker_tb.vhdl:144:17:@0ms:(assertion error): END OF TEST!

Component `C1` is not bound... Shouldn't it be connected to the Lattice's official [Ice technology library](https://www.latticesemi.com/view_document?document_id=52206) where `SB_SPI` hard macro is defined *(page 171, 117)*?

**Describe the solution you'd like**
I need `ghdl` to support the Lattice libraries, so that I can simulate my designs.

Paebbels, 2021-08-10:
You have 4 problems here:
1. You haven't pre-compile the Lattice primitives with e.g. the provided pre-compile scripts.  
   See https://ghdl.readthedocs.io/en/latest/getting/PrecompileVendorPrimitives.html
2. You haven't referenced that pre-compile libraries with `-P` option in your GHDL calls.  
   See https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#cmdoption-ghdl-p-directory
3. What's the purpose of a makefile when you actually write a linear Bash script?  
   Your sequence off calls doesn't nee `-s` for syntax check (is done anyway be analysis `-a`),  
   but might miss the elaborate step `-e`.  
   See https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#elaboration-e
4. Even with all enhancements, it doesn't change the fact that Lattice isn't providing simulation models for there hard IP macros.

umarcor, 2021-08-10:
As an alternative to @Paebbels' 1. and 2., you can have a "components.vhd" file in your HDL sources. See for instance the [sb_ice40_components.vhd](https://github.com/im-tomu/fomu-workshop/blob/master/hdl/sb_ice40_components.vhd) in the 'hdl' examples of Fomu workshop: [im-tomu/fomu-workshop: hdl](https://github.com/im-tomu/fomu-workshop/tree/master/hdl). There is a blink example in Verilog only, VHDL only, and mixed (Verilog-VHDL or VHDL-Verilog). All of the examples are equivalent. All of them are "built" with GHDL and Yosys, using Makefiles.

See similar component files used in NEORV32: https://github.com/stnolting/neorv32/tree/master/setups/osflow/devices. That uses GHDL and Yosys as well.

Additionally, I'm not sure about the 'component' being correct. Typically, Lattice requires generics to be of type "string" even though it makes little sense when the "natural" type would be "std_logic", "std_logic_vector" or "integer", as you did. Therefore, it might be possible that GHDL cannot find the piece, even if you satisfy all the points commented by @Paebbels.

Overall, as most traditional EDA vendors, Lattice is neither VHDL friendly nor open source friendly. Unlike other vendors, they are not beligerant with the bitstreams being reverse engineered, but they don't provide data/models for properly simulating or for using the hard IP in synthesis.