ghdl/ghdl #1837: libghdl reports no error

Paebbels, 2021-08-11:
**Description**
While testing `pyGHDL.libghdl` and it's translations to `pyGHDL.DOM`, it was found that the following code lets libghdl crash when reading the file but not print any syntax error messages.

```vhdl
entity e is
end entity;

architecture a of e is
begin
  end: std.env.stop;
end architecture;
```

The syntax mistake happend when setting the label of a concurrent procedure call to `end` instead of e.g. `ende`, as `end` is a keyword.

**Expected behaviour**
Report a syntax error.

**How to reproduce?**

**Context**
Please, provide the following information:

- OS: Windows 10 - MinGW64 build mcode64 - standalone
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `b1eb0b50` + Python modifications

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

Paebbels, 2021-08-11:
When running with GHDL (same build as libghdl), I get this:
```
PS C:\Temp\ghdl> C:\Tools\GHDL\2.0.0.dev0-mcode64\bin\ghdl.exe -s .\end-as-label.vhdl
.\end-as-label.vhdl:6:6: missing ";" at end of architecture
  end: std.env.stop;
     ^
.\end-as-label.vhdl:7:1: missing entity, architecture, package or configuration
end architecture;
^
.\end-as-label.vhdl:7:17: an identifier is expected instead of ';'
end architecture;
                ^
.\end-as-label.vhdl:7:17: 'of' is expected instead of ';'
end architecture;
                ^
.\end-as-label.vhdl:7:17: name expected here, found ';'
end architecture;
                ^
.\end-as-label.vhdl:7:17: 'is' is expected instead of ';'
end architecture;
                ^
.\end-as-label.vhdl:7:17: ';' (semi colon) not allowed alone
end architecture;
                ^
.\end-as-label.vhdl:8:1: 'begin' is expected instead of end of file

^
.\end-as-label.vhdl:8:1: unexpected end of file, 'END;' expected

^
.\end-as-label.vhdl:8:1: 'end' is expected instead of end of file

^
.\end-as-label.vhdl:7:18: missing ";" at end of architecture
end architecture;
                 ^
```

This on the other hand are a lot of messages for a single syntax error ... :)

Paebbels, 2021-08-11:
@tgingold I updated the issue description with more details. This was tested with my branch [paebbels/dom](https://github.com/Paebbels/ghdl/tree/paebbels/dom), which is based on latest `master` from GHDL.

Paebbels, 2021-08-12:
I did another update to the reproduces to investigate why GHDL and libghdl behave differently. I figured out, my reproducer was not accurate enough. A package (or other things on document level ?) must be present after the syntax error occurred. (See my updated VHDL snippet)

------------
Hmmm GHDL still prints syntax error messages while libghdl crashes ...
```
PS C:\Temp\ghdl> C:\Tools\GHDL\2.0.0.dev0-mcode32\bin\ghdl.exe -a .\end-as-label.vhdl
.\end-as-label.vhdl:6:6: missing ";" at end of architecture
  end: std.env.stop;
     ^
.\end-as-label.vhdl:7:1: missing entity, architecture, package or configuration
end architecture;
^
.\end-as-label.vhdl:7:17: an identifier is expected instead of ';'
end architecture;
                ^
.\end-as-label.vhdl:7:17: 'of' is expected instead of ';'
end architecture;
                ^
.\end-as-label.vhdl:7:17: name expected here, found ';'
end architecture;
                ^
.\end-as-label.vhdl:7:17: 'is' is expected instead of ';'
end architecture;
                ^
.\end-as-label.vhdl:7:17: ';' (semi colon) not allowed alone
end architecture;
                ^
.\end-as-label.vhdl:9:1: 'begin' is expected instead of 'package'
package p is
^
.\end-as-label.vhdl:9:1: unexpected token 'package' in a concurrent statement list
package p is
^
.\end-as-label.vhdl:11:4: missing ";" at end of architecture
end package;
   ^
.\end-as-label.vhdl:11:12: an identifier is expected instead of ';'
end package;
           ^
.\end-as-label.vhdl:11:12: 'is' is expected instead of ';'
end package;
           ^
.\end-as-label.vhdl:11:12: ';' (semi colon) not allowed alone
end package;
           ^
.\end-as-label.vhdl:12:1: 'end' is expected instead of end of file

^
.\end-as-label.vhdl:11:13: missing ";" at end of package declaration
end package;
            ^
```

I see the number of messages grows, the more text comes ...
Is the list of error records overflowing?

When switching from `errorout_memory.Install_Handler()` to `errorout_console.Install_Handler()`, the syntax error get printed, but still libghdl crashes. So no difference in the error handlers.

tgingold, 2021-08-24:
I have filled this test in `testsuite/gna/issue1837`.  When using `dom.py`, I now get:

```
dom.py pretty -F repro.vhdl 
================================================================================
                         pyGHDL.dom - Test Application                          
================================================================================
Parsing file 'repro.vhdl'
DOM: Error raised in libghdl.
libghdl: :6:5: missing ";" at end of architecture
libghdl: :7:0: missing entity, architecture, package or configuration
```

Is that OK ?

Paebbels, 2021-08-24:
Yes. Now it reaches the code to raise an exception and it can print all the details.

tgingold, 2021-08-24:
Did you try on Windows too ?

Paebbels, 2021-08-24:
With latest commits, it's fixed now.

closed 2021-08-24 by Paebbels