ghdl/ghdl #1838: synthesis: Better naming for inferred memory

antonblanchard, 2021-08-13:
When using ghdl+yosys synthesis to simulate Microwatt in verilator, the register file is not available in a vcd trace. A simplified example:

```library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity register_file is
    port(
        clk        : in std_logic;

        write_en   : in std_ulogic;
        write_addr : in std_ulogic_vector(6 downto 0);
        write_data : in std_ulogic_vector(63 downto 0);

        read_addr  : in std_ulogic_vector(6 downto 0);
        read_data  : out std_ulogic_vector(63 downto 0)
    );
end entity register_file;

architecture behaviour of register_file is
    type regfile is array(0 to 127) of std_ulogic_vector(63 downto 0);
    signal registers : regfile;
begin
    -- synchronous writes
    register_write_0: process(clk)
    begin
        if rising_edge(clk) then
            if write_en = '1' then
                registers(to_integer(unsigned(write_addr))) <= write_data;
            end if;
        end if;
    end process register_write_0;

    -- asynchronous reads
    register_read_0: process(all)
    begin
        read_data <= registers(to_integer(unsigned(read_addr)));
    end process register_read_0;

end architecture behaviour;
```

If I convert it to verilog directly via ghdl we get:
```
  reg [63:0] n21_oport[127:0] ; // memory
```

If we do the same through yosys+ghdl we get:

```
  reg [63:0] \19  [127:0];
```

In both cases we've lost the original array name, I presume because of memory inference.

tgingold, 2021-08-14:
I suppose you would like to have the name of the signal for the memory name...

antonblanchard, 2021-08-14:
@tgingold that would be ideal. If that proves difficult, having the name of the signal somewhere in the memory name (eg `register_file_oport`) seems good enough.

umarcor, 2021-08-16:
Ref: ghdl/ghdl-yosys-plugin#153

commit 2ef64244 (2021-08-25): synth: reuse signal name while creating memories. Fix #1838
closed 2021-08-25 by tgingold