ghdl/ghdl #1839: ghdl-gcc for v1.0.0 fails

wfjm, 2021-08-15:
**Description**
gcc build as described in `readthedocs` works for 0.37 but not for 1.0.0

**Expected behaviour**
That https://ghdl.readthedocs.io/en/latest/getting/GCC.html#build-gcc gives proper build instructions

**How to reproduce?**
I use `Ubuntu 20.04` and `gcc-9.3.0`.  
I've build the ghdl-gcc for version 0.37, no problems.

A build after `git checkout tags/v1.0.0` fails with
```
    pushd gcc-objs
    ../../gcc-9.3.0/configure --prefix=$PREFIX --enable-languages=c,vhdl \
       --disable-bootstrap \
       --disable-lto \
       --disable-multilib \
       --disable-libssp \
       --disable-libquadmath \
       --disable-libgomp 2>&1 |\
          tee configure.log

      configure: WARNING: using in-tree isl, disabling version check
      configure: error: 
      The following requested languages could not be built: vhdl
```
Even though 1.0.0 was released in 6 month ago in February, the only trace of this major release in `NEWS.md` is
```
**1.0-dev** (XXXX-XX-XX)
```

Xiretza, 2021-08-15:
Please post the entire build log.

umarcor, 2021-08-15:
@wfjm, I believe that guidelines for buiding GHDL with GCC backend did not change, however, the updated docs are available in [ghdl.github.io/ghdl](https://ghdl.github.io/ghdl). All references to `ghdl.rtfd.io` should be removed from the repo...

wfjm, 2021-08-16:
I've re-done the v1.0.0 build, and the second time it worked.  
No idea why it failed, I had done a v0.37 build before, which worked.  

So the best of this issue are the side effects :). Thanks for updating NEWS and the docs pointers !

closed 2021-08-16 by wfjm