ghdl/ghdl #1842: GHDL improperly handles the situation where only a part of output record is driven?

wzab, 2021-08-16:
**Description**
It seems that if only a part of the record being an output of the entity is driven, GHDL does not drive the whole record.

**Expected behaviour**
The undriven part of the record should be set to 'U'. Instead of that it behaves like it has no driver.

**How to reproduce?**
I attach the MWE together with the makefile. However, for easier reading, I include them also in the issue:

```vhd :file: signal_pkg.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
library work;

package signal_pkg is

  type t_sigs is array (natural range <>) of std_logic_vector(7 downto 0);
  
  type t_signals is record
    dta: t_sigs(0 to 7);
  end record;
                 
end signal_pkg;
```
```vhd :file: source.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
library work;
use work.signal_pkg.all;

entity source is
  generic (
    instance_number : integer := 0);
  port (
    rst_n_i   : in  std_logic;
    clk_i : in  std_logic;
    outs : out  t_signals
    );
end entity source;

architecture sim of source is

  signal toggle : std_logic := '0';

begin  -- architecture rtl
  process(clk_i)
  begin
    if rising_edge(clk_i) then
      if rst_n_i = '0' then
        outs.dta(instance_number) <= (others => '0');
      else
        toggle <= not toggle;
        if toggle='0' then
          outs.dta(instance_number) <= (others => '0');          
        end if;
          outs.dta(instance_number) <= std_logic_vector(to_unsigned(instance_number + 2, 8));
      end if;
    end if;
  end process;
end architecture sim;
```
```vhd :file: tester_tb.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
library work;
use work.signal_pkg.all;

entity tester_tb is
end tester_tb;

architecture beh1 of tester_tb is

  signal clk : std_logic := '0';
  signal rst_n : std_logic := '0';
  signal test : t_signals;
  
begin  -- beh1

  clk <= not clk after 10 ns;
  
  process
  begin
    rst_n <= '0';
    wait for 30 ns;
    rst_n <= '1';
    wait;
  end process;

  g1: for i in 0 to 7 generate
    source_1: entity work.source
      generic map (
        instance_number => i)
      port map (
        rst_n_i => rst_n,
        clk_i   => clk,
        outs    => test);
  end generate g1;
  
end beh1;
```
```makefile :file: makefile
VSTD=02
ENTITY=tester_tb
RUN_OPTIONS= --stop-time=2900ns --wave=${ENTITY}.ghw 
SOURCES = \
  signal_pkg.vhd \
  source.vhd \
  tester_tb.vhd \

OBJECTS=$(SOURCES:.vhd=.o)

all: $(OBJECTS)

%.o : %.vhd
	ghdl -a -g -C  --std=${VSTD} $<

all: show
show:   ${ENTITY} ${ENTITY}.ghw
	gtkwave ${ENTITY}.ghw ${ENTITY}.sav
${ENTITY}: $(SOURCES:.vhd=.o)
	ghdl -e -g --mb-comments  --std=${VSTD}  ${ENTITY}
${ENTITY}.ghw: ${ENTITY}
	./${ENTITY} ${RUN_OPTIONS} 
clean:
	rm -f comp/* *.o *.vcd *.ghw *.cf events* ${ENTITY} 
```

**Context**
Please, provide the following information:

- OS: Debian/testing
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ x] Built from sources: `b1eb0b50572ceef9efa19fe00ebdf7b952a49692`
```
$ ghdl --version
GHDL 2.0.0-dev (1.0.0.r565.gb1eb0b50) [Dunoon edition]
 Compiled with GNAT Version: 10.2.1 20210110
 llvm code generator
```

**Additional context**
When I run make, I get the following result:
[image]
It shows that each instance of the `source` drives only a part of the `test.dta` array. Therefore, we get all elements driven by individual `source` components. That's incorrect, as the whole `outs` output should be driven, causing the collision.
If I add the intermediate signal:

```vhd :file: source.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
library work;
use work.signal_pkg.all;

entity source is
  generic (
    instance_number : integer := 0);
  port (
    rst_n_i   : in  std_logic;
    clk_i : in  std_logic;
    outs : out  t_signals
    );

end entity source;

architecture sim of source is

  signal toggle : std_logic := '0';
  signal s_outs : t_signals;
  
begin  -- architecture rtl
  process(clk_i)
  begin
    if rising_edge(clk_i) then
      if rst_n_i = '0' then
        s_outs.dta(instance_number) <= (others => '0');
      else
        toggle <= not toggle;
        if toggle='0' then
          s_outs.dta(instance_number) <= (others => '0');          
        end if;
          s_outs.dta(instance_number) <= std_logic_vector(to_unsigned(instance_number + 2, 8));
      end if;
    end if;
  end process;

  outs <= s_outs;
end architecture sim;
```
(you may just change `source.vhd` into `source2.vhd` in the makefile)
Then of course I get the expected behavior:
[image]

I must admit, that it would be very nice, if VHDL would behave as in the first case. I have a lot of designs, where using a single record to concentrate data from multiple sources could significantly improve legibility and maintainability of the design. 
However, as far as I know, that's a violation of the standard.

[mwe.zip](https://github.com/ghdl/ghdl/files/6991751/mwe.zip)

Paebbels, 2021-08-16:
Independent of this issue, could a procedure help in your case like [assign_row](https://github.com/VLSI-EDA/PoC/blob/master/src/common/vectors.vhdl?ts=2#L225) and a `signal vector : T_SLM(..., ...) := (others => (others => 'Z'));`

Using `Z` solves the multiple driver problem on vectors and records that are driven as one signal in case of assignments in procedures. Moreover, this can also be synthesized and is compliant to the LRM.

wzab, 2021-08-16:
I have prepared a demo that compares the behavior of GHDL and Xilinx tools. Therefore I have added a `concentrator.vhd` entity that instantiates 8 `source.vhd` entities, and connects the output signal to all of them. 
I attach two zip archives. The first one `ghdl_demo.zip` shows that the design behaves as previously - nice, but incorrectly:
[image]

The second one `vivado_demo.zip` in the behavioral simulation shows the expected (according to LRM) behavior - the `test` signal remains set to 'U':
[image]
[ghdl_demo.zip](https://github.com/ghdl/ghdl/files/6994393/ghdl_demo.zip)
[vivado_demo.zip](https://github.com/ghdl/ghdl/files/6994401/vivado_demo.zip)
The funny thing is that the design synthesizes correctly (no multiple drivers reported).
However, the post-synthezis and post-implementation simulation in Vivado fails (due to xsim limitations?)

wzab, 2021-08-21:
The solution proposed by @Paebbels  in https://github.com/ghdl/ghdl/issues/1842#issuecomment-899462969 indeed may solve the design issue that lead to discovery of the problem with GHDL. 

That however requires providing the constant that will automatically fill all elements of the record with 'Z'. The problem is that in the real design that record may contain bit vectors, arrays of bit vectors, nested records and/or arrays of nested records. Of course the system creates also the whole hierarchy of types. The only solution, I can see is generation of the appropriate constant (containing 'Z' everywhere) together with the appropriate type. 

Another question is: do all synthesis tools handle properly 'Z' value appearing inside the design? I had some problems with that, but it was many years ago (either an early version of Vivado, or even ISE, I really don't remember). 

Of course, the above does not invalidate the fact that GHDL seems to behave in a way not conforming to LRM.

tgingold, 2021-08-23:
I think you're correct and there is a problem on sources when signals are collapsed.

wzab, 2021-08-24:
So that's indeed a bug in GHDL?

tgingold, 2021-08-24:
Yes, I do think so.