ghdl/ghdl #1844: GHDL compilation bug

nfrancque, 2021-08-19:
**Description**
GHDL is erroring on a file that is compiled in several other simulators without issue.  I am not sure of much beyond that as I am not sure how to check what line the issue is caused by.

**Expected behaviour**
I would expect the file to compile without error.

**How to reproduce?**
If someone can provide suggestions on getting more info I can work on an MWE.  It is failing on a large testbench package.

**Context**
Please, provide the following information:
This is my first time using ghdl, I am trying to run our regression suite that has been used in a few other simulators so far.  GHDL is encountering an exception shown below on one file that is mostly a lot of utils procedures for vunit testbenches.

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ x] Released binaries: `https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-llvm.tgz`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-568-gbbdff59e) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/nathan/ics/hdl_libs/bin/
Command line:
/home/nathan/ghdl/ghdl-gha-ubuntu-20.04-llvm/bin/ghdl1-llvm --workdir=/home/nathan/ics/hdl_libs/scratch/vunit_out/ghdl/libraries/hdl_libs_tb --work=hdl_libs_tb --std=08 -frelaxed -P/home/nathan/ics/hdl_libs/scratch/vunit_out/ghdl/libraries/vunit_lib/ -P/home/nathan/ics/hdl_libs/scratch/vunit_out/ghdl/libraries/adder_tree_dsn/ -P/home/nathan/ics/hdl_libs/scratch/vunit_out/ghdl/libraries/adder_tree_tb/ -P/home/nathan/ics/hdl_libs/scratch/vunit_out/ghdl/libraries/ics_vunit_dsn/ -P/home/nathan/ics/hdl_libs/scratch/vunit_out/ghdl/libraries/ics_vunit_tb/ -P/home/nathan/ics/hdl_libs/scratch/vunit_out/ghdl/libraries/hdl_libs_dsn/ -P/home/nathan/ics/hdl_libs/scratch/vunit_out/ghdl/libraries/hdl_libs_tb/ -P/home/nathan/ics/hdl_libs/scratch/vunit_out/ghdl/libraries/osvvm/ -P/home/nathan/ghdl/ghdl-gha-ubuntu-20.04-llvm/lib/ghdl/ieee/v08/ -P/home/nathan/ghdl/ghdl-gha-ubuntu-20.04-llvm/lib/ghdl/ -fpic -c -o /home/nathan/ics/hdl_libs/scratch/vunit_out/ghdl/libraries/hdl_libs_tb/axis_tb_utils_pkg.o /home/nathan/ics/hdl_libs/scratch/vunit_out/preprocessed/hdl_libs_tb/axis_tb_utils_pkg.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Is_Within_Flag
Call stack traceback locations:
0x7fb0a9de1d12 0x49c409 0x54de7d 0x54e05b 0x553f03 0x56c019 0x56d394 0x544ef2 0x546b4e 0x546cf9 0x58658c 0x578c92 0x578ed9 0x58822a 0x589376 0x53e182 0x64f91a 0x65168d 0x652cd7 0x40b40b 0x7fb0a967d0b1 0x40a58c 0xfffffffffffffffe
******************************************************************

```

**Additional context**

nfrancque, 2021-08-20:
Confirming that the same issue occurs with [mcode](https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-mcode.tgz) and [GCC](https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-gcc.tgz) backends as well.

nfrancque, 2021-08-20:
I am able to work around this for the time being by removing the following line:

```vhdl
return get_words(byte_arr, idx, word_width, idx)(0).data;
```
get_words returns an an array of words, here are the types:

```vhdl
  type word_t is record
    data         : std_logic_vector;
    byte_enables : std_logic_vector;
  end record;

  type words_arr_t is array (natural range <>) of word_t;
```

I am doing this to work around having to know the length of the vector and assign it to a temporary variable.  A suggestion for a change would be welcome, but it seems like this should still be compilable.

umarcor, 2021-08-20:
@nfrancque, please provide a MWE. A (maybe empty) entity and an architecture. Precisely, we are missing the body of `get_words`, which I assume returns a `words_arr_t` which you are then indexing.

Overall, unconstrained arrays/records as in VHDL 2008 are not fully supported yet. Tristan implemented several enhancements recently, but that's still a pending feature.

nfrancque, 2021-08-20:
@umarcor Good to know, I will work at on MWE.

I can work around unconstrained records but our whole test infrastructure is based around unconstrained arrays.  I encountered another exception in the same file that is mostly a lot of operations on unconstrained arrays after commenting out this line.  Will be reporting that as well.  Are you saying in general we should not expect unconstrained arrays to be compilable in ghdl yet?

umarcor, 2021-08-20:
> Are you saying in general we should not expect unconstrained arrays to be compilable in ghdl yet?

While unconstrained arrays are usable, unconstrained arrays of unconstrained arrays, or unconstrained arrays of records with unconstrained arrays, or any other similar variation of "complex" types introduced in VHDL 2008 might crash. Yet, it is a so wanted feature and we have an specific label for tracking the MWEs provided by users: https://github.com/ghdl/ghdl/labels/FeaReq%3A%20VHDL-2008%20%28unconst.%20arrrays%2Frecords%29. That's why I asked you to provide one, or many. We cannot guarantee that the feature will be completely supported soon, but having specific examples of what users need is very helpful. Tristan likes MWE driven development, given the vast complexity of completely supporting all the variants of the language. That helps him and us do some light "gamification" of the development.

As a matter of fact, VHDL 2008 features were Tristan's priority until he was sidetracked on implementing synthesis (both `ghdl synth` and ghdl-yosys-plugin). So, during 2019 and 2020, the implementation of VHDL 2008 features was kind of on hold. I believe we are all grateful regardless. As a community, having an open source VHDL frontend for synthesis has allowed achievements such as the Microwatt sent to the Skywater/Google/Efabless shuttle, or using VHDL 2008 for synthesis and implementation with Vivado.

Now, in 2021, Tristan switched focus back to missing VHDL 2008 features, and we are seeing a lot of fixes/enhancements in that regard. In parallel, Patrick and me are being inquisitive about libghdl and the Python interface, but Tristan is overall comitting fixes related to VHDL 2008. During the last weeks, 8 issues/bugs were fixed: https://github.com/ghdl/ghdl/issues?q=label%3A%22FeaReq%3A+VHDL-2008+%28unconst.+arrrays%2Frecords%29%22+is%3Aclosed+sort%3Aupdated-desc. Some of those were opened by me in 2019 and 2020, as compediums of MWEs about unconstrained multidimensional arrays of unconstrained/complex types. See #1055 and #1455. This is to say that enhancements in this area are now more probable than during the last 3 years, and many of your uses should not crash. However, there are 10 open bug reports yet, and you are getting crashes, so it is obviously not complete. That's why reproducible examples are very welcome.

nfrancque, 2021-08-20:
Thank you for the detailed explanation!  I will get MWE for this when I am back on the same machine.  For the other issue I will be reporting, it did not seem as easy to locate the line the issue occurred on.   Is there something like a compile flag that can print out the line it was processing when an exception occurred?  Otherwise it may take me a while to figure out.

umarcor, 2021-08-20:
On the one hand, Tristan is pretty good at finding the location of the bugs. Typically, if you provide a reproducible example (one he can `ghdl -a`, `ghdl -e` and/or `ghdl -r` to trigger the bug), he will find the problem and maybe fix it before you can understand what's going on. That depends on the complexity of the bug, of course. Therefore, as long as you can share the subsets of your code that trigger the bug, you don't need to minimize them. Naturally, the smaller the example, the better, but it's not a requirement.

On the other hand, if you build GHDL with backtrace support, you can get some more info. However, I'm not sure about that working in case of a crash of GHDL itself. The "developer" way is to run GHDL on gdb and have it crash there. There is some info in https://ghdl.github.io/ghdl/development/Debugging.html#gnu-debugger-gdb.

nfrancque, 2021-08-24:
@umarcor I have been trying to narrow this down but it is very strange.  There are lines I can comment out that will cause the exceptions to go away, but moving just those problem lines into their own files don't cause the exception.

I don't think I can post an entire file in an issue, can I send the file privately to you or @tgingold ?  It is about 1300 lines as is, but I should be able to at least remove some of it.  Most of the file is defining operations on some multi-dimensional array types.

tgingold, 2021-08-24:
Feel free to send it privately to me.

The size is not really an issue, but the reproducer should be standalone.

nfrancque, 2021-08-24:
Great, much appreciated @tgingold.

Keeps getting weirder though.  I went to go get an example for you on my work machine ([windows](https://github.com/ghdl/ghdl/releases/download/nightly/MINGW64-mcode-standalone.zip)), and the compile failures do not occur there.  My home PC is ubuntu 20.04.

But then once I start any simulation I encounter similar exceptions.  What's the best way to contact you?

umarcor, 2021-08-24:
@nfrancque, mcode backend is not supported on Windows 64 bits. The standalone zipfile you are using is expected to be used for libghdl (the language server, pyGHDL.dom, etc.), but simulation does not work. You need to pick MINGW64 LLVM or MINGW32 mcode. We don't have "standalone" zipfiles for those yet.

> What's the best way to contact you?

Check his e-mail in the commit history, or use gitter :wink:

tgingold, 2021-08-24:
If your reproducer is public, it would be better to put it in this issue, just attach a zip file if needed.

Otherwise, my email address is tgingold
 AT free DOT fr

nfrancque, 2021-08-24:
Unfortunately I don't think I can attach here.  We are still figuring out our open source model, what can be pushed back etc.  Tristan, I emailed you the example.

tgingold, 2021-08-25:
With the fix, I was able to compile your testcase.  But the error message was not the one in the first message of this issue.
So there might be a different issue.