ghdl/ghdl #1845: Building GHDL package fails for GCC backend, with gcc 11.2

marzoul, 2021-08-21:
**Description**
Building GHDL package fails for GCC backend, with gcc sources version 11.2 (no problem with gcc 10.3)
Error happens at compiling VHDL libraries, error message is the following:

`ghdl1: error: unrecognized command-line option ‘-auxbase’
/home/adrien/Dev/AUR/ghdl-gcc-git/src/gcc-build/gcc/ghdl:error: compilation error
make[1]: *** [libraries/Makefile.inc:172: lib/ghdl/std/v87/std-obj87.cf] Error 1
make[1]: Leaving directory '/home/adrien/Dev/AUR/ghdl-gcc-git/src/ghdl'
make: *** [Makefile:293: libs.vhdl.gcc] Error 2
`

It looks like GHDL is driving the binary `GHDL1` with an option `-auxbase` that is not recognized with gcc sources version 11.2.

**Expected behaviour**
Same compilation flow than with gcc version 10.3.
No error at compilation of VHDL libraries.

**How to reproduce?**
Need to use Archlinux distribution.
Compile Archlinux AUR package that is available at following page:
[https://aur.archlinux.org/packages/ghdl-gcc-git/](https://aur.archlinux.org/packages/ghdl-gcc-git/)

Edit the `PKGBUILD` file:
- set gcc version 11.2.0
- set the sha256sum for the gcc sources to `'SKIP'` which is enough for the context of this reproducer

Then compile the package with usual command `makepkg`.

**Context**
- OS: Archlinux
- Origin: Built from sources: `commit bbdff59e4e68e89033843f9344814c957b0b6250`

tgingold, 2021-08-25:
Should now be fixed.  Thank your for the report.

closed 2021-08-25 by tgingold

umarcor, 2021-08-25:
It seems that the fix broke the build with GCC 8: https://github.com/ghdl/ghdl/actions/runs/1165194186. @tgingold, is that something to be fixed/handled in https://github.com/ghdl/ghdl/blob/master/scripts/ci-run.sh ?

commit a5ff9b62 (2021-08-25): ghdldrv: handle auxbase option in ortho/gcc.  Fix #1845

Remove additional option in ghdldrv.adb,
automatically set aux_base_name in ortho-lang-9.c