ghdl/ghdl #1853: PSL: Enhance use cases of generate indexes to PSL operators

tmeissner, 2021-08-30:
**Is your feature request related to a problem? Please describe.**
After fixing ghdl#1850 is it possible to use indexes from generate loops in PSL vunits. For example the code from the issue:

```vhdl
-- generate pulse one cycle assertion
for J in 0 to 15 generate
  assert always {(not rst) and output_pulse(J) = '1'} |=> {output_pulse(J) = '0'};
end generate;
```

What is still not possible: Using that indexes in context of PSL operators like repetition `*`, `->` or `next[i]` etc. If I understand correctly the issue with them is the meaning of staticness of PSL. AFAIK the indexes in generate loops are locally static. Wouldn't that be enough to support them as parameters of PSL operators?

**Describe the solution you'd like**
I would like to use code like this simple example:

```vhdl
-- Cover for number of requests
cover_gen : for i in 1 to 8 generate
  cov_req : cover {req[*i]};
end generate cover_gen;
```

Or a more complex one:

```vhdl
-- Checking correct data flow in the FIFO
gen_checks : for i in 0 to 7 generate
  DATA_FLOW_GEN : assert always
    {{s_cnt = i and Wen_i = '1' and Din_i = s_data} : {Ren_i[->i+1]}}
    |=>
    {Dout_o = s_data};
end generate gen_checks;
```
If arithmetic is like `{Ren_i[->i+1]}` is hard, we could first support simple substitution. Then `{Ren_i[->i+1]}` could be rewritten as `{Ren_i; not Ren_i[*]; Ren_i[->i]}` for example to getting it work.

I could create a MWE if requested.

**Describe alternatives you've considered**
At the moment I have to roll out such loops by hand like this for the second example:

```vhdl
-- Data flow checks
-- GHDL only allows numerals in repetition operators
-- so we have to use separate checks for each fill state 
DATA_FLOW_0 : assert always
  {{s_cnt = 0 and Wen_i = '1' and Din_i = s_data} ; {Ren_i[->1]}}
  |=> {Dout_o = s_data};
DATA_FLOW_1 : assert always
  {{s_cnt = 1 and Wen_i = '1' and Din_i = s_data} : {Ren_i[->2]}}
  |=> {Dout_o = s_data};
DATA_FLOW_2 : assert always
  {{s_cnt = 2 and Wen_i = '1' and Din_i = s_data} : {Ren_i[->3]}}
  |=> {Dout_o = s_data};
DATA_FLOW_3 : assert always
  {{s_cnt = 3 and Wen_i = '1' and Din_i = s_data} : {Ren_i[->4]}}
  |=> {Dout_o = s_data};
DATA_FLOW_4 : assert always
  {{s_cnt = 4 and Wen_i = '1' and Din_i = s_data} : {Ren_i[->5]}}
  |=> {Dout_o = s_data};
DATA_FLOW_5 : assert always
  {{s_cnt = 5 and Wen_i = '1' and Din_i = s_data} : {Ren_i[->6]}}
  |=> {Dout_o = s_data};
DATA_FLOW_6 : assert always
  {{s_cnt = 6 and Wen_i = '1' and Din_i = s_data} : {Ren_i[->7]}}
  |=> {Dout_o = s_data};
DATA_FLOW_7 : assert always
  {{s_cnt = 7 and Wen_i = '1' and Din_i = s_data} : {Ren_i[->8]}}
  |=> {Dout_o = s_data};
```

tgingold, 2021-08-30:
Unfortunately, in a for-generate the iterator is globally constant (and not locally).

As GHDL is translating the properties to an NFA, it cannot handle non-locally constants.

tmeissner, 2021-09-11:
Maybe supporting PSL macros like `%for` would be a solution (IEEE 1850-2010 4.2.4 Macros). They would be processed during a before analyzing the code. We could try to implement them for `vunits`. Then, one could write the above as:

```vhdl
-- Cover for number of requests
%for ii in 1..8 do
  cov_req : cover {req[*ii]};
%end
```