ghdl/ghdl #1855: Cosim function buffering wavefile display in GTKWave

russellbdave, 2021-09-04:
Hi
**Description**
I'm attempting to write a sim environment that allows stepping through a pipeline with key presses via cosim. However, when the blocking cosim function is called, only a certain number of delta cycles from GHDL are displayed in GTKwave.

**Expected behaviour**
I setup a quick reproduction of this issue in a small testbench. On clock ticks 1000 and 2000, the cosim function is being called and gets blocked by scanf within the C code. At this time I refresh the viewer in GTKwave and expect to see 1000 or 2000, but only see up to about 900 and 1800 clocks (respectively). Eventually the sim finishes as expected, but I'm not sure why they don't show in the viewer at that time.

**How to reproduce?**
Makefile:
```sh :image: ghdl/ghdl:bullseye-mcode
GHDL = /opt/ghdl/bin/ghdl
GTKW = gtkwave
default:
	$(GHDL) -a --warn-no-hide tb.vhd
	$(GHDL) --bind tb
	gcc -no-pie main.c -o main.o -Wl,`$(GHDL) --list-link tb`
	./main.o --wave=tb.ghw
gui:
	$(GTKW) tb.ghw	
```

Testbench:
```vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity tb is
end tb;

architecture test of tb is

    signal clk : std_logic := '0';
    signal tick : integer := 0;
    signal kill_sim : std_logic := '0';
    signal return_int : integer := 0;

    function input_handler(test : integer) return integer is
    begin
    end function;
    attribute foreign of input_handler : function is "VHPIDIRECT input_handler";

begin
    clk_stim: process(clk)
    begin
        if(kill_sim = '0') then
            clk <= not clk after 5 ns;
        else
            assert false report "Clock stopped, end of sim!" severity note;
        end if;
    end process;
    process(clk)
    begin
        if(rising_edge(clk)) then
            tick <= tick + 1;
            if(tick = 1000 or tick = 2000) then
                return_int <= input_handler(0);
            end if;
            if(tick = 10000) then
                kill_sim <= '1';    
            end if;     
        end if;
    end process;
    process
    begin
        wait;
    end process;

end test;

```

```c
#include <stdio.h>
#include <stdlib.h>

extern int ghdl_main(int argc, char** argv);

int tick = 0;
int input_handler(int test){
    char c;
    printf("\nEnter character to advance sim: ");
    scanf(" %c",&c);
    tick++;
    return(tick);
}

int main(int argc, char** argv) {
    ghdl_main(argc,argv);
    return 0;
}

```

**Context**
Please, provide the following information:

OS - Debian 64-bit
GHDL - 2.0.0-dev (compiled with GNAT 8.3, GCC back-end)
GTKWave - 3.3.98

Thanks.

umarcor, 2021-09-04:
Let's try this and see if it makes any differene:

```vhdl
    begin
        if(rising_edge(clk)) then
            if(tick mod 1000 = 0) then
                if(input_handler(tick) = 2) then
                  assert false report "Clock stopped, end of sim!" severity note;
                end if;    
            end if;
            tick <= tick + 1;
         end if;
    end process;
```

russellbdave, 2021-09-04:
Modified the testbench to your reply and still have the same issue. I tried sprinkling wait statements like mentioned in #1187 but no dice. 

Order of events:
1. Run the sim, open GTKWave
2. Wait for the first key press prompt (tick value is 1000)
3. Load waveform in GTKWave. See last tick value in waveform is ~960
4. Enter the key press, sim advances tick to 2000
5. Reload GTKWave. See last tick value is ~1230

tgingold, 2021-09-04:
I think we need to add a function to flush the dump buffer.
Maybe you will have a better chance with vcd.

russellbdave, 2021-09-04:
Tested with vcd and got the same resullt unfortunately

tgingold, 2021-09-05:
Next test: call `fflush(NULL);` at the start of `input_handler`.

russellbdave, 2021-09-05:
Perfect, that was it. Appreciate the help everyone

closed 2021-09-05 by russellbdave