ghdl/ghdl #1856: cannot handle IIR_KIND_CONSTANT_DECLARATION in PSL Vunit

Martoni, 2021-09-08:
**Description**
If I declare constant under vunit bloc psl : 

```vhdl
vunit i_mymodule(mymodule(mymodule_1))
{
//...
    constant avalue: natural := 2;
//...
```

I get a GHDL bug  in yosys-smtbmc test : 

```console
 make bmc
sby --yosys "yosys -m /opt/ghdl-yosys-plugin/ghdl.so" -f mymodule.sby bmc
SBY 10:04:14 [mymodule_bmc] Removing directory 'mymodule_bmc'.
SBY 10:04:14 [mymodule_bmc] Copy '../../src/legacy/mymodule.vhd' to 'mymodule_bmc/src/mymodule.vhd'.
SBY 10:04:14 [mymodule_bmc] Copy '../../src/legacy/constellium_pkg.vhd' to 'mymodule_bmc/src/constellium_pkg.vhd'.
SBY 10:04:14 [mymodule_bmc] Copy 'mymodule.psl' to 'mymodule_bmc/src/mymodule.psl'.
SBY 10:04:14 [mymodule_bmc] engine_0: smtbmc
SBY 10:04:14 [mymodule_bmc] base: starting process "cd mymodule_bmc/src; yosys -m /opt/ghdl-yosys-plugin/ghdl.so -ql ../model/design.log ../model/design.ys"
SBY 10:04:14 [mymodule_bmc] base: annotate_vunit_declaration: cannot handle IIR_KIND_CONSTANT_DECLARATION (mymodule.psl:50:14)
SBY 10:04:14 [mymodule_bmc] base: ******************** GHDL Bug occurred ***************************
SBY 10:04:14 [mymodule_bmc] base: Please report this bug on https://github.com/ghdl/ghdl/issues
SBY 10:04:14 [mymodule_bmc] base: GHDL release: 2.0.0-dev (1.0.0.r720.g5a09c3b5) [Dunoon edition]
SBY 10:04:14 [mymodule_bmc] base: Compiled with unknown compiler version
SBY 10:04:14 [mymodule_bmc] base: Target: x86_64-linux-gnu
SBY 10:04:14 [mymodule_bmc] base: /usr/local/projects/mymodule_bmc/src/
SBY 10:04:14 [mymodule_bmc] base: Command line:
SBY 10:04:14 [mymodule_bmc] base: Exception TYPES.INTERNAL_ERROR raised
SBY 10:04:14 [mymodule_bmc] base: Exception information:
SBY 10:04:14 [mymodule_bmc] base: raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
SBY 10:04:14 [mymodule_bmc] base: ******************************************************************
SBY 10:04:14 [mymodule_bmc] base: ERROR: vhdl import failed.
SBY 10:04:14 [mymodule_bmc] base: finished (returncode=1)
SBY 10:04:14 [mymodule_bmc] base: job failed. ERROR.
SBY 10:04:14 [mymodule_bmc] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 10:04:14 [mymodule_bmc] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 10:04:14 [mymodule_bmc] DONE (ERROR, rc=16)
SBY 10:04:14 One or more tasks produced a non-zero return code.
make: *** [Makefile:19: mymodule_bmc/PASS] Error 16
```

**Expected behaviour**
it may be forbidden to declare `constant` in `vunit` blocks but in this case, why is it a GHDL bug and not a simple error ?

- OS:
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.2 LTS"
```
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [* ] Built from sources: `commit 5a09c3b571749808a8aee26f71aaa963f78cf094`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
SBY 10:04:14 [mymodule_bmc] base: ******************** GHDL Bug occurred ***************************
SBY 10:04:14 [mymodule_bmc] base: Please report this bug on https://github.com/ghdl/ghdl/issues
SBY 10:04:14 [mymodule_bmc] base: GHDL release: 2.0.0-dev (1.0.0.r720.g5a09c3b5) [Dunoon edition]
SBY 10:04:14 [mymodule_bmc] base: Compiled with unknown compiler version
SBY 10:04:14 [mymodule_bmc] base: Target: x86_64-linux-gnu
SBY 10:04:14 [mymodule_bmc] base: /usr/local/projects/mymodule_bmc/src/
SBY 10:04:14 [mymodule_bmc] base: Command line:
SBY 10:04:14 [mymodule_bmc] base: Exception TYPES.INTERNAL_ERROR raised
SBY 10:04:14 [mymodule_bmc] base: Exception information:
SBY 10:04:14 [mymodule_bmc] base: raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
SBY 10:04:14 [mymodule_bmc] base: ******************************************************************
```

tmeissner, 2021-09-08:
I think it is legal and should be supported. AFAIK vunits can contain HDL declarations and statements without further restrictions (Section 7.2 of IEEE 1850-2010):

> Vunit_Item ::=
HDL_DECL
| HDL_STMT
| PSL_Declaration
| PSL_Directive
| Vunit_Instance

> A Vunit Item can be any of the following:
a) Any modeling layer statement or declaration.
b) A property, sequence, or default clock declaration.
c) Any verification directive.

tgingold, 2021-09-08:
Now fixed, but please it is always better to provide a reproducer.

closed 2021-09-08 by tgingold

Martoni, 2021-09-08:
thanks @tgingold .
Sorry for the reproducer, next time I will.

commit 604d6b28 (2021-09-08): vhdl: allow constants in vunit declarations. Fix #1856