ghdl/ghdl #1857: VHDL-2008 : 'subtype attribute used with a unconstrained port crashes GHDL

NicoPy, 2021-09-09:
Here are the two files used for the test :

The DUT :
```vhd :file: subtype_test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity subtype_test is
port
(
    i_Data : in  unsigned; -- (3 downto 0);
    o_Data : out unsigned
);
end entity subtype_test;

architecture RTL of subtype_test is

-- signal Data : unsigned(i_Data'Range) := (others=>'0');
signal Data : i_Data'subtype := (others=>'0');

begin

    Data   <= i_Data;
    o_Data <= Data;

end architecture RTL;
```

The test bench :
```vhd :file: subtype_test_tb.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

--  A testbench has no ports.
entity subtype_test_tb is
end subtype_test_tb;

architecture behav of subtype_test_tb is

signal Data1 : unsigned (4-1 downto 0) := x"0";
signal Data2 : unsigned (4-1 downto 0);

begin
    c_pwm : entity work.subtype_test
        port map 
        (
            i_Data => Data1,
            o_Data => Data2
        );
end behav;
```
`ghdl -i --std=08 --workdir=work *.vhd` runs without error
`ghdl -m --std=08 --workdir=work subtype_test_tb` crashes GHDL.

In `subtype_test.vhd` , constraining either `i_Data` or `Data` (or both) removes the crash.

**Context**

- OS: Windows 10 (MSYS2 64bit)
- Origin:
  - [ ] Package manager: `GHDL 2.0.0-dev (1.0.0.r517.g9262a810) [Dunoon edition]`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r517.g9262a810) [Dunoon edition]
Compiled with GNAT Version: 10.3.0
Target: x86_64-w64-mingw32
Z:\ghdl\subtype_test\
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 --workdir=work -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o work\subtype_test.o subtype_test.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap7.adb:3555
Call stack traceback locations:
0x7ff666dacb06 0x7ff665b08bfc 0x7ff665b0974f 0x7ff665b0c429 0x7ff665b0c945 0x7ff665b0c6b9 0x7ff665b3ed47 0x7ff665b3ee2f 0x7ff665b44a93 0x7ff665b75716 0x7ff665b2755e 0x7ff665b1962c 0x7ff665b79aed 0x7ff665b7b56e 0x7ff665b7ce45 0x7ff6659426f2 0x7ff6659413af 0x7ff6659414e4 0x7ff9aee87032 0x7ff9af98264f
******************************************************************
```

Paebbels, 2021-09-09:
@jimlewis is this one of the fixes OSVVM needs to reduce the number of workarounds?

tgingold, 2021-09-10:
I can reproduce the crash.  But it is due to the initialization of the signal with an aggregate rather than the subtype attribute.

NicoPy, 2021-09-10:
How the signal should be initialized ?

tgingold, 2021-09-10:
Your initialization is correct, but not handled by GHDL (yet).

NicoPy, 2021-09-10:
Ok. Sorry for the misunderstanding.

commit 50e34d98 (2021-09-16): trans-chap4: handle unbounded aggregate initial value to unbounded signal.

Fix #1857
closed 2021-09-16 by tgingold