ghdl/ghdl #1858: error: type "myrecord" cannot be selected by name

Martoni, 2021-09-10:
**Description**
Can't get property of declared record : 

```vhdl
     type myrecord is record
        un : std_logic;
        deux : std_logic_vector(7 downto 0);
        trois : unsigned(7 downto 0);
    end record myrecord;

   signal vout_reg : std_logic_vector(myrecord.deux'length - 1 downto 0);
```
Elaboration give the following error : 
```console
issue_module.vhd:23:40:error: type "myrecord" cannot be selected by name
    signal vout_reg : std_logic_vector(myrecord.deux'length - 1 downto 0);
```

Vivado can synthesize it, then it should be possible no ?

**How to reproduce?**
This minimal module reproduce the error : 

```vhd :file: issue_module.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.numeric_std.all;

Entity issue_module is
port
(
    clk : in std_logic;
    vout : out std_logic_vector(7 downto 0)
);
end entity;

Architecture issue_module_1 of issue_module is

    type myrecord is record
        un : std_logic;
        deux : std_logic_vector(7 downto 0);
        trois : unsigned(7 downto 0);
    end record myrecord;

    constant COMPTE: myrecord := (un => '1', deux => "00000010", trois => x"03");

    signal vout_reg : std_logic_vector(myrecord.deux'length - 1 downto 0);
begin

    process(clk)
    begin
        if rising_edge(clk) then
            vout_reg <= COMPTE.deux; 
            vout <= vout_reg;
        end if;
    end process;

end architecture issue_module_1;

```

And elaborate:

```sh :image: ghdl/ghdl:bullseye-mcode
$ ghdl -a issue_module.vhd 
issue_module.vhd:23:40:error: type "myrecord" cannot be selected by name
    signal vout_reg : std_logic_vector(myrecord.deux'length - 1 downto 0);
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [*] Built from sources: `commit 054f40e6bd1e20f8f655d0a6beaf5789548ea030`

tgingold, 2021-09-10:
Sorry, but this is not valid VHDL.

You can report this issue to Xilinx...

Martoni, 2021-09-10:
Ok. Thanks

closed 2021-09-10 by Martoni