ghdl/ghdl #1860: Support for entity instantiation in VUnit

Hoffti, 2021-09-11:
**Description**
GHDL crashes if a VUnit contains an entity instantiation.

**Expected behaviour**
GHDL is expected to support that statement but emits a 'GHDL Bug occurred' message instead (see below).

**How to reproduce?**
Put the following code in the respective files and invoke GHDL with the command below.

```vhd :file: test_rtl.vhd
ENTITY test IS
END ENTITY;

ARCHITECTURE rtl OF test IS
BEGIN
END ARCHITECTURE;
```

```vhd :file: test.psl
VUNIT test_formal (test) {
    u_test_formal_test: ENTITY work.test_formal_test(formal);
}
```

```vhd :file: test_formal_test_formal.vhd
ENTITY test_formal_test IS
END ENTITY;

ARCHITECTURE formal OF test_formal_test IS
BEGIN
END ARCHITECTURE;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl --synth --std=08 test_rtl.vhd test.psl test_formal_test_formal.vhd -e test
```

**Context**

- OS: 7.3
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: ae3e6932f7cecfce0dce4777644fc29040886651

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r726.gae3e6932) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-pc-linux-gnu
/home/TIHO101/git_ws/projekte/1034-A-64003A00/viper/source/general_purpose/hdl/
Command line:
ghdl --synth --std=08 test_rtl.vhd test.psl test_formal_test_formal.vhd -e test
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:1622
Call stack traceback locations:
0x7f3ed0 0x58de1e 0x6fa842 0x6d7772 0x6d80b3 0x6fbf27 0x6fc2f8 0x6fc3ca 0x6f47c1 0x72d759 0x650ef8 0x7d9d8e 0x4091d1 0x7f5f01930b33 0x4078d7 0xfffffffffffffffe
******************************************************************
```

**Additional context**

None.

tgingold, 2021-09-11:
Interesting.  What is the use-case for having instances within vunits ?

closed 2021-09-12 by tgingold

HoT-1, 2021-09-13:
> Interesting. What is the use-case for having instances within vunits ?

The use-case is having the capability to easily attach re-usable verification cores (e.g. AXI4) to an implementation.
The verification core is parameterized and can be 'glued' to the RTL using the vunit.

tmeissner, 2021-09-13:
Ha, nice idea. "Binding" a verification component without needing an extra wrapper around the RTL.

commit 55a6aefe (2021-09-12): vhdl-configuration: also consider units instantiated in vunit.  Fix #1860