ghdl/ghdl #1864: Mcode: signal in pkg of integer_vector array causes "gen_insn error: cannot match reg s*i  with stmt OE_ADD"

avelure, 2021-09-14:
**Description**
Using Mcode backend I get a bug report and the error `gen_insn error: cannot match reg s*i  with stmt OE_ADD` when I have a signal in a package declared as an array of any of the new VHDL-2008 vector types real_vector, integer_vector, time_vector.
Changing to a std_logic_vector makes it work.

**Expected behaviour**
No error as in LLVM.

**How to reproduce?**
MWE:
```vhd :file: int_vec_pkgv.vhd
package int_vec_pkg is
  type t_integer_array is array (0 downto 0) of integer_vector(0 downto 0);
  type t_real_array is array (0 downto 0) of real_vector(0 downto 0);
  type t_time_array is array (0 downto 0) of time_vector(0 downto 0);
  signal int_vec : t_integer_array;
  signal real_vec : t_real_array;
  signal time_vec : t_time_array;
end package;
```

```vhd :file: test.vhd
use work.int_vec_pkg.all;

entity test is
end entity test;
architecture beh of test is
begin
  process
  begin
    wait;
  end process;
end architecture beh;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 int_vec_pkg.vhd
ghdl -a --std=08 test.vhd
ghdl --elab-run --std=08 test
```
**Context**
Please, provide the following information:

- OS: win10
- Origin: Nightly mcode i686-w64-mingw32 95cbcc92

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
gen_insn error: cannot match reg s*i  with stmt OE_ADD

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r740.g95cbcc92) [Dunoon edition]
Compiled with GNAT Version: 10.3.0
Target: i686-w64-mingw32
C:\proj\public\minimal\
Command line:
c:\prog\GHDL\nightly-mingw64-llvm\bin\ghdl.exe --elab-run --std=08 test
Exception PROGRAM_ERROR raised
Exception information:
raised PROGRAM_ERROR : ortho_code-x86-insns.adb:336 explicit raise
Call stack traceback locations:
0x997bba 0x999a5b 0x99a0cb 0x99a468 0x99a6ad 0x98e904 0x9a5557 0xc0affb 0xc0a7d8 0xc0a873 0xbf7b01 0xc07019 0xc01463 0xc447cf 0xb92343 0xadf43e 0xc47524 0x8e2cc6 0x8e1384 0x7777fa27 0x77ea7a9c 0x77ea7a6c
******************************************************************
```

tgingold, 2021-09-14:
I can reproduce this issue and I will investigate.

It looks like you are testing many features!

avelure, 2021-09-14:
I have a relative big design under redesign and modernization so I hit cases here and there.
Recently I had some spare time to recheck the compatibility and reduce the MWEs on gcc and mcode, which had not been working on my design previously, but I see it has all been fixed this autumn (on Linux at least), so thanks for all the effort!

tgingold, 2021-09-14:
This should now be fixed.

Thank you for the issue and waiting for the next one...

commit ecd3da07 (2021-09-14): ortho_code-x86-insns: handle OE_ADD R_I_Off + R_I.  Fix #1864