ghdl/ghdl #1867: crash on non-globally static actual expression in VHDL 2008 with generics

wouterscm, 2021-09-15:
**Description**
When assigning an expression to an input port of an instantiated entity with generics, GHDL crashes while running (after elaboration) if the expression is not globally static (which should be fine in VHDL-2008). I compared this issue with [https://github.com/ghdl/ghdl/issues/827](url), but it seems to be a different cause.

**How to reproduce?**

```vhd :file: crash.vhd
library ieee;
use ieee.std_logic_1164.all;

entity one is
    generic (
        PORT_WIDTH : integer
    );
    port (
        INPUT : in std_logic_vector(PORT_WIDTH-1 downto 0)
    );
end entity;

architecture rtl of one is
begin
end architecture;

-----------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std_unsigned.all;

entity two is
end entity;

architecture rtl of two is
    signal ctr : integer range 0 to 7;
begin
    one_inst : entity work.one
    generic map (
        PORT_WIDTH => 3
    )
    port map (
        INPUT => to_slv(ctr, 3)
    );
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 crash.vhd && ghdl -r --std=08 two
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [x] Built from sources: `2fb2384dbcfea0b2b6e6efe00934ccdf869d6ee6`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (tarball) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/mnt/c/Users/chris/Nextcloud/Windows/Hyperion/
Command line:
ghdl -r --std=08 two
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:553
Call stack traceback locations:
0x558726fa2c0b 0x558726fa2feb 0x558726fa3267 0x5587271b4952 0x5587271c3733 0x5587271c37eb 0x5587271ad560 0x55872718287f 0x558727182b37 0x558727183033 0x55872718348d 0x558727190141 0x55872718f636 0x55872715a104 0x55872715a22f 0x55872715a888 0x55872715b01e 0x55872715cbb8 0x55872715cd0d 0x558727160129 0x558727164a0c 0x5587271d294c 0x55872713aa5d 0x558727137bb0 0x558727145cb4 0x5587271d72bb 0x5587271164ed 0x55872704ef8b 0x5587271dabd8 0x558726e05a67 0x7f5ceea100b1 0x558726e041cc 0xfffffffffffffffe
******************************************************************
```

**Additional context**
When replacing the non-static expression with a static one, and remove --std=08, there is no crash.

```vhd :file: no-crash.vhd
library ieee;
use ieee.std_logic_1164.all;

entity one is
    generic (
        PORT_WIDTH : integer
    );
    port (
        INPUT : in std_logic_vector(PORT_WIDTH-1 downto 0)
    );
end entity;

architecture rtl of one is
begin
end architecture;

-----------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity two is
end entity;

architecture rtl of two is
    signal ctr : integer range 0 to 7;
    signal one_input : std_logic_vector(2 downto 0);
begin
    one_inst : entity work.one
    generic map (
        PORT_WIDTH => 3
    )
    port map (
        INPUT => one_input
    );
    one_input <= std_logic_vector(to_unsigned(ctr, 3));
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a no-crash.vhd && ghdl -r two
```

tgingold, 2021-09-15:
This issue has been fixed in the development version.

closed 2021-09-16 by tgingold

umarcor, 2021-09-16:
I can confirm that it works ok with `GHDL 2.0.0-dev (1.0.0.r651.g243bb15e)`.