ghdl/ghdl #1871: Build fails with latest revision: 'use clause for package "Str_Table" has no effect' in vhdl-evaluation.adb

SiliconWizard, 2021-09-17:
Building the latest revision yields:

> vhdl-evaluation.adb:2181:07: warning: use clause for package "Str_Table" has no effect
> vhdl-evaluation.adb:2203:07: warning: use clause for package "Str_Table" has no effect
> gnatmake: "/home/akasa/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/vhdl-evaluation.adb" compilation error
> make: *** [Makefile:203 : ghdl_mcode] Error 4

GNAT 11.1.0

Paebbels, 2021-09-17:
I can confirm this issue on my local machine and GHA also lists the [latest CI run](https://github.com/ghdl/ghdl/actions/runs/1246307367) as failed.

tgingold, 2021-09-17:
My mistake.  Fix in progress.

SiliconWizard, 2021-09-17:
Fixed :)

umarcor, 2021-09-17:
Fast reaction and good job everyone! Open source is awesome when done properly!

closed 2021-09-17 by umarcor