ghdl/ghdl #1875: VITAL port type must be Std_Logic_Vector or Std_Ulogic - VHDL2008

avelure, 2021-09-22:
**Description**
I'm trying to make a vendor library compilable with GHDL with VITAL enabled, however it doesn't parse a port as std_logic_vector when compiling for 2008. 

Looking at vhdl-ieee-vital_timing.adb:263 it seems to be because in 08 std_logic_vector is a subtype of std_ulogic_vector and not a base type anymore. There is potentially a similar issue at line 565.

**Expected behaviour**
Parsed ok like in v93

**How to reproduce?**

```vhd :file: test.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.VITAL_timing.all;
use IEEE.VITAL_primitives.all;

entity test is
  port (
    CLK : in std_logic_vector(1 downto 0)
    );
  attribute VITAL_LEVEL0 of test : entity is TRUE;
end test;

architecture VITAL_ACT of test is
begin
  process
  begin
  wait;
  end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 -frelaxedx test.vhd
```

**Context**
Please, provide the following information:

- OS: Win10
- Origin: Nightly fe9e78d4

```
test_vital.vhd:8:5: VITAL port type must be Std_Logic_Vector or Std_Ulogic
    CLK : in std_logic_vector(1 downto 0)
    ^
test_vital.vhd:13:14: circular reference of entity "test"
architecture VITAL_ACT of test is
             ^
c:\prog\GHDL\nightly-mingw64-llvm\bin\ghdl.exe: compilation error
```

umarcor, 2021-09-22:
I think that VITAL packages were not updated in the last 15-20 years. That is the reason for the license of VITAL not matching the Apache 2.0 of VHDL 2019 libraries. Therefore, this issue might be more complex than it seems, in the sense that VITAL libraries might need to be reviewed and updated for them to be compatible with the active revision of the language.

/cc @JimLewis @Paebbels

JimLewis, 2021-09-22:
VITAL needs an update.   VHDL-2019 could not take it on as it was not proposed in the par to do that.   We need to add this to the issue list and make sure it gets on the PAR when we ask for one.    We should ask for a PAR soon - maybe after we get the draft of the Latex done.

avelure, 2021-09-22:
The IEEE 1076.4 4.3.1 specifies
```The type mark in an entity port declaration shall denote a type or subtype that is declared in package
Std_Logic_1164. The type mark in the declaration of a scalar port shall denote the subtype
Std_Ulogic or a subtype of Std_Ulogic. The type mark in the declaration of an array port shall denote the type
Std_Logic_Vector.```

I agree the VITAL text probably should be updated to reflect the change in 08,  but the I think the bug is only in the syntax checker for VITAL in GHDL, where it assumes the base type of a Std_Logic_Vector is  Std_Logic_Vector, while in 08 it is instead a Std_Ulogic_Vector. If you want to follow the LRM to the dot I guess you need an exception for 08 and check directly against the type. However, the checker is already allowing the types to be subtypes of Std_Logic_Vector which is not explicitly allowed by the standard.

On another note, I also did a diff between the GHDL, mentor and LRM VITAL libraries (except for the vital_timing_p which is missing from the LRM where vital_timing_b is instead repeated twice by error) and there seem to be various bugfixes in both GHDL and Mentor already, and I assume the other vendors have more. In other words, the VITAL library included with GHDL is already not pure.

JimLewis, 2021-09-22:
@avelure   You can find 1076 WG issues at:  https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues   

The 1076 WG can take on this issue.    We would need someone to lead it though.   Interested?

umarcor, 2021-09-23:
> I agree the VITAL text probably should be updated to reflect the change in 08

It's not just a matter of willingness. VITAL packages are "Inactive-Withdrawn" since 2009: https://standards.ieee.org/standard/1076_4-2000.html. "Abandoned" resources should not be used for meaningful tasks. Hence, the LRM compliant solution to this issue is that VITAL cannot be used with VHDL > 2002.

It might be possible to have some non-standard tweaks to use VITAL with VHDL 2008 in GHDL. However, the sensible approach would be to first reactivate the standard, and then have it cleanly supported (not only in GHDL, but by any vendor). Having outdated VITAL in GHDL is quite unpleasant already (from a packaging perspective), because it forces us to have a different GPL build option to be used in projects such as Debian.

> However, the checker is already allowing the types to be subtypes of Std_Logic_Vector which is not explicitly allowed by the standard.

Is this related to `-frelaxed`?

> there seem to be various bugfixes in both GHDL and Mentor already, and I assume the other vendors have more. In other words, the VITAL library included with GHDL is already not pure.

There is a difference between having bugfixes for VITAL libraries to work with the revisions of the standard released at the time VITAL was actively maintained, and implementing modifications to deal with later revisions. Should this issue be a matter of touching a few lines, that might be doable. However, I would refrain from maintaining non-standard variants of the sources for 08 or 19 which are significantly different.

closed 2021-09-23 by tgingold

JimLewis, 2021-09-23:
@umarcor  Do you disagree with subsuming 1076.4 (VITAL) into 1076?    That is at least one of the thoughts.   Also that way the code would automatically become IEEE open source.   If you initiated it as a separate project, then you would need ask to be part of IEEE open source.     This is what happened with 1076.3 (numeric_std + friends) and 1164 (std_logic).

avelure, 2021-09-23:
Thanks for fixint it @tgingold !

@umarcor 
I think there was some confusion, I was not proposing any change to the VITAL VHDL packages included with GHDL, instead the issue is in the parser implemented in GHDL that checks my vendor's code is conforming with the standard. As far as I understand I cannot mix 08 and 02 in GHDL, so being strictly LRM compliant here would be a step backwards compared to the industry I feel.

I agree there should be a new VITAL release and that it is not good to work with a redacted one (however on IEEE explore it is still listed as Active), but looking at the time it took to get 2019 released, I don't see that being completed in a few years.

> There is a difference between having bugfixes for VITAL libraries to work with the revisions of the standard released at the time VITAL was actively maintained, and implementing modifications to [...]

I'm not sure I'm following. The VITAL packages were released as appendixes to the standard, however there are modifications in the ones that are included with GHDL and Mentor. Some of the fixes are pure/impure changes, others are what seems to be bug fixes. I'm not sure what source GHDL had for their VITAL packages, maybe there was some official repository that collected changes. They were modified before the creation of the GIT repo in 2005. There are no changelog changes indicating that the modifications were official. 

08/19 is mostly backwards compatible and the rest is presently handled with -frelaxed without the need to change any sources.

> Is this related to -frelaxed?

No, that is related to pure/impure #716 which should be fixed in a new Vital update. 

@JimLewis 
If by issue you mean to update the IEEE 1076.4 to modern standards, then that seems like a monumental task, though I might be able to participate depending on my workload.

umarcor, 2021-09-23:
@JimLewis, I agree with any of the following:

- Having 1076.4 subsumed into 1076. That is something we MUST do as a community. As you said, that would solve the licensing and compatibility issues "cleanly". However, it's also the task that most effort requires. At least, someone is required for coordination, and it's not a short-term task. Anyway, we should include it in the PAR.
- Having bugfixes applied to the "last" VITAL packages, so that they can still be used with modern simulators using VHDL 93 or 02. This is what Mentor, GHDL and maybe others are doing already. All those fixes should be tracked as bugfixes/LCS in the VASG.
- Implementing non-disruptive tweaks to the simulators so that "last" VITAL packages can be used with VHDL 2008 or 2019, without modifying the VITAL HDL sources. That might be the default, or opt-in as `-frelaxed`.

I do *not* agree with:

- Having variants of the VITAL HDL sources for making them compatible with VHDL 2008 or 2019, without proper coordination with the VASG. However, it is reasonable to develop/implement those in the context of GHDL, and have them later donated to the VASG. In practice, GHDL is the only simulator used in the VASG for CI purposes, so it is the *de facto* reference implementation of the language, as well as the ideal prototyping playground.

> I think there was some confusion, I was not proposing any change to the VITAL VHDL packages included with GHDL, instead the issue is in the parser implemented in GHDL that checks my vendor's code is conforming with the standard.

@avelure, I got that. I was thinking in advance, in the sense that you are trying to use VITAL packages with VHDL 2008 and that is an strict "currently not possible officially, we can only provide best-effort workarounds". You hit this issue for now (which Tristan could fix on the simulator/parser side), but I would expect you to find further and more complicated problems if you go along that path. Don't take me wrong: I do want you to keep pushing in that direction, because as a community it is desirable to have VITAL revived and brought up to date (hopefully to 2019 and 202X). However, since VHDL is not only a language but also an IEEE standard, there are further implications to take into account.

> As far as I understand I cannot mix 08 and 02 in GHDL, so being strictly LRM compliant here would be a step backwards compared to the industry I feel.

While the first half is correct, the second is arguable. GHDL cannot mix multiple revisions of the LRM (yet) and that might be an step behind the industry. However:

- Being strictly compliant with the LRM (up to our capabilities) is a design decision in GHDL and the motivation for many users to rely on it as a serious and robust tool/project. IMHO, that is something we cannot disregard. Naturally, GHDL supports multiple non-standard features, mostly because of bad habits in the industry as a result of vendors abusing their market position instead of following the standards. However, all non-standard behaviours should be explicitly acknowledged and made opt-in to not disturb the default compliance.
- There is a single officially valid VHDL revision at each time. Using VHDL 1987, 1993, 2002 or 2008 nowadays is a non-standard/non-official usage of the language. No vendor/tool is strictly compliant at the moment, for rather obvious reasons. Aldec is probably the one with most 2019 features. Assuming that 2019 might be considered "too new yet", we might analyse 2008 support. GHDL's VHDL 2008 support is comparable to vendors (if not better), and language support for simulation and synthesis in GHDL is much more consistent than vendors. GHDL is (far) ahead of the industry in a value for price, but probably also ahead in terms of absolute value compared to e.g. Xilinx's VHDL support. Very precisely, XSIM cannot handle VUnit because Xilinx did not implement some VHDL 1993 features yet, and JSON-for-VHDL (using VHDL 1993) did need 3h to be synthesised by Vivado 2018.
- It is planned to revisit how GHDL supports multiple revisions of the language (see #980). However, that'd be "GHDL's solution to the problem" because there is no standard approach to make it strictly compatible with vendors. BTW, the same applies to mixed-language designs. Therefore, users which need strict/deterministic behaviour should be careful with mixing revisions in any case.
  - My understanding is that Tristan tweaked the analysis of VITAL libraries in order to handle them as "valid but relaxed 2008". It is not reading them as "VITAL 2002 mixed with 2008 sources".

> I agree there should be a new VITAL release and that it is not good to work with a redacted one (however on IEEE explore it is still listed as Active)

That is an interestich catch!

- https://standards.ieee.org/standard/1076_4-2000.html
- https://ieeexplore.ieee.org/document/954750

Same document, different status.

> but looking at the time it took to get 2019 released, I don't see that being completed in a few years.

That is literally up to us (all the members of the VHDL community), for good and bad. Major vendors decided to ditch VHDL ~15y ago. Unlike SV, which had huge human and economic support (primarily due to UVM being a SV only top-tier feature in most vendor portfolios), VHDL development has not been explicitly funded since the 90s. In the last 2y, it seems that some vendors are showing interest in VHDL again, probably because 70-80% of the european market is still dominated by VHDL (acc [... cut ...]
Jim asked whether you would like to lead all the way. It's ok if you cannot afford that. However, it'd be very desirable if you could share the knowledge you have already.
Furthermore, bugfixes can be accepted in a feature branch of https://gitlab.com/IEEE-P1076/Packages, so we have some "canonical" location for the fixed sources before we go ahead with the formalities.
As a matter of fact, we discussed in the VASG about having more frequent (pre-)releases of the Packages and LRM (maybe every 1-2y) without having all of them made official by IEEE.

JimLewis, 2021-09-24:
@umarcor  @avelure 
My thought is that if we can get this into next IEEE standard, we fix the parts that are causing issues.   

We don't have to address everything at once or even have a single person do it.   Instead, my 
strategy to get it done is to get the packages into an environment that supports incremental
update and empower the reporter to propose a fix (via a pull request) - that way we as issues
are talked about they also get done.   It would also be good to have a test environment so 
that the reporter can both propose a fix and add a test case to the test environment.

avelure, 2021-09-24:
@umarcor 
Thanks for the through summary of the status.

> Jim's entry in the wiki mentions that pure/impure functions need to be changed for them to be compatible with >=2008.
However, I guess that -frelaxed might be hiding other modifications that ought to be applied to the VITAL VHDL sources.

I analysed the sources included in 1076.4. 
There are only pure/impure that GHDL complains about:
  - result subtype of a pure function cannot have access subelements
  - result subtype of a pure function cannot denote an access type
  - pure function "xxx" cannot call (impure) function "yyy"
  - pure function "xxx" cannot call (impure) procedure "yyy"
  - pure subprogram "xxx" cannot call (impure) procedure "yyy"'

`-frelaxed` fixes all of these. There are some warnings on variables in functions hiding ones from the outer scope, though this doesn't seem to be an issue looking at the code.

>  in the sense that you are trying to use VITAL packages with VHDL 2008

GHDL comes with VITAL packages precompiled for 08, so I assumed it would be ok.

>  mostly because of bad habits in the industry as a result of vendors abusing their market position instead of following the standards

I would think this is rather due to user requests and flexibility. The user might have old IPcores they bought that are encrypted and built for v87 and then another one for v93 and they would like to write their TB in v08. If a tool vendor does not facilitate, the user might move to another tool.

> Xilinx did not implement some VHDL 1993 features yet

It is also the youngest simulator, not steered by economics (except maybe that Xilinx now doesn't need to pay license fees to distribute a free version of Mentor with Vivado anymore) and I would think only used seriously in academics, so they might be excused for not following the standard. Cadence on the other hand has no excuse.

> my strategy to get it done is to get the packages into an environment that supports incremental
update and empower the reporter to propose a fix (via a pull request)

Sounds like a good plan.

umarcor, 2021-09-24:
> I would think this is rather due to user requests and flexibility. The user might have old IPcores they bought that are encrypted and built for v87 and then another one for v93 and they would like to write their TB in v08. If a tool vendor does not facilitate, the user might move to another tool.

@avelure, that is the point: using `-frelaxed` is not meant for users to mix different versions of the standard. By doing so you are analysing all the files as VHDL 2008 and accepting non-compliant (read invalid) VHDL code; the 87 and 93 sources are not analysed for compliance with their respective revisions. See http://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-frelaxed:

> *Slightly relax some rules to be compatible with various other simulators or synthesizers*
> *Some code (such as Xilinx packages) have such constructs, which are invalid*
> *This option also relaxes the rules about pure functions. Violations result in warnings instead of errors.*

The fact that you can mix 87, 93 and 08 by using this "trick" is just a very fortunate outcome, not a feature nor something you should rely on. In order to mix language revisions as a feature, #980 is required.

Again, I don't want you to refrain from mixing sources or using VITAL with VHDL 2008. I do want you to do so. I'm just concerned about your expectations, and I'm playing the bad guy role here.

> It is also the youngest simulator, not steered by economics (except maybe that Xilinx now doesn't need to pay license fees to distribute a free version of Mentor with Vivado anymore) and I would [...]

I don't think there is excuse for not supporting a 30y old language revision, or for providing non-standard and uglily patched fixed point packages as they are doing. If academics are using Xilinx's simulator seriously, I would doubt about the seriousness of their work from a (V)HDL point of view. Which is not surprising, because academics are not paid for doing good implementations, but for telling what they might do in hypothetical situations. Disclaimer: I'm in Academia.

avelure, 2021-09-24:
I'm not relying on GHDL anyway, it is mostly for regression testing, my final verification will be in Mentor, so I will see it then if there is an issue.

> If academics are using Xilinx's simulator seriously

I meant academic in the sense of students learning to program VHDL. In higher education I have not seen it in use. Besides, I would not trust the Xilinx sim any less than GHDL, assuming the code is parseable by both tools. Being pedantic on syntax does not necessarily mean the simulation result is more correct.

>I don't think there is excuse for not supporting a 30y old language revision

That the standard is old does not reduce the complexity of implementing it if you are starting out with a new simulator. Especially if there is no money incentive for them to implement the less used parts. There is plenty of things in the standard I have never seen in use.

umarcor, 2021-09-25:
> my final verification will be in Mentor, so I will see it then if there is an issue.

That should be safer with regard to mixing multiple revisions of the standard, because Mentor does explicitly support it. I.e., in case of failure, it's a legit bug on their side.

> I meant academic in the sense of students learning to program VHDL.

Then, their professors or whoever designed their studies are the ones to blame.

> Besides, I would not trust the Xilinx sim any less than GHDL, assuming the code is parseable by both tools.

You should trust GHDL significantly more than Xilinx with regard to VHDL. Parsing is a very minor part of interpreting a VHDL codebase. After parsing, analysis is required, then elaboration. Parsing is not so complex, indeed, that's why most available HDL parsers do that only. The complexity is analysis (precisely semantic analysis) and elaboration. In other words, both GHDL and Xilinx can understand what the VHDL looks like, but not both of them can understand what it means (properly). This is because GHDL is not pedantic with the syntax, but with the LRM (the standard). Being strict with the syntax is just a requirement for being able to implement the standard, because the VHDL syntax is known to be ambiguous. In case of conflict, GHDL tries to understand, vendors used to do their own interpretation and shove it into the codebase.

> Being pedantic on syntax does not necessarily mean the simulation result is more correct.

I beg to disagree. I want to use an standard language so that I can write code once and use it with any vendor that is compliant. I don't want to rewrite the codebase once for each vendor who advertises VHDL support but then decides to implement their own language, which is 99% similar, but not compliant. Deviating from the standard means that simulation and synthesis in Vivado might not match, which is a serious risk. Fortunately, FPGAs nowadays almost never die because of broken bitstreams, so time is lost only.

> That the standard is old does not reduce the complexity of implementing it if you are starting out with a new simulator. Especially if there is no money incentive for them to implement the less used parts. There is plenty of things in the standard I have never seen in use.

I might understand that, if Xilinx was not a +30y old company with an absolute market lead during at least 20y. They decided to ditch what they had and implement a new simulator from scratch, even though that's absolutely out of their expertise. They might have also ditched GCC and implement their own compilers. They did not. Maybe they should think about why they did not do it for C/C++ but they decided to do it for VHDL.
Can't feel sorry for them. If they ever want to collaborate with others instead of double betting into autarchy, they will be very welcome. In fact, RapidWright, fpga-interchange-format, Vitis, etc. are all open source. They are starting to understand that the scope is too large for them to handle alone.

To be specific, last time I checked XSIM could not create/initialise a pointer (in terms of GHDL, create an access): `newp := new vec_t'(0 to length-1 => value);`. I cannot understand why they did not implement that yet, neither can I imagine how long will they take to do it. One of the [most popular](https://larsasplund.github.io/github-facts/verification-practices.html) verification frameworks cannot be used in Vivado because of that. That's a non-negligible user base.

It's legit for them not to have interest in VHDL, I'm not blaming them for that. They are free to not invest into VHDL or into XSIM, if they think that's not profitable for them. However, it is quite offenssive that they keep advertising "improved VHDL support" given that the state is "not even minimally acceptable for any real world project" (something they know because they did the no-so-elegant implementations). Dishonesty is very unwelcome in open source.

commit 565967d1 (2021-09-23): vhdl-ieee-vital_timing.adb: handle vhdl 2008.  Fix #1875