ghdl/ghdl #1880: Documentaion on ghdl.readthedocs.io seems outdated

abyszuk, 2021-09-26:
**Description**
There's a discrepancy between documentation content available on
https://ghdl.github.io/ghdl/ and https://ghdl.readthedocs.io/en/latest/
The readthedocs.io version looks like it stopped being updated in 2020.

This is confusing because the http://ghdl.free.fr/ webpage still points to the old version.
Google results for "ghdl documentation" also point mostly to the outdated Read the Docs version, in my case the github.io link appears on the second results page.

Looking at the [RTD build log](https://readthedocs.org/api/v2/build/14809507.txt) looks like it still uses Python 2 and can't find sufficiently new version of `pydecor`:
```
ERROR: Could not find a version that satisfies the requirement pydecor>=2.0.1 (from -r doc/../pyGHDL/libghdl/requirements.txt (line 1)) (from versions: 0.1.0.dev1, 0.1.0.dev2, 0.1.0.dev3, 0.1.0, 1.0.0, 1.1.0, 1.1.1, 1.1.2, 1.1.3)
ERROR: No matching distribution found for pydecor>=2.0.1 (from -r doc/../pyGHDL/libghdl/requirements.txt (line 1))
```

**Expected behaviour**
Keep both pages in sync or maybe take down the old version completely to avoid confusion.

**How to reproduce?**
Just open the webpage ;-)

umarcor, 2021-09-26:
There is branch https://github.com/ghdl/ghdl/tree/doc-RTD, which should be the only one built and served in the RTD project, so that users are properly redirected to ghdl.github.io/ghdl.
@Paebbels, can you please review the settings of the RTD project: https://readthedocs.org/projects/ghdl/. Some weeks ago, we enabled branch doc-RTD, but setting it as the default (latest) and disabling others is pending.

Ref: https://github.com/ghdl/ghdl/issues/1839#issuecomment-899104041

umarcor, 2021-10-16:
Some days ago, I reviewed the settings of RTD with @Paebbels. Now a single version is built and shown in https://ghdl.readthedocs.io/en/latest/, which explains the readers to go to ghdl.github.io/ghdl.

closed 2021-10-16 by umarcor