ghdl/ghdl #1882: Use local GCC as CC linker in Docker images

abyszuk, 2021-09-30:
**Description**
This issue is strongly connected to [code coverage issues](https://github.com/ghdl/docker/issues/42) I've noticed in the GHDL Docker images since beginning in August.
I originally thought that this is somehow a cointainer problem caused by switching to newer Debian version, but now I think the root cause is commit b72f46a0 introduced because of #1629 

Quick recap of discussion from docker repo:
There are two gcc binaries in the GCC backend images:
- `/usr/bin/gcc` - system GCC
- `/usr/local/bin/gcc` - GCC used to build GHDL

Additionally there's a CC symlink for system GCC `/usr/bin/cc -> /usr/bin/gcc`, but there's **no symlink** for `/usr/local/bin/cc`.
Because of `PATH` priorities default GCC version (as reported by `which gcc`) is local GHDL version in `/usr/local/bin/gcc`.
Before commit b72f46a0 this binary was called both for compilation and linking.
Since this change, `/usr/bin/cc` is used for linking and `/usr/local/bin/gcc` for compilation.
As it turns out, using different GCC versions for compilation and linking breaks code coverage.

**Expected behaviour**
Code coverage works in GHDL-gcc Docker images.
I think this can be done by simply adding `/usr/local/bin/cc -> /usr/local/bin/gcc` symlink in Docker images. No need to modify GHDL itself.
I wonder if this will show up not only in Docker images, but also in local installations.

**How to reproduce?**
Try using code coverage feature in Docker images with GCC backend.

tgingold, 2021-09-30:
You can also set the environment variable `CC` to `/usr/local/bin/gcc`.
(Or use `--LINKER=/usr/local/bin/gcc`)

abyszuk, 2021-09-30:
That's true. In case of env variable I think the proper fix would be to modify `ghdl:run` images. If I understood the CI flow properly that's currently root of other images.
I would be against `--LINKER` as a final solution. This will require additional knowledge from users about which linker to use and why.

abyszuk, 2021-09-30:
I can test this tomorrow, but I think this commit also broke traditional, local installations.
Maybe the best solution would be simply that `make install` would also create `cc -> gcc` symlink?
I can create PR for that.

tgingold, 2021-10-01:
No, `make install` should install `ghdl` binaries, not modify your C compiler.

The simplest solution for this issue is to slightly change the docker image.

abyszuk, 2021-10-01:
OK, so Docker it is :-)
Thanks for quick feedback!

closed 2021-10-01 by abyszuk

umarcor, 2021-10-02:
Some days ago, Tristan added an enhancement for honoring envvar `CC`: https://github.com/ghdl/ghdl/commit/a076bcb8121ea03b626447f5dace625415697600
That's used in extended tests, to avoid the powershell jobs from failing: https://github.com/ghdl/extended-tests/commit/f9e33be42b7c22ad331eae43b6531c62072f3bd8
I now followed the same approach with the docker images: https://github.com/ghdl/docker/commit/3fd9453fe994702641ebca28b80335887e8a1878