ghdl/ghdl #1893: Analysis doesn't always include external libraries

abyszuk, 2021-10-15:
**Description**
I first spotted this problem because I'm using `ghdl-ls`, but after some poking I think it's a problem with GHDL analysis stage itself.
I think there's more than one issue here, but I'd like to start with the most basic one.
If a file constains more than one design unit, say `module_pkg` and `module`, and if uses some external libraries then GHDL will recognize only some uses of this library.

**Expected behaviour**
Analysis properly recognizes all uses of a library for all design units in a file.

**How to reproduce?**
Analyse this file with following command (you need to have Vunit library available):
`ghdl -a -v --std=08 -Pvunit_out/ghdl/libraries/vunit_lib mwe.vhd`

```vhd :file: mwe.vhd
library ieee;
use ieee.std_logic_1164.all;
library vunit_lib;
context vunit_lib.vc_context;  -- <=== this fails with 'unit "vc_context" not found in library "vunit_lib"

package mwe_pkg is
  constant c_wbm_handle : bus_master_t := new_bus(data_length => 32, address_length => 32);
end package;

------------------------------------------------------
library ieee;
use ieee.std_logic_1164.all;

use ieee.numeric_std.all;
use ieee.math_real.all;

library vunit_lib;
context vunit_lib.vunit_context;  -- <=== this is OK

entity mwe is
end mwe;

architecture arch of mwe is

begin

end arch;
```

**Context**
- OS: Centos 8
- Origin:
  -  Built from sources: `348629d` with `mcode` backend

**Additional context**
There are a few other variants that can trigger this, but I still have problem designing proper MWEs for them.

abyszuk, 2021-10-16:
Yes, I'm sure.
I've just double checked this - if I remove the upper part of MWE (the one about package), then the lower part passes without any errors or warnings.

tgingold, 2021-10-16:
I am not sure to understand well this issue.
If you keep only the first part, does it analyze without error ?

Alternatively, it would help if you could provide a standalone reproducer (which will be composed of multiple files).  I am not sure how vunit should be configured/installed to reproduce your issue.

umarcor, 2021-10-16:
The VUnit user_guide example does not include/build verification components, because VUnit does not add them by default. You need to use `VU.add_verification_components()` in the `run.py` script (as in e.g. https://github.com/VUnit/vunit/blob/master/examples/vhdl/array_axis_vcs/run.py). However, vunit_context is always included. That's why `context vunit_lib.vunit_context` is ok but `context vunit_lib.vc_context` fails.