ghdl/ghdl #1904: `warning: static expression violates bounds` should be an error ?

NicoPy, 2021-10-30:
**Description**
With the following code, ghdl outputs a warning at elaboration step.
In a project where ghdl is used with cocotb, I did not see the warning since cocotb outputs a lot of messages. I run all my tests without noticing the warning. Luckily, the Xilinx synthesizer outputs an error instead of a warning.

**Expected behaviour**
I think ghdl should output an error instead of a warning.

**How to reproduce?**
Use the following code :
```vhd :file: integer_init_test.vhd
library ieee;
use ieee.std_logic_1164.all;

entity Integer_init_test is
end entity Integer_init_test;

architecture RTL of Integer_init_test is

subtype t_Counter is Natural range 16 to 32;
signal Counter : t_Counter := 0;

begin

    process
    begin
        wait;
    end process;

end architecture RTL;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a integer_init_test.vhd
```
I get the following output.
```batch
integer_init_test.vhd:11:31:warning: static expression violates bounds [-Wruntime-error]
signal Counter : t_Counter := 0;
                              ^
```

**Context**

- OS: Windows 10
- Origin: MSYS2 nightly package
```batch
$ ghdl -v                                                                      
GHDL 2.0.0-dev (1.0.0.r840.g5665fedb) [Dunoon edition]
 Compiled with GNAT Version: 11.2.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

Paebbels, 2021-10-31:
I suspect the language defines this only as elaboration error, but not an analysis error, therefore GHDL gives you an early warning that is will raise an elaboration error.

tgingold, 2021-10-31:
That's correct.
The error will be raised during elaboration

Paebbels, 2021-10-31:
@tgingold how can he make this warning an error?

tgingold, 2021-10-31:
You can use `-Werror` or `-Werror=runtime-error` but it is not recommended.

NicoPy, 2021-10-31:
Hmmm... I have to investigate further and modify my test bench since in my actual project I made all my simulations without any error (but the initialization error was there).

Paebbels, 2021-11-01:
@tgingold is it possible to add a warning subclass (or id) for `static expression violates bounds`. In such a case, this single warning could be promoted to an error. `-Wruntime-error` would behave like a super-switch, it would be a little bit like `-Wall` enables a bunch of switches in GCC.

I think a user-defined promotion of `static expression violates bounds` would create a benefit, because it can same lot's of analyze time, as it is already clear an elaboration will fail.

tgingold, 2021-11-01:
-Wruntime-error is about these kinds of checks.  But it could trigger false warnings if the code is never executed.
That's the reason why it cannot be an error.

closed 2021-11-03 by tgingold