ghdl/ghdl #1906: What LLVM versions to support?

Paebbels, 2021-11-01:
In a short timeframe, LLVM advanced to version 13.  
GHDL still has code for LLVM 3.5 in it, right?

My questions are:
* What LLVM version do we need to support?
* What LLVM versions are offered by e.g. Debian 10+ or Ubuntu 2010.04+?
* Can we drop some version and maybe reduce maintenance work?

--------------
/cc @tgingold, @umarcor

umarcor, 2021-11-01:
I don't know whether dropping older versions will reduce maintenance work. I believe that effort is required for newer versions, but old ones just keep working.
IMHO, we should drop LLVM versions if they don't support debugging, or in case we add coverage and they don't support that.

tgingold, 2021-11-03:
Yes, maintainance work is not high (currently).  If it requires more work, the oldest versions will be removed (in particular 3.5)

closed 2021-11-03 by tgingold