ghdl/ghdl #1911: Regression: Null range vectors?

stnolting, 2021-11-13:
**Description**
The GHDL (+yosys) based implementation workflow of the [NEORV32](https://github.com/stnolting/neorv32) master branch fails since ~7 hour ago.

**How to reproduce?**
All implementation task fail, for example `make -C setups/osflow BOARD=UPduino MinimalBoot`. The full console log for this configuration can be found here: https://github.com/stnolting/neorv32/runs/4198423846?check_suite_focus=true

**GHDL Bug occurred**

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-859-g1f3b1857) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/github/workspace/setups/osflow/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
******************************************************************
```

**Additional context**

The last error message before GHDL crashes is:
```
synth_name: cannot handle IIR_KIND_SELECTED_ELEMENT (../../rtl/core/neorv32_fifo.vhd:127:43)
```

The message references this code line:
(see https://github.com/stnolting/neorv32/blob/7ae64d7d95973fbf77ed0a5cd602aa3105219103/rtl/core/neorv32_fifo.vhd#L127)
```vhdl
fifo.match <= '1' when (fifo.r_pnt(fifo.r_pnt'left-1 downto 0) = fifo.w_pnt(fifo.w_pnt'left-1 downto 0)) or (FIFO_DEPTH = 1) else '0';
```

`fifo.r_pnt` and `fifo.w_pnt` might be defined as `std_ulogic_vector(0 downto 0)` here so there might be a comparison of two null range vectors. This specific part of the code has not bee modified for quite some time. 7 hours ago everything still worked fine.

umarcor, 2021-11-13:
Reproducible with `GHDL release: 2.0.0-dev (v1.0.0-864-g5df8168b) [Dunoon edition]` as well: https://github.com/stnolting/neorv32/runs/4200189784?check_suite_focus=true#step:4:109

closed 2021-11-13 by tgingold

umarcor, 2021-11-13:
Thanks @tgingold!

@stnolting, this will take ~2h to run CI here and to update the docker images (in ghdl/docker, then in stnolting/neorv32). I'll take care of babysitting and restarting CI in NEORV32.

stnolting, 2021-11-13:
Thanks for fixing!

> I'll take care of babysitting and restarting CI in NEORV32.

@umarcor Thank you :wink:

umarcor, 2021-11-13:
I restarted CI in NEORV32, after letting the images update. Unfortunately, it seems that the issue is not fixed: https://github.com/stnolting/neorv32/runs/4200680998?check_suite_focus=true. Maybe it is related to the values of the generics?

@stnolting, would you mind copying line https://github.com/ghdl/ghdl/commit/e3c865e88bd12919a65e7230ce4b9772ea58d004#diff-63ad2b89a7d1da94733e55373af9e8c0a50f39488e0b851ceb13f56ae750c133R5 in the test and modifying it to make it crash?

stnolting, 2021-11-13:
I haven't figured out yet which actual instance/configuration of the FIFO causes the problem... However, I think the following configuration might be the most self-evident one:
```
FIFO_DEPTH=1
FIFO_WIDTH=32
FIFO_RSYNC=false
FIFO_SAFE=false
```

`FIFO_DEPTH=1` is the most relevant here as this leads to null-range comparisons.

umarcor, 2021-11-14:
My bad. An update was missing in the chain of images. It had updated GHDL, but 4 commits behind `master`. Now, with properly updated containers, everything is green: https://github.com/stnolting/neorv32/actions

commit d4e80643 (2021-11-13): synth: add exec_name_subtype.  Fix #1911