ghdl/ghdl #1912: Recursive instantiation of component makes ghdl crash on assertion

marzoul, 2021-11-15:
**Description**
Synthesis of recursive instantiation leads to GHDL crashing on an assertion.
See the "GHDL Bug occurred" section at end of this message.

**Expected behaviour**
GHDL should not crash because the code is synthesizable.
At least, in case there is a typo somewhere, GHDL should provide a reason and point to a line of input VHDL code.

**How to reproduce?**
Below is a reduced reproducer code, based on what was originally a popcount synthesizable code.
It used to work well, simulate and synthesize, but I have not used it in recent months.
It uses recursive instantiation of component, with a declaration of the local entity as a component, just because that is the way it was coded originally.

If the component declaration is removed, and the component instantiation is replaced by entity instantiation, then no chash happens and the code is synthesized.

```vhd :file: popcount.vhd
-- Reduced code to exhibit a GHDL crash
-- raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : elab-vhdl_insts.adb:270

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.all;

entity popcount is
	generic (
		SIZE  : natural := 32
	);
	port (
		-- Input data
		data : in std_logic_vector(SIZE-1 downto 0);
		-- Results
		num  : out std_logic
	);
end entity;

architecture synth of popcount is

	-- Component declaration of itself, for recursive instantiation

	component popcount is
		generic (
			SIZE  : natural := 32
		);
		port (
			-- Input data
			data : in std_logic_vector(SIZE-1 downto 0);
			-- Results
			num  : out std_logic
		);
	end component;

begin

	n5 : if SIZE <= 5 generate

		num <= '1';

	end generate;

	more : if SIZE >= 6 generate

		signal num_1 : std_logic;
		signal num_2 : std_logic;

		constant SIZE_LOWER : natural := 4 * (SIZE / 5);

	begin

		inst1 : popcount
			generic map (
				SIZE  => SIZE_LOWER
			)
			port map (
				data => data(SIZE_LOWER-1 downto 0),
				num  => num_1
			);

		inst2 : popcount
			generic map (
				SIZE  => SIZE - SIZE_LOWER
			)
			port map (
				data => data(SIZE-1 downto SIZE_LOWER),
				num  => num_2
			);

		num <= num_1 or num_2;

	end generate;

end architecture;
```

To compile:

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 popcount.vhd
ghdl --synth --std=08 -gSIZE=11 popcount
```

**Context**
Please, provide the following information:

- OS: Archlinux
- Origin: Built from sources `commit e3c865e88bd12919a65e7230ce4b9772ea58d004` (nov 13th, 2021)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r870.ge3c865e88) [Dunoon edition]
Compiled with GNAT Version: 11.1.0
Target: x86_64-pc-linux-gnu
/home/adrien/Dev/C/cellver/fpga/yosys/
Command line:
ghdl --synth --std=08 -gSIZE=11 popcount
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : elab-vhdl_insts.adb:270
Call stack traceback locations:
0x558fcd5e1f72 0x558fcd4d9d1e 0x558fcd4d8fc3 0x558fcd4d923d 0x558fcd4d92be 0x558fcd4da2b1 0x558fcd4da70b 0x558fcd4d9204 0x558fcd4d92be 0x558fcd4d8b23 0x558fcd4d8fef 0x558fcd4d923d 0x558fcd4d92be 0x558fcd4da2b1 0x558fcd4daba8 0x558fcd5076ae 0x558fcd419b44 0x558fcd5c5d77 0x558fcd1a9226 0x7f6fd45ceb23 0x558fcd1a784c 0xfffffffffffffffe
******************************************************************
```

umarcor, 2021-11-15:
Interestingly, on MSYS2 (MINGW64), I don't get a crash:

```
# ghdl version
GHDL 2.0.0-dev (1.0.0.r870.ge3c865e8) [Dunoon edition]
 Compiled with GNAT Version: 11.2.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO   
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.      

# ghdl -a popcount.vhd
popcount.vhd:27:19:warning: declaration of "popcount" hides entity "popcount" [-Whide]
        component popcount is
                  ^

# ghdl --synth --std=08 -gSIZE=11 popcount
C:\msys64\mingw64\bin\ghdl.exe: cannot find entity or configuration popcount 
```

tgingold, 2021-11-15:
I was able to reproduce the issue.  You are simply analyzing using the default standard (93) but synthesizing with --std=08
(Hence unit not found message).

umarcor, 2021-11-15:
@tgingold, correct. I updated my message a few minutes after posting the initial comment. However, that was not sent through e-mail.

closed 2021-11-16 by tgingold

marzoul, 2021-11-16:
The fix works. Thank you :-)