ghdl/ghdl #1916: ieee.math_real numeric accuracy disappointing

mfro0, 2021-11-16:
I know it's not ghdl's fault (but rather the weak IEEE implementation's), but the numeric accuracy of `math_real` is not to the abilities of ghdl (even after the recent replacement with the latest VHDL 2019 sources).

`report "2**32=" & to_string(2.0 ** 32.0, "%20.15f") severity note;`

`140:9:@0ms:(report note): 2**32=4294967295.999997615814209`

IEEE 754 can do (much) better. Modelsim, for example, is to the point:

`** Note: 2**32=4294967296.000000000000000`

(this is only a simple example, other numeric functions lack accuracy even worse, to the point getting unusable, e.g. to convert real integral numbers into >32 Bit signed/unsigned constants to circumvent integer limits).

`openieee.math_real` appears attempt to fix that, but is unfinished?

I quickly hacked together a `math_real` replacement at https://github.com/mfro0/math_real.git that maps all `math_real` functions to VHPI calls "upgrading" ghdl `math_real` accuracy to `modelsim`'s standard.

If there is interest, I would offer to contribute to ghdl, but I'm unsure what it needs to meet the requirements (proper error checking, for example, is still missing).

Would appreciate any feedback/pointers. Thank you!

tgingold, 2021-11-16:
I think that math_real from openieee is complete.  But the complex package is not implemented.
Did you spotted a missing function ?

I am not sure what to do.  Ideally, there could be a switch to use one version or the other.
But, math_real is supposed to be the reference, so it's inaccuracy is what the standard wants!

mfro0, 2021-11-17:
> I think that math_real from openieee is complete. But the complex package is not implemented. Did you spotted a missing function ? 

I apologize; didn't look into openiee deep enough and oversaw that some VHPI calls are already implemented in the package itself and not in the package body. It indeed appears pretty complete.

>I am not sure what to do. Ideally, there could be a switch to use one version or the other. But, math_real is supposed to be the reference, so it's inaccuracy is what the standard wants!

I fully understand ghdl primarily aims to standards compliance and that is of course perfectly fine. Anyway, if somebody (like me) wants/needs to have better accuracy, my github repository is there ;).

Thank you (for this answer and for ghdl in general) and keep up the good work!

closed 2021-11-24 by tgingold