ghdl/ghdl #1917: Item of array is not assigned in the concurrent signal assignment if another item is assigned in the sequential process.

m-kru, 2021-11-18:
**Description**
I am not 100% sure whether this is a bug, or maybe this is inline with the VHDL standard.

**Expected behaviour**
Items assigned in the concurrent signal assignment should be updated. Now, they are stuck to `'U'`. They are simply never updated.

**How to reproduce?**
This is not a working example, but it presents the concept and what is going on.

```vhd
library ieee;
   use ieee.std_logic_1164.all;

package main_pkg is

   type t_add is record
      a : std_logic_vector(15 downto 0);
      b : std_logic_vector(15 downto 0);
      stb : std_logic;
   end record;

end package;

library ieee;
   use ieee.std_logic_1164.all;
   use ieee.numeric_std.all;

library general_cores;
   use general_cores.wishbone_pkg.all;

library work;
   use work.wbfbd.all;

use work.main_pkg.all;

entity main is
   generic (
      G_REGISTERED : boolean := true
   );
   port (
      clk_i : in std_logic;
      rst_i : in std_logic;
      slave_i : in  t_wishbone_slave_in_array (1 - 1 downto 0);
      slave_o : out t_wishbone_slave_out_array(1 - 1 downto 0);
      add_o : out t_add;
      result_i : in std_logic_vector(17 - 1 downto 0)
   );
end entity;

architecture rtl of main is

   constant C_ADDRESSES : t_wishbone_address_array(0 downto 0) := (0 => "00000000000000000000000000000000");
   constant C_MASKS     : t_wishbone_address_array(0 downto 0) := (0 => "00000000000000000000000000000000");

   constant C_REGISTER_UNINITIALIZED_VALUE : std_logic_vector(32 - 1 downto 0) := (others => 'U');
   subtype t_register_vector is t_slv_vector(4 - 1 downto 0)(32 - 1 downto 0);
   signal registers : t_register_vector := (others => C_REGISTER_UNINITIALIZED_VALUE);

   signal internal_master_out : t_wishbone_master_out;
   signal internal_master_in  : t_wishbone_master_in;

begin

   crossbar: entity general_cores.xwb_crossbar
   generic map (
      G_NUM_MASTERS => 1,
      G_NUM_SLAVES  => 0 + 1,
      G_REGISTERED  => G_REGISTERED,
      G_ADDRESS     => C_ADDRESSES,
      G_MASK        => C_MASKS
   )
   port map (
      clk_sys_i   => clk_i,
      rst_n_i     => not rst_i,
      slave_i     => slave_i,
      slave_o     => slave_o,
      master_i(0) => internal_master_in,
      master_o(0) => internal_master_out
   );

   -- Statuses Routing
   registers(3)(16 downto 0) <= result_i(16 downto 0);
   registers(0)(31 downto 0) <= x"deadbeef";
   registers(1)(31 downto 0) <= x"61961bf7";

   -- Funcs Routing
   add_o.a <= registers(2)(15 downto 0);
   add_o.b <= registers(2)(31 downto 16);

   register_access : process (clk_i) is
      variable internal_addr : natural range 0 to 4 - 1;
   begin
      if rising_edge(clk_i) then
         -- Normal operation.
         internal_master_in.rty <= '0';
         internal_master_in.ack <= '0';
         internal_master_in.err <= '0';

         -- Funcs Strobes Clear
         add_o.stb <= '0';

         transfer : if
            internal_master_out.cyc = '1'
            and internal_master_out.stb = '1'
            and internal_master_in.err = '0'
            and internal_master_in.rty = '0'
            and internal_master_in.ack = '0'
         then
            internal_addr := to_integer(unsigned(internal_master_out.adr(2 - 1 downto 0)));

            -- First assume there is some kind of error.
            -- For example internal address is invalid or there is a try to write status.
            internal_master_in.err <= '1';
            -- '0' for security reasons, '-' can lead to the information leak.
            internal_master_in.dat <= (others => '0');
            internal_master_in.ack <= '0';

            -- Statuses Access
            result : if internal_addr = 3 then
               internal_master_in.dat(16 downto 0) <= registers(internal_addr)(16 downto 0);

               if internal_master_out.we = '0' then
                  internal_master_in.ack <= '1';
                  internal_master_in.err <= '0';
               end if;
            end if;

            x_uuid_x : if internal_addr = 0 then
               internal_master_in.dat(31 downto 0) <= registers(internal_addr)(31 downto 0);

               if internal_master_out.we = '0' then
                  internal_master_in.ack <= '1';
                  internal_master_in.err <= '0';
               end if;
            end if;

            x_timestamp_x : if internal_addr = 1 then
               internal_master_in.dat(31 downto 0) <= registers(internal_addr)(31 downto 0);

               if internal_master_out.we = '0' then
                  internal_master_in.ack <= '1';
                  internal_master_in.err <= '0';
               end if;
            end if;

            -- Funcs Access
            add_a : if internal_addr = 2 then
               if internal_master_out.we = '1' then
                  registers(internal_addr)(15 downto 0) <= internal_master_out.dat(15 downto 0);
                  internal_master_in.ack <= '1';
                  internal_master_in.err <= '0';
               end if;
            end if;

            add_b : if internal_addr = 2 then
               if internal_master_out.we = '1' then
                  registers(internal_addr)(31 downto 16) <= internal_master_out.dat(31 downto 16);
                  internal_master_in.ack <= '1';
                  internal_master_in.err <= '0';
               end if;
            end if;

            -- Funcs Strobes Set
            add_stb : if internal_addr = 2 then
               if internal_master_out.we = '1' then
                  add_o.stb <= '1';
               end if;
            end if;

         end if transfer;

         if rst_i = '1' then
            internal_master_in <= C_DUMMY_WB_MASTER_IN;
         end if;
      end if;
   end process;

end architecture;
```

**Context**
- OS: `Linux 5.11.0-40-generic #44~20.04.2-Ubuntu`
- GHDL version: `GHDL 2.0.0-dev (1.0.0.r889.g29b56efcb) [Dunoon edition]`

**Additional context**
There are is an array of `registers` modeled with `signal`. The registers are accessible via Wishbone. Each register (or more precisely each bit of each register) can be either:
1. Written in the concurrent part and read in the sequential part.
2. Written or read in the sequential part and read in the concurrent part.

It implies that given bit is always written only in one of the processes (concurrent xor sequential). However, if any of the registers is written in the sequential process, none of the registers assigned in the concurrent part is ever updated, they are always stuck to `'U'`, even `registers(0)` and `registers(1)` that have constant values.
[image]

If I comment out all assignments to the `registers()` in the sequential process, the assignments in the concurrent part behave as expected.
[image]

tgingold, 2021-11-18:
ghdl is correct here as you have multiple drivers for registers 0 to 3.
In your case, you'd better to use a single process.

m-kru, 2021-11-19:
@tgingold What is meant by the "driver" in such case? I thought that process drives the signal when it changes its value. Registers 0, 1, 3 are never set in the sequential process and register 2 is never set in the concurrent part. Does the VHDL standard impose that a whole array is treated as a "single" signal?

tgingold, 2021-11-19:
The rule is more complex than what you have said, but in this case your conclusion is true: there are drivers for registers 0-3 in the process and also for the concurrent assignments.

closed 2021-11-24 by tgingold