ghdl/ghdl #1918: Bug when using recursive functions and nested records containing unconstrained arrays

bewimm, 2021-11-19:
_Originally posted by @bewimm in https://github.com/ghdl/ghdl/issues/1914#issuecomment-973295334_

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-889-g29b56efc) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/user/Desktop/ghdl_bug/
Command line:
/home/user/Desktop/ghdl-gha-ubuntu-20.04-gcc/libexec/gcc/x86_64-pc-linux-gnu/9.3.0/ghdl1 --ghdl--std=08 --ghdl-frelaxed-rules -P/home/user/Desktop/ghdl-gha-ubuntu-20.04-gcc/lib/ghdl/ieee/v08/ -P/home/user/Desktop/ghdl-gha-ubuntu-20.04-gcc/lib/ghdl/ -quiet -o test_pkg.s test_pkg.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap3.adb:60
Call stack traceback locations:
0x7f1d0a449ae2 0x75f2d3 0x7649cf 0x764af1 0x75ba65 0x75c73b 0x75e84c 0x7a3669 0x77984b 0x7a353d 0x7a7f59 0x7804de 0x77f715 0x77f7b9 0x77fa97 0x7819dd 0x76dbca 0x797493 0x76ecfa 0x74fbf8 0x74151b 0x73ee38 0x64da58 0xba486d 0x64424e 0x652c2e 0x7b1329 0x6420dd 0x7f1d09f25fce 0x7f1d09f2607b 0x64901c 0xfffffffffffffffe
******************************************************************

Execution of /home/user/Desktop/ghdl-gha-ubuntu-20.04-gcc/libexec/gcc/x86_64-pc-linux-gnu/9.3.0/ghdl1 terminated by unhandled exception
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap3.adb:60
Call stack traceback locations:
0x7f1d0a449ae2 0x75f2d3 0x7649cf 0x764af1 0x75ba65 0x75c73b 0x75e84c 0x7a3669 0x77984b 0x7a353d 0x7a7f59 0x7804de 0x77f715 0x77f7b9 0x77fa97 0x7819dd 0x76dbca 0x797493 0x76ecfa 0x74fbf8 0x74151b 0x73ee38 0x64da58 0xba486d 0x64424e 0x652c2e 0x7b1329 0x6420dd 0x7f1d09f25fce 0x7f1d09f2607b 0x64901c 0xfffffffffffffffe
ghdl:error: compilation error
```

Here is the code to reproduce it. This was made by stripping down a bigger package, so it's not really meaningful as is.
In any case ModelSim accepts it.

```
package test_pkg is

	type value_t is record
		value : natural;
	end record;

	type value_array_t is array(natural range<>) of value_t;

	type state_t is record
		values : value_array_t;
	end record;

	type state_array_t is array(natural range<>) of state_t;
	
end test_pkg;

package body test_pkg is

	function to_string(prefix : string; val : state_array_t) return string is
	begin
		if val'length = 0 then
			return "";
		else
			return to_string(prefix, val(val'low+1 to val'high));
		end if;
	end function;

end package body test_pkg;
```

dalex78, 2026-08-20:
# Issue #1918 — Crash: recursive function on nested unconstrained-array records

Analysis by an AI agent.

## High-level summary

A recursive function (`to_string`) operating on `state_array_t` (array of `state_t`, a record containing `value_array_t`, itself an unconstrained array of records) crashed the gcc-backend translator with an internal `ASSERT_FAILURE` at `trans-chap3.adb:60`.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master`, on both mcode and the gcc backend (prebuilt at this repo's exact `master` commit, matching the backend originally reported): both analyze the design cleanly with `-frelaxed-rules --std=08`, no crash. This must have been fixed as part of general improvements to nested-unconstrained-composite-type handling since this was reported (originally against a 2021 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/gna/issue1918/` (the exact repro from the report) whose `testsuite.sh` analyzes the design and expects success. This locks in the current, correct behavior as a regression guard.

dalex78, 2026-08-24:
@tgingold should this ticket be closed ? (PR merged)

closed 2026-08-24 by tgingold