ghdl/ghdl #1919: possibly invalid line causes fatal error - cannot handle IIR_KIND_FLOATING_TYPE_DEFINITION

Juyas, 2021-11-20:
**Description**
A line of code caused a fatal error - even though the line might not be valid.

**Expected behaviour**
Tell the user what the error is, at least the line or anything helpful.

**How to reproduce?**
Use a statement like below should be sufficient.

```vhd :file: ent.vhd
alias reg_acc : std_logic_vector(3 downto 0) is "0111";
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -s file.vhdl
```

- OS: Windows 10

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
finish_sem_name_1: cannot handle IIR_KIND_FLOATING_TYPE_DEFINITION (*std_standard*:1:1)
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2017 (20170515-63)
Target: i686-pc-mingw32
In directory: ***************** somewhere *************
Command line:
***********\Ghdl\Bin\ghdl.exe -s cu.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:246
******************************************************************
```

umarcor, 2021-11-20:
@Juyas, v0.35 is too old. Please, try a more recent version of GHDL. This is the result I get:

```
# ghdl -s file.vhdl 
file.vhdl:1:1: missing entity, architecture, package or configuration
alias reg_acc : std_logic_vector(3 downto 0) is "0111";
^
```

diogratia, 2021-11-20:
(And that's a declaration, not a statement.)

tgingold, 2021-11-21:
I can reproduce the crash if the reproducer is completed.  Fix in progress.

commit db8354e4 (2021-11-22): vhdl-sem_decls: avoid a crash on invalid alias name.  Fix #1919
closed 2021-11-22 by tgingold