ghdl/ghdl #1921: no function declarations for operator

Tonigau, 2021-11-23:
**Description**
I get the following error...
PWMxPhDT.vhd:43:26: no function declarations for operator "*"

with either of following...
Half_Duty_New <= (Duty_in * period) / (2 ** n_bits) / 2;

Temporary workaround...  (hard code value just to get overall success) 
Half_Duty_New <= (128 * period) / (2 ** n_bits) / 2;

I can compile success in quartus19, but maybe related to following error in GHDL ?
--USE ieee.std_logic_unsigned.all; -- Error primary unit "std_logic_unsigned" not found in library "ieee"

Command line... 
** PWMxPhDT > Analyze main module **
$ C:\Programs\UMH\ghdl-0.33\bin\ghdl -a --workdir=d:\Data\CPLD\GHDL\work.ghdl PWMxPhDT.vhd
Failed (98 ms.)

**Expected behaviour**
Not this error.

Its very likely an issue with my code, I have only been learning VHDL for a week
My TestBench is working, (but I dont think I can access internal signals to display is GTKwave for debugging)

OS=win7P x64
```
Code
--------------------------------------------------------------------------------
LIBRARY ieee;
USE ieee.std_logic_1164.all;
 --USE ieee.std_logic_unsigned.all; -- <--GHDL Error primary unit "std_logic_unsigned" not found in library "ieee"

ENTITY PWMxPhDT IS
  GENERIC (
      sys_clk         : INTEGER := 50_000_000; --system clock frequency in Hz
      pwm_freq        : INTEGER := 50_000;     --PWM switching frequency in Hz
      n_bits          : INTEGER := 8;          --bits of resolution setting the duty cycle
      phases          : INTEGER := 2;          --number of PWM outputs and phases
      DT_val          : INTEGER := 1000         --Dead Time value in ns
		  );
	
  PORT    (    ---- info "STD_LOGIC_VECTOR" is a group of signals
      clk       : IN  STD_LOGIC;                             --system clock
      reset_n   : IN  STD_LOGIC;                             --asynchronous reset
      EN        : IN  STD_LOGIC;                              --latches in new duty cycle
      Duty_in   : IN  STD_LOGIC_VECTOR(n_bits-1 DOWNTO 0);   --duty cycle
      pwm_H     : OUT STD_LOGIC_VECTOR(phases-1 DOWNTO 0);   --pwm outputs
      pwm_L     : OUT STD_LOGIC_VECTOR(phases-1 DOWNTO 0)    --pwm inverse outputs
 --     DeadTime  : IN  STD_LOGIC_VECTOR(phases-1 DOWNTO 0)  --Delay between complementary outputs
          );
			 
END PWMxPhDT;

ARCHITECTURE logic OF PWMxPhDT IS
  CONSTANT period     :  INTEGER := sys_clk/pwm_freq;                           
  CONSTANT DT_cnt     :  INTEGER := DT_val/(1_000_000_000/sys_clk);             
  CONSTANT DutyVal     :  INTEGER := 64; -- compiler not like port val                   
  TYPE     counters IS ARRAY (0 TO phases-1) OF INTEGER RANGE 0 TO period - 1;  
  SIGNAL   Count          :  counters := (OTHERS => 0);                         
  SIGNAL   Half_Duty_New  :  INTEGER RANGE 0 TO period/2 := 0;                  
  TYPE     half_duties IS ARRAY (0 TO phases-1) OF INTEGER RANGE 0 TO period/2; 
  SIGNAL   Half_Duty      :  half_duties := (OTHERS => 0);

BEGIN
  PROCESS(clk, reset_n)
  BEGIN 
    IF(reset_n = '0') THEN                                                      
      Count <= (OTHERS => 0);                                                   
      pwm_H <= (OTHERS => '0');                                                 
      pwm_L <= (OTHERS => '0');                                                 
    ELSIF(clk'EVENT AND clk = '1') THEN                                         
      IF(EN = '1') THEN                                                         
       Half_Duty_New <= Duty_in * period / (2 ** n_bits) / 2; --determine clocks in 1/2 duty cycle
--       Half_Duty_New <= (128 * period) / (2 ** n_bits) / 2; -- Temp workaround
      END IF;
      FOR i IN 0 to phases-1 LOOP                                            
        IF(Count(0) = (period-1) - (i*period/phases)) THEN                   
          Count(i) <= 0;                                                     
          Half_Duty(i) <= Half_Duty_New;                                     
        ELSE                                                                 
          Count(i) <= Count(i) + 1;                                          
        END IF;
      END LOOP;

      FOR i IN 0 to phases-1 LOOP                                            
        IF(Count(i) = Half_Duty(i)) THEN                                     
          pwm_H(i) <= '0';                                                   
        ELSIF(Count(i) = Half_Duty(i) + DT_cnt ) THEN                        
          pwm_L(i) <= '1';                                                   
        ELSIF(Count(i) = (period - DT_cnt) - Half_Duty(i)) THEN              
          pwm_L(i) <= '0';                                                   
	     ELSIF(Count(i) = period - Half_Duty(i)) THEN                    
          pwm_H(i) <= '1';                                                   
        END IF;
      END LOOP;
    END IF;
  END PROCESS;
END logic;
```

umarcor, 2021-11-23:
Hi @Tonigau!

The version of GHDL you are using is rather old (from 2015). I recommend you use a more recent version.

On the other hand, see the following references with regard to the usage of `std_logic_unsigned`:

- https://stackoverflow.com/questions/45704135/when-to-use-vhdl-library-std-logic-unsigned-and-numeric-std
- http://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-fsynopsys
- http://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-ieee
- http://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-frelaxed

Tonigau, 2021-11-23:
Thank you for the reply,   I will digest & create some more neural connections.

closed 2021-11-23 by Tonigau