ghdl/ghdl #1922: How could I use my vhd libs with ghdl?

szhou888, 2021-11-23:
**Is your feature request related to a problem? Please describe.**
I write a lib which contains xilinx primitive,how could I use it by ghdl?

**Describe the solution you'd like**
I want to use a lib which contain some xilinx primitive (such as io\dsp ). I want to use ghdl synth vhdl designs which contain module in the libs. I want the used modules be transfer into nelist !

**Describe alternatives you've considered**
 I have tried to use "ghdl -a ..." to generate libs,but I found that ghdl use it as blackbox,how could ghdl synth it into netlist?

tgingold, 2021-11-23:
If it is in the library, it should work.
Do you have any warning about unbound components ?

szhou888, 2021-11-24:
> If it is in the library, it should work. Do you have any warning about unbound components ?

Year,such as
`/root/Desktop/ip/c_addsub_v12_0_vh_rfs.vhd:5737:5:warning: instance "i_baseblox_addsub" of component "c_addsub_v12_0_11_legacy" is not bound [-Wbinding]
    i_baseblox_addsub : c_addsub_v12_0_11_legacy
`
I have tried with mentor,it could synth the module into netlist!

tgingold, 2021-11-24:
Did you try the `--syn-binding` option ?

szhou888, 2021-11-25:
I used the command as"ghdl --synth -Wbinding -fsynopsys -fexplicit test.vhdl -e add_jidai1 >test.log" and warning just like this"warning: instance "i_baseblox_addsub" of component "c_addsub_v12_0_11_legacy" is not bound [-Wbinding]

    i_baseblox_addsub : c_addsub_v12_0_11_legacy;......"

tgingold, 2021-11-25:
So, did you try to add th `--syn-binding` option ?

szhou888, 2021-11-25:
> So, did you try to add th `--syn-binding` option ?
Year,I have tried it! [the second is the warning after use --syn-binding](https://github.com/ghdl/ghdl/issues/1922#issuecomment-978679200).
Currently,using syn-binding,I meet another problem as this
`/usr/local/lib/ghdl/ise/src/unisims/primitive/PLLE2_ADV.vhd:972:28:error: no declaration for "pll_lockg" in package "vcomponents"
        unisim.VCOMPONENTS.PLL_LOCKG <= '0' when (STARTUP_WAIT_sig = '1' and locked_out_tmp = '0') else 'H';
                           ^
/usr/local/lib/ghdl/ise/src/unisims/primitive/PLLE2_ADV.vhd:972:28:error: target is not a signal name
        unisim.VCOMPONENTS.PLL_LOCKG <= '0' when (STARTUP_WAIT_sig = '1' and locked_out_tmp = '0') else 'H';`
But I found that VCOMPONENTS.vhd had define the signal.

tgingold, 2021-11-25:
If you still have unbound components, then some entities are missing and must be analyzed.

For `pll_lockg`, are you sure it is declared ?  If so, please create a reproducer.
On the other hand, I doubt that PLLE2_ADV is synthesizable.  It must be a blackbox.

szhou888, 2021-11-25:
> If you still have unbound components, then some entities are missing and must be analyzed.
> 
> For `pll_lockg`, are you sure it is declared ? If so, please create a reproducer. On the other hand, I doubt that PLLE2_ADV is synthesizable. It must be a blackbox.

First,the lib I analyzed as this
`file / "/root/Desktop/ip/c_addsub_v12_0_vh_rfs.vhd" "ee4a81ca864d9174c481d9d6c82612e802389984" "20211124072430.608":
  package c_addsub_v12_0_11_comp at 48( 2419) + 0 on 65;
  package c_addsub_v12_0_11_viv_comp at 156( 7618) + 0 on 66;
  package c_addsub_v12_0_11_pkg at 267( 13056) + 0 on 67 body;
  package body c_addsub_v12_0_11_pkg at 664( 30185) + 0 on 68;
  package c_addsub_v12_0_11_pkg_legacy at 1288( 59013) + 0 on 69 body;
  package body c_addsub_v12_0_11_pkg_legacy at 1727( 77325) + 0 on 70;
  entity c_addsub_v12_0_11_lut6_legacy at 2196( 97601) + 0 on 71;
  architecture rtl of c_addsub_v12_0_11_lut6_legacy at 2281( 101043) + 0 on 72;
  entity c_addsub_v12_0_11_base_legacy at 3650( 154195) + 0 on 73;
  architecture rtl of c_addsub_v12_0_11_base_legacy at 3735( 157589) + 0 on 74;
  entity c_addsub_v12_0_11_fabric_legacy at 5177( 212131) + 0 on 75;
  architecture synth of c_addsub_v12_0_11_fabric_legacy at 5268( 215605) + 0 on 76;
  entity c_addsub_v12_0_11_legacy at 5458( 223085) + 0 on 77;
  architecture synth of c_addsub_v12_0_11_legacy at 5556( 227326) + 0 on 78;
  entity c_addsub_v12_0_11_viv at 6370( 263414) + 0 on 79;
  architecture synth of c_addsub_v12_0_11_viv at 6453( 266987) + 0 on 80;
  entity c_addsub_v12_0_11 at 7020( 291128) + 0 on 81;
  architecture xilinx of c_addsub_v12_0_11 at 7073( 294648) + 0 on 82;`
It contained the the  c_addsub_v12_0_11_legacy.   

Secondly,In the unism_vcomp had the defined of pll_lockg,such as:
`library IEEE;
use IEEE.STD_LOGIC_1164.all;
package VCOMPONENTS is
attribute BOX_TYPE : string;
signal GSR : std_logic := '0';
signal GTS : std_logic := '0';
signal GWE : std_logic;
signal PLL_LOCKG : std_logic := 'H';
signal PROGB_GLBL : std_logic;
signal CCLKO_GLBL : std_logic; `

tgingold, 2021-11-25:
I think I can now see the problem:

If you pass a file during synthesis on the command line, the work library is not read.
So you should either analyze all the files and do: `ghdl synth [OPTIONS] -e TOP`, or pass all the files on the command line.

szhou888, 2021-11-26:
> So you should either analyze all the files and do: `ghdl synth [OPTIONS] -e TOP`, or pass all the files on the command line.

Why this error appeared :
**/root/Desktop/ip/c_addsub_v12_0_vh_rfs.vhd:5545:5:error: multiple assignments for "ovfl" offsets 0:0  
    ovfl     : OUT STD_LOGIC := '0';  
/root/Desktop/ip/c_addsub_v12_0_vh_rfs.vhd:5546:5:error: multiple assignments for "c_out" offsets 0:0 
    c_out    : OUT STD_LOGIC := '0'; 
/root/Desktop/ip/c_addsub_v12_0_vh_rfs.vhd:5548:5:error: multiple assignments for "q_ovfl" offsets 0:0 
    q_ovfl   : OUT STD_LOGIC := '0';
/root/Desktop/ip/c_addsub_v12_0_vh_rfs.vhd:5549:5:error: multiple assignments for "q_c_out" offsets 0:0** 

But in the file,there are two entity. such as:
### ENTITY c_addsub_v12_0_11_base_legacy IS
  GENERIC (
    c_width               : INTEGER;
    c_add_mode            : INTEGER;
    c_b_constant          : INTEGER;
    ci_b_value            : STD_LOGIC_VECTOR;
    c_ainit_val           : STRING;
    c_sinit_val           : STRING;
    c_bypass_enable       : INTEGER;
    c_bypass_low          : INTEGER;
    c_sync_enable         : INTEGER;
    c_sync_priority       : INTEGER;
    c_has_s               : INTEGER;
    c_has_q               : INTEGER;
    c_has_c_in            : INTEGER; -- borrow in and borrow out is handled through c_in and c_out ports to make life easier
    c_has_c_out           : INTEGER;
    c_has_q_c_out         : INTEGER;
    c_has_ovfl            : INTEGER;
    c_has_q_ovfl          : INTEGER;
    c_has_ce              : INTEGER;
    c_has_bypass          : INTEGER;
    c_has_aclr            : INTEGER;
    c_has_aset            : INTEGER;
    c_has_ainit           : INTEGER;
    c_has_sclr            : INTEGER;
    c_has_sset            : INTEGER;
    c_has_sinit           : INTEGER
    );

  PORT (
    a        : IN  STD_LOGIC_VECTOR(c_width-1 DOWNTO 0) := (OTHERS => '0');  -- input value
    b        : IN  STD_LOGIC_VECTOR(c_width-1 DOWNTO 0) := (OTHERS => '0');  -- input value
    clk      : IN  STD_LOGIC                              := '0';  -- clock
    add      : IN  STD_LOGIC                              := '1';  -- valid when addsub mode selected. 1 = add, 0 = subtract
    c_in     : IN  STD_LOGIC                              := '0';  -- carry in
    ce       : IN  STD_LOGIC                              := '1';  -- clock enable
    bypass   : IN  STD_LOGIC                              := '0';  -- sync load of b into reg
    aclr     : IN  STD_LOGIC                              := '0';  -- asynch init.
    aset     : IN  STD_LOGIC                              := '0';  -- asynch set.
    ainit    : IN  STD_LOGIC                              := '0';  -- asynch init.
    sclr     : IN  STD_LOGIC                              := '0';  -- synch init.
    sset     : IN  STD_LOGIC                              := '0';  -- synch set.
    sinit    : IN  STD_LOGIC                              := '0';  -- synch init.
    ovfl     : OUT STD_LOGIC := '0';
    c_out    : OUT STD_LOGIC := '0';
    **q_ovfl   : OUT STD_LOGIC := '0';**
    q_c_out  : OUT STD_LOGIC := '0';
    s        : OUT STD_LOGIC_VECTOR(c_width - 1 DOWNTO 0) := (others =>'0');  -- asynch output value
    q        : OUT STD_LOGIC_VECTOR(c_width - 1 DOWNTO 0) := (others =>'0')  -- synch output value
    );
END c_addsub_v12_0_11_base_legacy;

### ENTITY c_addsub_v12_0_11_legacy IS
  GENERIC (
    c_family              : STRING := "virtex7";
    c_a_width             : INTEGER;
    c_b_width             : INTEGER;
    c_out_width           : INTEGER;
[... 44 lines cut ...]
    b        : IN  STD_LOGIC_VECTOR(c_b_width-1 DOWNTO 0) := (OTHERS => '0');  -- input value
    clk      : IN  STD_LOGIC                              := '0';  -- clock
    add      : IN  STD_LOGIC                              := '1';  -- valid when addsub mode selected. 1 = add, 0 = subtract
    c_in     : IN  STD_LOGIC                              := '0';  -- carry in
    b_in     : IN  STD_LOGIC                              := '1';  -- borrow in
    ce       : IN  STD_LOGIC                              := '1';  -- clock enable
    bypass   : IN  STD_LOGIC                              := '0';  -- sync load of b into reg
    aclr     : IN  STD_LOGIC                              := '0';  -- asynch init.
    aset     : IN  STD_LOGIC                              := '0';  -- asynch set.
    ainit    : IN  STD_LOGIC                              := '0';  -- asynch init.
    sclr     : IN  STD_LOGIC                              := '0';  -- synch init.
    sset     : IN  STD_LOGIC                              := '0';  -- synch set.
    sinit    : IN  STD_LOGIC                              := '0';  -- synch init.
    a_signed : IN  STD_LOGIC                              := '0';  -- '0' = unsigned, '1' = signed.
    b_signed : IN  STD_LOGIC                              := '0';  -- '0' = unsigned, '1' = signed.
    ovfl     : OUT STD_LOGIC := '0';
    c_out    : OUT STD_LOGIC := '0';
    b_out    : OUT STD_LOGIC := '0';           -- borrow out
    **q_ovfl   : OUT STD_LOGIC := '0';**
    q_c_out  : OUT STD_LOGIC := '0';
    q_b_out  : OUT STD_LOGIC := '0';
    s        : OUT STD_LOGIC_VECTOR(c_out_width - 1 DOWNTO 0) := (others => '0');  -- asynch output value
    q        : OUT STD_LOGIC_VECTOR(c_out_width - 1 DOWNTO 0) := (others => '0')  -- synch output value
    );
END c_addsub_v12_0_11_legacy;

tgingold, 2021-11-26:
It means those signals are assigned in different processes, which is not valid.
A reproducer is needed to investigate.

szhou888, 2021-11-26:
> reproducer

Ok,thanks so much! Although，I don't know how to investigate it!

tgingold, 2021-11-29:
You have to look at the sources and check how the erroneous signals are assigned.  Probably, they are assigned twice.

szhou888, 2021-12-03:
> You have to look at the sources and check how the erroneous signals are assigned. Probably, they are assigned twice.

ok,thanks so much! I found this problem as you said.Next,I would try to find the reason of difference between mentor with ghdl !