ghdl/ghdl #1923: assertion warning does not give enought information

NicoPy, 2021-11-24:
When starting debugging a design, I often get _assertion warning_ that are only one line like this one :
`../../src/ieee2008/numeric_std-body.vhdl:3036:7:@40690104fs:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0`

The problem is to find the culprit. There is not enough information to easily find it.
Is it possible to print where in the hierarchy the assertion has been raised.

I have added the `--backtrace-severity=warning` option to `ghdl -r` without success.
I use ghdl on MSYS2 (nightly release).
```
GHDL 2.0.0-dev (1.0.0.r893.gba783ea0) [Dunoon edition]
 Compiled with GNAT Version: 11.2.0
 llvm code generator
```

I'm aware of issue #1578 but wonder if it applies to my use case.

tgingold, 2021-11-24:
Backtraces are not reliable on windows x32.  Unfortunately.

NicoPy, 2021-11-24:
What about Windows x64 ?

tgingold, 2021-11-25:
Worth trying!

NicoPy, 2021-11-25:
Is adding `--backtrace-severity=warning` option to `ghdl -r` enough ?

NicoPy, 2021-11-25:
I've added _libbacktrace_ to my MSYS2 installation with `pacman -S mingw64/mingw-w64-x86_64-libbacktrace`.
This had no effect.
This is not a surprise since, if I understood correctly, LLVM has to be compiled with _libbacktrace_ which doesn't seems to be done here. Right ?

umarcor, 2021-11-25:
Enabling libbacktraces on Windows (MSYS2) and containers (ghdl/docker#35) is in the backlog. Hence, it would be nice if you could give it a try!

Apart from installing libbacktrace, as you did, the recipe in https://github.com/ghdl/ghdl/blob/master/scripts/msys2-llvm/PKGBUILD needs to be modified in order to add argument `--with-backtrace-lib=` in the call to configure. I vaguely recall trying it, but I can't find the branch...

NicoPy, 2021-11-29:
> Apart from installing libbacktrace, as you did, the recipe in https://github.com/ghdl/ghdl/blob/master/scripts/msys2-llvm/PKGBUILD needs to be modified in order to add argument `--with-backtrace-lib=` in the call to configure. I vaguely recall trying it, but I can't find the branch...

For this I have to compile GHDL myself. Right ?

umarcor, 2021-11-29:
Yes. However, it's pretty straightforward on MSYS2: https://github.com/ghdl/ghdl/blob/852b1e064f81c7c7108c45847872a266e995d31c/.github/workflows/Test.yml#L263-L264

First, you need to install the base dependencies: https://github.com/ghdl/ghdl/blob/852b1e064f81c7c7108c45847872a266e995d31c/.github/workflows/Test.yml#L248-L249

Assuming you are going to build GHDL with LLVM backend on MINGW64: `pacman -S base-devel git mingw-w64-x86_64-toolchain`.

NicoPy, 2021-11-30:
I did a full update of MSYS2 to get the toolchain in version 11.2.0 .
```
pacman -Syu
pacman -Su
```
I cloned the GHDL git repo locally.

I added `--with-backtrace-lib` to `scripts/msys2-llvm/PKGBUILD` :
```
build() {
  mkdir "${srcdir}/builddir"
  cd "${srcdir}/builddir"
  export CC=clang
  export CXX=clang++
  ../../../../configure \
    --prefix=${MINGW_PREFIX} \
    --with-llvm-config="llvm-config --link-static" \
    LDFLAGS="-static" \
    --enable-libghdl \
    --enable-synth \
    --with-backtrace-lib="/c/msys64/mingw64/lib/libbacktrace.a"
  make GNATMAKE="gnatmake -j$(nproc)"
}
```
I know the path to `libbacktrace` should not be hardcoded. This is just for the test, to be sure it points to a correct location.

I then run 
` makepkg-mingw --noconfirm --noprogressbar -sCLf` 

This created a package that I installed with 
`pacman -U mingw-w64-x86_64-ghdl-llvm-ci-1-any.pkg.tar.zst`

Looks like it has been installed correctly :
```
$ ghdl -v
GHDL 2.0.0-dev (1.0.0.r901.g852b1e06.dirty) [Dunoon edition]
 Compiled with GNAT Version: 11.2.0
 llvm code generator
Written by Tristan Gingold.
```

Alas, I still don't get any backtrace on warning assertions.

NicoPy, 2021-11-30:
I can check that the generated `.exe` is linked with `libbacktrace` :
```
$ ldd test_tb.exe
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ffd71ed0000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ffd70f50000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ffd6fae0000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ffd71d90000)
        dbghelp.dll => /c/WINDOWS/SYSTEM32/dbghelp.dll (0x7ffd56ca0000)
        zlib1.dll => /mingw64/bin/zlib1.dll (0x7ffd4b490000)
        libbacktrace-0.dll => /mingw64/bin/libbacktrace-0.dll (0x7ffd4eae0000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ffd6f9e0000)
        libgcc_s_seh-1.dll => /mingw64/bin/libgcc_s_seh-1.dll (0x7ffd4b470000)
        libwinpthread-1.dll => /mingw64/bin/libwinpthread-1.dll (0x7ffd4b450000)
```

tgingold, 2021-11-30:
Ok.  Few questions:

1) Do you use only Windows ?

2) What is the output when you run test_tb.exe ?

3) Can you run on gdb, set a breakpoint on <grt__lib__do_report>, and do `bt` when the breakpoint is hit ?

umarcor, 2021-11-30:
> I know the path to `libbacktrace` should not be hardcoded. This is just for the test, to be sure it points to a correct location.

You might have used `${MINGW_PREFIX}`, however, that is irrelevant. The procedure you explained looks correct from a build/install point of view. Please, follow @tgingold's guidelines, since he can better know what is failing "internally".

NicoPy, 2021-12-01:
> @NicoPy, you might have used `${MINGW_PREFIX}`, however, that is irrelevant. 

Thanks for the information. I will not have to search for it when I need it 😉

NicoPy, 2021-12-01:
> 1) Do you use only Windows ? 

Yes

> 2) What is the output when you run test_tb.exe ? 

`../../src/ieee2008/numeric_std-body.vhdl:2025:7:@5ns:(assertion warning): NUMERIC_STD."/=": metavalue detected, returning TRUE`

> 3) Can you run on gdb, set a breakpoint on <grt__lib__do_report>, and do `bt` when the breakpoint is hit ?

The command I used is `gdb --args test_tb  --backtrace-severity=warning`.

```
Reading symbols from test_tb...
(gdb) break grt__lib__do_report
Breakpoint 1 at 0x1400570eb
(gdb) run
Starting program: Z:\ghdl\test\test_backtrace\test_tb.exe "--backtrace-severity=warning"
[New Thread 11972.0x32a8]
[New Thread 11972.0x30ac]
[New Thread 11972.0x2608]

Thread 1 hit Breakpoint 1, 0x00007ff780b270eb in grt.lib.do_report ()
(gdb) bt
#0  0x00007ff780b270eb in grt.lib.do_report ()
#1  0x00007ff780b27650 in __ghdl_ieee_assert_failed ()
#2  0x00007ff780af9184 in ieee__numeric_std__OPNeO5 ()
    at ../../src/ieee2008/numeric_std-body.vhdl:2018
#3  0x00007ff780b0c51a in work__test_tb__ARCH__behave__check_process__PROC ()
    at test_tb.vhdl:38
#4  0x00007ff780b0f902 in grt.processes.run_processes ()
#5  0x00007ff780b0fea5 in grt.processes.simulation_cycle ()
#6  0x00007ff780b10431 in grt.processes.simulation_main_loop ()
#7  0x00007ff780b104a9 in grt.processes.simulation ()
#8  0x00007ff780b28efd in __ghdl_run_through_longjump ()
#9  0x00007ff780b28da6 in grt.main.run_simul ()
#10 0x00007ff780b28e7c in grt.main.run ()
#11 0x00007ff780b43f8a in ghdl_main ()
#12 0x00007ff780b29393 in main ()
```
The files I use : [test_backtrace.zip](https://github.com/ghdl/ghdl/files/7635401/test_backtrace.zip)

umarcor, 2021-12-01:
> I vaguely recall trying it, but I can't find the branch...

I found it now: https://github.com/umarcor/MINGW-packages/commit/8bf684d466a67013bcea81b66fbbbec9f383b955.

Anyway, as commented above, it is equivalent to what you did.

tgingold, 2021-12-01:
Ok, so using gdb you can have the full backtrace: "/=" is called from test_tb.vhdl:38

If gdb is able to get the backtrace, then the debug info are correctly generated.
I don't know why libbacktrace is not able to show all the lines.  I currently don't have a windows box, so it will be difficult to investigate.

tgingold, 2021-12-03:
Yes.  Can you also run under gdb but set a breakpoint on <grt__backtraces__put_err_bracktrace>.
When the breakpoint is hit, do:
(gdb) bt
(gdb) print bt
(gdb) c

The purpose is to know if the backtrace is known or if it is not fully symbolized.

Thanks!

NicoPy, 2021-12-06:
Here are the commands and output :
```
$ gdb --args test_tb  --backtrace-severity=warning
GNU gdb (GDB) 11.1
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from test_tb...
(gdb) set breakpoint pending on
(gdb) break grt__backtraces__put_err_bracktrace
Function "grt__backtraces__put_err_bracktrace" not defined.
Breakpoint 1 (grt__backtraces__put_err_bracktrace) pending.
(gdb) run
Starting program: Z:\CantarMini\Xilinx\CantarMini\Sources\test\test_backtrace\test_tb.exe "--backtrace-severity=warning"
[New Thread 12412.0x1a34]
[New Thread 12412.0x2e94]
[New Thread 12412.0x67d8]
../../src/ieee2008/numeric_std-body.vhdl:2025:7:@5ns:(assertion warning): NUMERIC_STD."/=": metavalue detected, returning TRUE
../../src/ieee2008/numeric_std-body.vhdl:2025:7:@15ns:(assertion warning): NUMERIC_STD."/=": metavalue detected, returning TRUE
simulation finished @20ns
[Thread 12412.0x67d8 exited with code 0]
[Thread 12412.0x1a34 exited with code 0]
[Thread 12412.0x2e94 exited with code 0]
[Inferior 1 (process 12412) exited normally]
(gdb)
```
So, function `grt__backtraces__put_err_bracktrace` is _statically_ unknown and not found at run time either.

avelure, 2021-12-13:
Is it possible `grt__backtraces__put_err_bracktrace` was supposed to be `grt__backtraces__put_err_backtrace` ?

tgingold, 2021-12-13:
Yes, sorry for the typo (and for missing your previous reply.

NicoPy, 2021-12-14:
> Is it possible `grt__backtraces__put_err_bracktrace` was supposed to be `grt__backtraces__put_err_backtrace` ?

You are right, I didn't noticed it (a side effect of copy/paste).

Here is the output with the correct spelling :
```
$ gdb --args test_tb  --backtrace-severity=warning
GNU gdb (GDB) 11.1
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from test_tb...
(gdb) break grt__backtraces__put_err_backtrace
Breakpoint 1 at 0x14006b270
(gdb) run
Starting program: Z:\CantarMini\Xilinx\CantarMini\Sources\test\test_backtrace\test_tb.exe "--backtrace-severity=warning"
[New Thread 14884.0x2598]
[New Thread 14884.0x2688]
[New Thread 14884.0x5884]

Thread 1 hit Breakpoint 1, 0x00007ff6eb88b270 in grt.backtraces.put_err_backtrace ()
```

tgingold, 2021-12-14:
Yes, but when the breakpoint is hit, you need to enter `bt` and `print bt` to see if the backtrace succeed.

I may eventually have access to a Win64 machine on January, that might help.

NicoPy, 2021-12-15:
> Yes, but when the breakpoint is hit, you need to enter `bt` and `print bt` to see if the backtrace succeed.

Sorry, I forgot these steps. Here it is :
```
$ gdb --args test_tb  --backtrace-severity=warning
GNU gdb (GDB) 11.1
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from test_tb...
(gdb) break grt__backtraces__put_err_backtrace
Breakpoint 1 at 0x14006b270
(gdb) run
Starting program: Z:\CantarMini\Xilinx\CantarMini\Sources\test\test_backtrace\test_tb.exe "--backtrace-severity=warning"
[New Thread 8808.0x4644]
[New Thread 8808.0x3de8]
[New Thread 8808.0x2244]

Thread 1 hit Breakpoint 1, 0x00007ff62848b270 in grt.backtraces.put_err_backtrace ()
(gdb) bt
#0  0x00007ff62848b270 in grt.backtraces.put_err_backtrace ()
#1  0x00007ff6284774e4 in grt.lib.do_report ()
#2  0x00007ff628477650 in __ghdl_ieee_assert_failed ()
#3  0x00007ff628449184 in ieee__numeric_std__OPNeO5 ()
    at ../../src/ieee2008/numeric_std-body.vhdl:2018
#4  0x00007ff62845c51a in work__test_tb__ARCH__behave__check_process__PROC ()
    at test_tb.vhdl:38
#5  0x00007ff62845f902 in grt.processes.run_processes ()
#6  0x00007ff62845fea5 in grt.processes.simulation_cycle ()
#7  0x00007ff628460431 in grt.processes.simulation_main_loop ()
#8  0x00007ff6284604a9 in grt.processes.simulation ()
#9  0x00007ff628478efd in __ghdl_run_through_longjump ()
#10 0x00007ff628478da6 in grt.main.run_simul ()
#11 0x00007ff628478e7c in grt.main.run ()
#12 0x00007ff628493f8a in ghdl_main ()
#13 0x00007ff628479393 in main ()
(gdb) print bt
No symbol "bt" in current context.
```

tgingold, 2021-12-17:
Apparently, the grt library was not compiled with debug info.