ghdl/ghdl #1924: ieee.float_generic_pkg with generic FLOAT_ROUND_STYLE = ieee.fixed_float_types.ROUND_ZERO does not truncate

NicoPy, 2021-11-25:
I use the following packages

```vhd :file: Package_Float32.vhd
library ieee;

package Package_Float32 is new ieee.float_generic_pkg
  generic map (
    FLOAT_EXPONENT_WIDTH => 8,    -- float32'high
    FLOAT_FRACTION_WIDTH => 23,   -- -float32'low
    FLOAT_ROUND_STYLE    => ieee.fixed_float_types.ROUND_ZERO,  -- Truncate
    FLOAT_DENORMALIZE    => false,  -- Use IEEE extended floating
    FLOAT_CHECK_ERROR    => false,  -- Turn on NAN and overflow processing
    FLOAT_GUARD_BITS     => 0,     -- number of guard bits
    NO_WARNING           => false, -- show warnings
    FIXED_PKG            => work.Package_Fixed
    );
```
```vhd :file: Package_Fixed.vhd
library ieee;

package Package_Fixed is new ieee.fixed_generic_pkg
  generic map (
    FIXED_ROUND_STYLE    => ieee.fixed_float_types.FIXED_TRUNCATE,
    FIXED_OVERFLOW_STYLE => ieee.fixed_float_types.FIXED_SATURATE,
    FIXED_GUARD_BITS     => 0,
    NO_WARNING           => false
    );
```

Please note the use of `ieee.fixed_float_types.ROUND_ZERO`  (Truncate) and `ieee.fixed_float_types.FIXED_TRUNCATE` since I want all calculations done with truncation.

Here is my test bench :
```vhd :file: test_Float_To_sFixed.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.Package_Float32.all;
use work.Package_Fixed.all;

entity test_Float_To_sFixed is
end entity test_Float_To_sFixed;

architecture RTL of test_Float_To_sFixed is

procedure compute_and_show(sign : in std_logic; exponent : in Integer; fraction : in std_logic_vector(23-1 downto 0)) is
    variable s  : std_logic_vector(32-1 downto 0);
    variable f  : float32;
    variable sf : sFixed(4-1 downto -2);
    variable r  : Real; 
begin
        s  := sign & std_logic_vector(to_unsigned(exponent, 8)) & fraction;
        f  := to_float(s, f);
        sf := to_sFixed(f, sf);
        r  := to_real(f);
        
        report "s  : " & to_string(sign) & "." 
                       & to_string(to_unsigned(exponent, 8)) & "." 
                       & to_string(fraction)
                       & " -> " & to_string(r);
        report "sf : " & to_string(sf);
        report "";
end compute_and_show;
        

begin
    
    process
    begin
        compute_and_show('0',   0, "00000000000000000000000");  -- 0
        compute_and_show('0', 123, "00000000000000000000000");  -- 0.0625
        compute_and_show('0', 124, "00000000000000000000000");  -- 0.125
        compute_and_show('0', 125, "00000000000000000000000");  -- 0.25
        compute_and_show('0', 126, "00000000000000000000000");  -- 0.5
        compute_and_show('0', 127, "00000000000000000000000");  -- 1
        compute_and_show('0', 127, "10000000000000000000000");  -- 1.5
        compute_and_show('0', 127, "11000000000000000000000");  -- 1.75
        compute_and_show('0', 127, "11100000000000000000000");  -- 1.875
        compute_and_show('0', 127, "11110000000000000000000");  -- 1.875
        report "------";       
        report ""; 
        compute_and_show('1',   0, "00000000000000000000000");  -- -0
        compute_and_show('1', 123, "00000000000000000000000");  -- -0.0625
        compute_and_show('1', 124, "00000000000000000000000");  -- -0.125
        compute_and_show('1', 125, "00000000000000000000000");  -- -0.25
        compute_and_show('1', 126, "00000000000000000000000");  -- -0.5
        compute_and_show('1', 127, "00000000000000000000000");  -- -1
        compute_and_show('1', 127, "10000000000000000000000");  -- -1.5
        compute_and_show('1', 127, "11000000000000000000000");  -- -1.75
        compute_and_show('1', 127, "11100000000000000000000");  -- -1.875
        compute_and_show('1', 127, "11110000000000000000000");  -- -1.875
        wait;
    end process;

end architecture RTL;
```
In this code, a `Float32` number is created and converted to a `sFixed`number. For readability, the integer part of the fixed point number is 4 bits. The fractional part is 2 bits to easily show the problem.

The test output is the following :
```
s  : 0.00000000.00000000000000000000000 -> 0.0
sf : 0000.00

s  : 0.01111011.00000000000000000000000 -> 6.25e-2
sf : 0000.00

s  : 0.01111100.00000000000000000000000 -> 1.25e-1
sf : 0000.00

s  : 0.01111101.00000000000000000000000 -> 2.5e-1
sf : 0000.01

s  : 0.01111110.00000000000000000000000 -> 5.0e-1
sf : 0000.10

s  : 0.01111111.00000000000000000000000 -> 1.0
sf : 0001.00

s  : 0.01111111.10000000000000000000000 -> 1.5
sf : 0001.10

s  : 0.01111111.11000000000000000000000 -> 1.75
sf : 0001.11

s  : 0.01111111.11100000000000000000000 -> 1.875
sf : 0001.11

s  : 0.01111111.11110000000000000000000 -> 1.9375
sf : 0001.11

------

s  : 1.00000000.00000000000000000000000 -> 0.0
sf : 0000.00

s  : 1.01111011.00000000000000000000000 -> -6.25e-2
sf : 1111.11

s  : 1.01111100.00000000000000000000000 -> -1.25e-1
sf : 1111.11

s  : 1.01111101.00000000000000000000000 -> -2.5e-1
sf : 1111.11

s  : 1.01111110.00000000000000000000000 -> -5.0e-1
sf : 1111.10

s  : 1.01111111.00000000000000000000000 -> -1.0
sf : 1111.00

s  : 1.01111111.10000000000000000000000 -> -1.5
sf : 1110.10

s  : 1.01111111.11000000000000000000000 -> -1.75
sf : 1110.01

s  : 1.01111111.11100000000000000000000 -> -1.875
sf : 1110.00

s  : 1.01111111.11110000000000000000000 -> -1.9375
sf : 1110.00
```
One can see that with positive numbers, the conversion correctly truncates numbers :
```
0.0625 -> 0000.00  ok
0.125  -> 0000.00 ok
0.25   -> 0000.01 ok
0.5    -> 0000.10 ok
1.75   -> 0001.11 ok
1.875  -> 0001.11 ok (truncated)
1.9375 -> 0001.11 ok (truncated)
```

However, with negative values, this is not true :
```
-0.0625 -> 1111.11 -> -0.25 NOT ok, should be 0
-0.125  -> 1111.11 -> -0.25 NOT ok, should be 0
-0.25   -> 1111.11 ok
-0.5    -> 1111.10 ok
-1.75   -> 1110.01 ok
-1.875  -> 1110.00 -> -2 NOT ok, should be -1.75
-1.9375 -> 1110.00 -> -2 NOT ok, should be -1.75
```

Unless I misunderstood something, negative numbers are not truncated correctly.
Here is a zip with files : [test.zip](https://github.com/ghdl/ghdl/files/7603955/test.zip)

NicoPy, 2021-11-25:
I forgot the compilation commands :
```batch
ghdl -i  --std=08  --workdir=sim_build --work=work test_Float_To_sFixed.vhd Package_Float32.vhd Package_Fixed.vhd
ghdl -m  --std=08  --workdir=sim_build -Psim_build --work=work test_Float_To_sFixed
ghdl -r  --workdir=sim_build -Psim_build --work=work test_Float_To_sFixed
```

The updated zip : [test.zip](https://github.com/ghdl/ghdl/files/7603988/test.zip)

tgingold, 2021-11-25:
Those packages come from the IEEE standard.  I will have a closer look at your test, but I don't plan to fix these packages.
You can open an issue to the VHDL standard.

tgingold, 2021-11-25:
Maybe it is a specification issue.  I have the impression that the number is truncated, which is different from rounded toward zero (if 2 complement is used).

tgingold, 2021-11-25:
So if you consider the fixed point rounding mode should be used to round the result of a conversion from fp to fixed, the results look OK, don't they ?

NicoPy, 2021-11-25:
> Those packages come from the IEEE standard.  

That's what I thought but I was not sure.

> I will have a closer look at your test, but I don't plan to fix these packages.

That's not a problem for my current project.  I've written my own Float_To_sFixed entity for minimum resource usage. Well... I hope so ;) 
I use the ieee package to check my design by comparing the output of my entity and the output of to_sFixed(). 
During my tests, I noticed inconsistencies. That's what I've reported here. 

> You can open an issue to the VHDL standard.

I have no idea how to do that ;)

> Maybe it is a specification issue.

Maybe. Is the LRM freely accessible ? 

> So if you consider the fixed point rounding mode should be used to round the result of a conversion from fp to fixed, the results look OK, don't they ?

Hmmm.... Not sure about that.

umarcor, 2021-11-25:
> > You can open an issue to the VHDL standard.
> 
> I have no idea how to do that ;)

See [gitlab.com/IEEE-P1076](https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues) and [IEEE-P1076.gitlab.io](https://IEEE-P1076.gitlab.io/).

> > Maybe it is a specification issue.
> 
> Maybe. Is the LRM freely accessible ?

It is not publicly available. However, see https://ieee-p1076.gitlab.io/index.html#participating and https://ieee-p1076.gitlab.io/About/index.html#about-admin. Let me know your GitLab handle, and/or your Name, Surname, Country and e-mail. You can send me an e-mail, or contact through gitter.

tgingold, 2021-11-26:
Looking again, I tend to agree with you.

Sorry for being slow, but I have never used those packages.

I will have a closer look.

NicoPy, 2021-11-26:
> Sorry for being slow, but I have never used those packages.

No problem Tristan. This is the first time I use those packages so I'm not very confident with my assertions.

> I will have a closer look.

Thanks

I think the behavior I observed is equivalent to `FLOAT_ROUND_STYLE => ieee.fixed_float_types.ROUND_NEGINF`.
I will do complementary tests and report the results here.

NicoPy, 2021-11-26:
@umarcor Thanks. I will have a look at those information later.

tgingold, 2021-11-26:
Wait.  The problem occurs when a real is converted to a fixed point value.  For that, only the FIXED_ROUND_STYLE matters (and it is set to FIXED_TRUNCATE).

The body of to_sfixed is declared in float_generic_pkg-body.vhdl at line 3532 and call the to_sfixed function declared at line 3370.

NicoPy, 2021-11-26:
> The problem occurs when a real is converted to a fixed point value.

Which line do you refer to ?

NicoPy, 2021-11-26:
>  For that, only the FIXED_ROUND_STYLE matters (and it is set to FIXED_TRUNCATE).

Right. Not obvious. Well done. 👍 
That explains why when I tried all `float_round_style` values I got no difference on the results 🙄 
When reading source code, I confused `float_round_style`  with `fixed_round_style` in the function definition.

Changing `FIXED_ROUND_STYLE    => ieee.fixed_float_types.FIXED_TRUNCATE` to `FIXED_ROUND_STYLE    => ieee.fixed_float_types.FIXED_ROUND` in Package_Fixed.vhd gives correct results (for what I want to achieve).

Many thanks for your support.

closed 2021-11-27 by tgingold