ghdl/ghdl #1925: Installation of gnat and gcc-7 issue

arielpodlubne, 2021-11-26:
Hi,

I am trying to install GHDL with:

    sudo apt update
    sudo apt install -y git make gnat zlib1g-dev
    git clone https://github.com/ghdl/ghdl
    cd ghdl
    ./configure --prefix=/usr/local
    make
    sudo make install

The issue comes when I want to install gnat:

    sudo apt install gnat    
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    Some packages could not be installed. This may mean that you have
    requested an impossible situation or if you are using the unstable
    distribution that some required packages have not yet been created
    or been moved out of Incoming.
    The following information may help to resolve the situation:
    
    The following packages have unmet dependencies:
     gnat : Depends: gnat-7 but it is not going to be installed
    E: Unable to correct problems, you have held broken packages.

I found [this post](https://askubuntu.com/questions/1260658/error-trying-to-install-dependencies-for-gnat) where the first answer states that it is related to gcc-7 (which is what I currently also have). The solution is to downgrade to gcc-5. So, does that mean that ghdl won't work with gcc-7?

Thanks for the help.

umarcor, 2021-11-26:
libghdl and synthesis features do require gnat-8, so rather than downgrading you might want to update your system. Which distribution are you using?

arielpodlubne, 2021-11-26:
```
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic
```

umarcor, 2021-11-26:
@arielpodlubne on Ubuntu 18.04, you can use the following packages:

For mcode:

```
    ca-certificates
    gcc
    gnat-7
    make
    zlib1g-dev
```

For LLVM, also:

```
    clang-5.0
    llvm-5.0-dev
```

See https://github.com/ghdl/docker/blob/master/build_debian.dockerfile and https://github.com/ghdl/docker/blob/master/run.sh#L120-L123. See the complete build log:

- With mcode: https://github.com/ghdl/docker/runs/4329322050?check_suite_focus=true
- With LLVM: https://github.com/ghdl/docker/runs/4329322169?check_suite_focus=true

arielpodlubne, 2021-11-29:
@umarcor thanks for the help. I also tried with docker and it worked.....there is something on my system that I need to figure out what's wrong.

closed 2021-11-29 by arielpodlubne