ghdl/ghdl #1926: recent GHDL's won't compile ecp5pll.vhd with moderately complex compile-time function

emard, 2021-11-28:
here is the source that fails to compile ecp5pll.vhd
https://github.com/emard/ulx3s-misc/tree/master/examples/ecp5pll

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r893.gba783ea0.dirty) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/home/guest/src/fpga/ulx3s/ulx3s-misc/examples/ecp5pll/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : elab-vhdl_expr.adb:1204
******************************************************************
this happens with GHDL binary from
https://github.com/YosysHQ/oss-cad-suite-build/releases

somewhat older GHDL binary compies it ok
https://github.com/YosysHQ/fpga-toolchain/releases/tag/nightly-20211006
vendor specific lattice diamond compiles it too

this function is most complex so I expect to problem be around it:
https://github.com/emard/ulx3s-misc/blob/master/examples/ecp5pll/hdl/vhd/ecp5pll.vhd#L112

umarcor, 2021-11-28:
@emard, can you please provide reproducible steps? What do you mean with "compile", synthesis or simulation?

emard, 2021-11-29:
Synthesis to ECP5 for yosys and nextpnr
I just type make here but don't know how to adapt to
your compilation environment

tgingold, 2021-11-29:
I can reproduce the issue and I confirm this is a regression.
I am investigating.

tgingold, 2021-11-29:
Should be fixed.  thank you for the issue.

closed 2021-11-29 by tgingold

emard, 2021-11-29:
Wonderful, it was fast! I'll pull new nightly build of ghdl and check
how it works now

umarcor, 2021-11-29:
@emard, note that you need to wait ~1h for the nightly to be updated: https://github.com/ghdl/ghdl/actions/runs/1517727123

emard, 2021-11-29:
Yes yes I will pull in 1-2 h!

commit 49a142e5 (2021-11-29): synth/elab-vhdl_expr: handle slices and indexed names.  Fix #1926